john30/ebusd #1201: Cannot set WLAN client 64 char password

BastelBaus, 2024-02-21:
### Description

Trying to configure my WLAN password but seams I cannot set a 64 char password (only 63 chars)
[image]
[image]

Build: 20240106

### Actual behavior

can only use 63 chars

### Expected behavior

64 chars should still be OK

### ebusd version

current source from git

### ebusd arguments

used 

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via USB

### Logs

n/a

BastelBaus, 2024-02-21:
My quick fix was using repl command and set the nvs values manually ...
[image]

john30, 2024-02-27:
wrong repo. moved to https://github.com/john30/ebusd-esp32/issues/50

closed 2024-02-27 by john30