john30/ebusd #1204: ebusd Valliant Arotherm polling problem

chheiss, 2024-02-24:
### Description

The new 08.csv value hmu BuildingCircuitFlow is not polling/updating. 

### Actual behavior

Value is only put on mqtt when I issue: ebusctl read -f -V -c hmu BuildingCircuitFlow
No automatic polling.

### Expected behavior

Value should be polled/updated automatically

### ebusd version

23.3

### ebusd arguments

ebusctl read -f -V -c hmu BuildingCircuitFlow

### Operating system

Debian 10 (Buster) / Ubuntu 18-19 / Raspbian 10 / Raspberry Pi OS 10 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter v3 USB

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

2024-02-24 06:25:21.212 [update notice] received poll-read basv z1ActualRoomTempDesired QQ=f1: 22
2024-02-24 06:25:21.429 [update notice] received read vr_71 SensorData2 QQ=10: -;-;-;-;0.00;8.00;80 57 05
2024-02-24 06:25:21.565 [update notice] received poll-read basv z1OpMode QQ=f1: auto
2024-02-24 06:25:21.754 [update notice] received update-write hmu SetMode QQ=10: auto;58.0;-;-;0;1;1;0;0;0
2024-02-24 06:25:22.215 [update notice] received unknown MS cmd: f115b52406020003012200 / 08020322000000a841
2024-02-24 06:25:22.419 [update notice] received write vr_71 SetActorState QQ=10: -;-;-;off;off;off;-;-;-;-;-;-;00 00
2024-02-24 06:25:22.576 [update notice] received poll-read basv z2ActualRoomTempDesired QQ=f1: 22
2024-02-24 06:25:22.712 [update notice] received read hmu State QQ=71: 100;253;193;8
2024-02-24 06:25:22.810 [update notice] received write vr_71 Mc1FlowTempDesired QQ=10: on;34.0;on;0
2024-02-24 06:25:22.951 [update notice] received poll-read basv z2OpMode QQ=f1: auto
2024-02-24 06:25:23.083 [update notice] received write vr_71 Mc2FlowTempDesired QQ=10: on;38.0;on;25
2024-02-24 06:25:23.234 [update notice] received unknown MS cmd: f115b52406020003011500 / 08010315000000c642
2024-02-24 06:25:23.313 [update notice] received unknown MS cmd: 0376b5120613000b4b0400 / 0200ff
2024-02-24 06:25:23.428 [update notice] received write vr_71 Mc3FlowTempDesired QQ=10: on;35.5;on;25
2024-02-24 06:25:23.598 [update notice] received unknown MS cmd: f115b52406020003022200 / 080203220000009041
2024-02-24 06:25:23.774 [update notice] received poll-read basv z3ActualRoomTempDesired QQ=f1: 19

mwildbolz, 2024-02-24:
Automatic polling has to be activated within config files. You have to set the message type to r1-r9 - depending on the needed poll priority.

See [ebusd Wiki](https://github.com/john30/ebusd/wiki/4.1.-Message-definition) for further details.

BR

john30, 2024-02-29:
does not sound like an ebusd issue to me, rather ebusd-configuration patch needed.
indeed: https://github.com/john30/ebusd-configuration/blob/cf37c58578857a723a14871a50feb6c5bdd53cd8/ebusd-2.1.x/de/vaillant/08.hmu.csv#L48 reveals it is not set to polling. so issue a PR there

closed 2024-02-29 by john30