john30/ebusd #1210: Docker-Compose EBUSD_READONLY leads to "cannot combine readonly with answer/generatesyn/initsend/scanconfig"

Kolbi, 2024-02-29:
### Description

My Portainer docker-compose:

services:
  ebusd:
    image: john30/ebusd
    container_name: ebusd
    restart: unless-stopped
    ports:
      - 8888:8888
    environment:
      EBUSD_DEVICE: "ens:192.168.21.239:9999"
      EBUSD_READONLY: ""
    network_mode: "host"
    volumes:
      - "/etc/localtime:/etc/localtime:ro"

### Actual behavior

Throws errors in log

### Expected behavior

Starts ebusd in Readonly mode

### ebusd version

current source from git

### ebusd arguments

see above

### Operating system

other

### CPU architecture

other

### Dockerized

latest

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

2024-02-29T06:10:18.438469411Z cannot combine readonly with answer/generatesyn/initsend/scanconfig
2024-02-29T06:10:18.438528480Z 2024-02-29 07:10:18.438 [main error] invalid arguments

john30, 2024-02-29:
how would ebusd know which files to load if it can't scan the bus participants??
you can run it readonly with local config files only, thats why it tells you:
"cannot combine readonly with answer/generatesyn/initsend/scanconfig"

Kolbi, 2024-03-01:
Okay currently I'm a little bit confused.

Because if I don't specifiy EBUSD_CONFIGPATH it should use default 
`#ifdef HAVE_SSL
34 | #define CONFIG_PATH "https" CONFIG_PATH_SUFFIX
35 | #else  // HAVE_SSL
36 | #define CONFIG_PATH "http" CONFIG_PATH_SUFFIX
37 | #endif  // HAVE_SSL`

john30, 2024-03-01:
yes, but the default is only available with --scanconfig (which is set in the Dockerfile) which required non-readonly