john30/ebusd #1219: MQTT readable fields not published if write of one field exists

pvyleta, 2024-03-06:
### Description

When the `.csv` configuration contains two lines - one for the write and one for a read of the same message name, and the read contains multiple fields while the write contains only one field, only the config of that one writable field is published to MQTT.

This is especially problematic for Brink HRU units, which have configuration stored in the format:

Reading:
Field1 = Current value
Field2 = Maximum
Field3 = Minimum
Field4 = Step size
Field5 = Default

Writing:
Field1 = New Current Value

An example of a `.csv` file of this format:

```
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,field2,part (m/s),datatypes/templates ,divider/values,unit,comment,field3,part (m/s),datatypes/templates,divider/values,unit,comment,field4,part (m/s),datatypes/templates,divider/values,unit,comment,field5,part (m/s),datatypes/templates,divider/values,unit,comment
w,test_device,AirFlowLevel0,,,3c,4080,21,,,SIR,,,
r1,test_device,AirFlowLevel0,,,3c,4050,21,,,SIR,,,,Minimum,,SIR,,,,Maximum,,SIR,,,,StepSize,,SIR,,,,FactorySetting,,SIR,,,
```

### Actual behavior

with the above `.csv` file, only the following configuration is published to MQTT:
```
2024-03-06 17:44:58.029 [mqtt debug] publish homeassistant/number/ebusd_test_device_2_AirFlowLevel0_CurrentValue/config {
  "unique_id":"ebusd_test_device_2_AirFlowLevel0_CurrentValue",
  "name":"AirFlowLevel0 CurrentValue",
  "device":{
   "identifiers":"ebusd_test_device_2",
   "manufacturer":"ebusd.eu",
   "name":"ebusd test_device_2",
   "via_device":"ebusd",
   "sw_version":"23.2",
   "suggested_area":"Heating"
  },
  "value_template":"{{value_json[\"curval\"].value}}",
  "state_topic":"ebusd/test_device_2/AirFlowLevel0",
  "command_topic":"ebusd/test_device_2/AirFlowLevel0/set",
  "min":-32767,
  "max":32767,
  "step":1
 }
```

### Expected behavior

on top of the above Write, Read of remaining fields should be published as well.

### ebusd version

23.2

### ebusd arguments

ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=Rae9ooYeekaigi9OoPhoodahCheel7aequaidiis1oocee8iw5ieG3maingietho --scanconfig --mqttjson --mqttlog --configpath=/config/ebusd-configuration/latest/en --mqttint=/config/ebusd/contrib/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.100.60:9999 --log=all:debug --log=bus:notice

### Operating system

other

### CPU architecture

arm64

### Dockerized

other

### Hardware interface

Adapter Shield v5 via USB

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

see above in 'Actual Behavior'

john30, 2024-03-24:
it does not make a lot of sense to have the same name for something that is just not comparable, e.g. when the number of fields are different. the idea of having read/write messages being considered as something belonging to each other requires of course having the same fields and even in the same sequence. so if that is not the case, then give those messages different names would be th eonly logical thing to do

closed 2024-03-24 by john30

pvyleta, 2024-03-24:
I understand the reasoning, and this may be true for the Vaillant devices, but the full portfolio of Brink HRU units (which is something I am trying to integrate at the moment), reading of the parameter has one PBSB, and gives not only the parameter value, but also the range of one values, step size and default value, and writing that value has a distinct PBSB and allows writing only the value itself (while presumably following the range and step size obtained from the read operation).

Giving different names to the messages is **not** logical in that case, because then the MQTT integration does not pair that one writable field with the readable field.

Please, reconsider this. Thanks!

john30, 2024-03-29:
why don't you just skip the extra fields in the read message as these should not be changeable anyway?

pvyleta, 2024-03-29:
Well that's what I do right now, and yes, there is only negligible impact on functionality, since I know most of the min/max/step/default anyway, but in case these would change in some future SW version of the device, this might be a nice way to confirm that.

What especially puzzled me though is, that publish of reading of those additional fields works, but only the initial message that registers the MQTT topic is omitted.

claytonn73, 2026-09-05:
I was looking at the same issue here and at least for my Ariston heatpump there are both 2000 and 2001 PBSB values to read values. The 2000 value only returns the current setting so can be paired with the write command using PBSB 2020 while the 2001 value returns the setting and minimum/maximum values. 

Here's some example CSV values - also need to add percent to the writable list in mqtt-hassio.cfg

w,heatpump,pump_min_pwm,HP pump min PWM,,1e,2020,d22c,setting,m,percent
r,heatpump,pump_min_pwm,HP pump min PWM,,1e,2000,d22c,,s,ignore_1,,,,setting,,percent
r9,heatpump,pump_min_pwm_2001,HP pump min PWM,,1e,2001,d22c,setting,,percent,,,,minimum,,percent,,,,maximum,,percent

This does mean that there is some duplication in home assistant entities but it does solve the issue of both having a writable value and getting the min and max.

Not expecting anything to be done here but leaving the comment for others to find.