john30/ebusd #1221: Datei für Vaillant 18.v32.csv

ralfnap2, 2024-03-09:
### Description

Die Datei 18.v32.csv wird geladen, jedoch finde ich diese datei nirgendwo, auch nicht im Netz.

address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 13: master #12
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903", loaded "vaillant/15.700.csv"
address 18: slave #12, scanned "MF=Vaillant;ID=V32;SW=0106;HW=6004", loaded "vaillant/18.v32.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0100;HW=5904", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"

### Actual behavior

Die Datei 18.v32.csv wird geladen, jedoch finde ich diese datei nirgendwo, auch nicht im Netz

### Expected behavior

Die Datei sollte irgendwo zu finden sein.

### ebusd version

current source from git

### ebusd arguments

xxx

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via WiFi

### Logs

address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 13: master #12
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903", loaded "vaillant/15.700.csv"
address 18: slave #12, scanned "MF=Vaillant;ID=V32;SW=0106;HW=6004", loaded "vaillant/18.v32.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0100;HW=5904", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"

JonesGW-MGD, 2024-03-09:
Are you sure your ebusd config files are coming from the repository?  Because the V32 is a bus coupler - and any device can sit behind it such as HRU or a boiler etc. etc. consequently I can't see there can ever be a repo v32 file!  

Can you share your --configpath? (from startup logs)

ralfnap2, 2024-03-10:
Ok, danke für die schnelle Antwort. Meine Installation ist neu und stammt aus dem Repository. Ich habe da lokal nichts verändert. Eigentlich geht es mir um Daten, die ich in den Logs und den CSV-Files nicht finde. Ich habe eine Vaillant-Hybrid-Anlage. Alle Komponenten - außer ECOVit Brennwertkessel - werden gefunden. In der CSV-Datei 15.700 fehlen aber Einträge für den Brennwertkessel, die im Steuermodul von Vaillant angezeigt werden. Z.B. Hybridmanager, Bivalenzpunkte, Kosten (Strom, Gas).

JonesGW-MGD, 2024-03-10:
Ok - that is a different, but related problem I think!

So this is what I think is happening - your Heat Pump is allocated slave address 08 and the autoscan therefore loads 08.hmu.csv to support the known that pump message definitions.

Normally in a single heat source scenario, the Ecovit would also identify on address 08, so the purpose of the bus coupler (Vaillant VR32) is to present the ecovit with a different address - in your case 18.  Of course neither the V32 or ebusd 'knows' what is behind the VR32.

To get Ecovit data, you can try the following - 

1. Clone the repository locally
2. Rename a boiler csv  to 32.v32.csv - you can look at all the bai.......csv files - I think one specifically mentioned ecovit - but I'd be tempted to start with bai.305832.inc which is effectively the 'default' Vaillant boiler message file.
3. Add --latency=100000 to your ebusd startup (do you run it stand-alone, in a container, or using the HA ebusd add-on?)

But that doesn't answer your original query as to why your installation loads 18.v32.csv - because that file doesn't (and can't for the reasons I explained above) exist in the official repository!

To satisfy my curiosity, can you show the setting you have for --configpath as your instance starts up?

Ok – das ist ein anderes, aber verwandtes Problem, denke ich!

Ich denke also, dass dies geschieht: Ihrer Wärmepumpe wird die Slave-Adresse 08 zugewiesen und der automatische Scan lädt daher 08.hmu.csv, um die bekannten Nachrichtendefinitionen dieser Pumpe zu unterstützen.

Normalerweise würde sich der Ecovit in einem Einzelwärmequellen-Szenario auch an der Adresse 08 identifizieren, daher besteht der Zweck des Buskopplers (Vaillant VR32) darin, dem Ecovit eine andere Adresse zuzuweisen – in Ihrem Fall 18. Natürlich weder der V32 noch der ebusd „weiß“, was sich hinter dem VR32 verbirgt.

Um Ecovit-Daten zu erhalten, können Sie Folgendes versuchen:

1. Klonen Sie das Repository lokal
2. Benennen Sie eine Boiler-CSV-Datei in 32.v32.csv um – Sie können sich alle bai.......csv-Dateien ansehen – ich glaube, in einer davon wurde speziell „ecovit“ erwähnt – aber ich wäre versucht, mit bai.305832 zu beginnen. inc, was praktisch die „Standard“-Meldungsdatei für den Vaillant-Kessel ist.
3. Fügen Sie --latency=100000 zu Ihrem ebusd-Startup hinzu (führen Sie es eigenständig, in einem Container oder mit dem HA-ebusd-Add-on aus?)

Aber das beantwortet nicht Ihre ursprüngliche Frage, warum Ihre Installation 18.v32.csv lädt – denn diese Datei existiert nicht (und kann aus den oben erläuterten Gründen auch nicht) im offiziellen Repository vorhanden sein!

Können Sie, um meine Neugier zu befriedigen, die Einstellung, die Sie für --configpath haben, beim Start Ihrer Instanz anzeigen?

ralfnap2, 2024-03-11:
Danke für die Antwort.
Ich habe in etc/default/ebusd nur folgenden Eintrag:
EBUSD_OPTS="--scanconfig -d ens:...:9999"

ralfnap2, 2024-03-11:
Ich könnte bai.0010007508.inc verwenden. "BAI00,ecoVIT pneumatic/4,0010007508 207,,,,"

ralfnap2, 2024-03-11:
Habe lokal geklont und Boiler-CSV-Datei in 18.v32.csv umgenannt. 
signal: acquired
symbol rate: 23
max symbol rate: 141
min arbitration micros: 2
max arbitration micros: 18
min symbol latency: 7
max symbol latency: 30
scan: finished
reconnects: 0
masters: 5
messages: 550
conditional: 0
poll: 0
update: 11
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 13: master #12
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903", loaded "vaillant/15.700.csv"
address 18: slave #12, scanned "MF=Vaillant;ID=V32;SW=0106;HW=6004", loaded "vaillant/18.v32.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0100;HW=5904", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"

# Aus 18.v32.csv (von Boiler-Dat) WPPostrunTime (Kann im Steuermodul 700 ausgelesen werden)
ebusctl read WPPostrunTime
ERR: element not found

ralfnap2, 2024-03-11:
Habe folgendes geändert:
EBUSD_OPTS="--scanconfig -d ens:...:9999 --latency=100000 --configpath=/home/ralfnap/ebusd-configuration/ebusd-2.1.x/de"

Jetzt funktioniert es.
ebusctl read WPPostrunTime
5

Bleibt noch die Frage nach den Werten für Gas- und Strompreis. Ansonsten schon mal vielen Dank für die Hilfe. Ein wirklich professioneller Service.

ralfnap2, 2024-03-13:
PrEnergyCountHc1/1000000 is m³ gas but for water it is PrEnergySumHwc1/1000000

john30, 2024-03-24:
not an ebusd issue, so closing this one. please use the appropriate repo [ebusd-configuration](https://github.com/john30/ebusd-configuration)

closed 2024-03-24 by john30