john30/ebusd #1226: Using --mqttchanges occasionally misses changes.

DavidR1953, 2024-03-14:
### Description

I am using ebusd to read and control a Vaillant ecoTEC plus System Boiler. There are no other controllers on the bus.
Messages are polled using r1 in the message definition.

When a value changes, it is occasionally not reported to MQTT, resulting in a wrong value for a time, until the value changes again, at which time it usually corrects.

ebusd.log shows that the poll is successful, and reads the value correctly.

Generally the system is pretty solid, with very few bus errors.
If the --mqttchanges option is removed from EBUSD_OPTS, then the value is reported to MQTT every time it is polled, as expected.
I have chosen to show the message RemainingBoilerblocktime here just as an example, but other values do this too. Clearly, for values that change often, it is not so much of a problem.

### Actual behavior

from ebusd.log:
	Line 5954: 2024-03-14 09:22:34.227 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 0
	Line 5966: 2024-03-14 09:23:22.203 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 0
	Line 5976: 2024-03-14 09:24:10.187 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 14
	Line 5986: 2024-03-14 09:24:58.209 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 13
	Line 5996: 2024-03-14 09:25:46.208 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 12
	Line 6004: 2024-03-14 09:26:34.230 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 12
	Line 6014: 2024-03-14 09:27:22.205 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 11
	Line 6026: 2024-03-14 09:28:10.207 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 10
	Line 6036: 2024-03-14 09:28:58.194 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 9
	Line 6047: 2024-03-14 09:29:52.201 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 9
	Line 6057: 2024-03-14 09:30:40.185 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 8
	Line 6065: 2024-03-14 09:31:28.221 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 7
	Line 6075: 2024-03-14 09:32:16.205 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 6
	Line 6088: 2024-03-14 09:33:04.221 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 5
	Line 6098: 2024-03-14 09:33:52.197 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 4
	Line 6106: 2024-03-14 09:34:40.240 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 3
	Line 6116: 2024-03-14 09:35:28.216 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 3
	Line 6124: 2024-03-14 09:36:16.190 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 2
	Line 6135: 2024-03-14 09:37:10.198 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 1
	Line 6145: 2024-03-14 09:37:58.198 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 0

and the values reported to MQTT are:
0, 14, 12, 11, 10, 8, 6, 5, 4, 3, 2, 1

### Expected behavior

When a value changes, it should be reported to MQTT.

### ebusd version

23.3

### ebusd arguments

EBUSD_OPTS="-d enh:192.168.10.140:9999 --scanconfig  --configpath=/etc/ebusd/ --mqttport=1883 --mqttjson --mqttchanges --accesslevel=*"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter v3 WiFi

### Related integration

[... 199 lines cut ...]
2024-03-14 09:37:43.369 [update notice] sent write bai SetModeOverride QQ=31: 0;40.5;-;-;-;0;0;0;-;0;0;0
2024-03-14 09:37:43.369 [mqtt notice] write bai SetModeOverride: 0;40.5;-;-;-;0;0;0;-;0;0;0
2024-03-14 09:37:46.177 [update notice] sent poll-read bai ACRoomthermostat QQ=31: on
2024-03-14 09:37:52.230 [update notice] sent poll-read bai WP QQ=31: on
2024-03-14 09:37:58.198 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 0
2024-03-14 09:38:01.097 [update notice] sent write bai SetModeOverride QQ=31: 0;40.5;-;-;-;0;0;0;-;0;0;0
2024-03-14 09:38:01.097 [mqtt notice] write bai SetModeOverride: 0;40.5;-;-;-;0;0;0;-;0;0;0
2024-03-14 09:38:04.242 [update notice] sent poll-read bai Flame QQ=31: off
2024-03-14 09:38:10.204 [update notice] sent poll-read bai Modulation QQ=31: 19.0
2024-03-14 09:38:16.210 [update notice] sent poll-read bai Statenumber QQ=31: 0
2024-03-14 09:38:22.240 [update notice] sent poll-read bai Status01 QQ=31: 36.5;36.5;-;-;-;off
2024-03-14 09:38:28.213 [update notice] sent poll-read bai FlowTempDesired QQ=31: 40.50
2024-03-14 09:38:34.243 [update notice] sent poll-read bai ACRoomthermostat QQ=31: on
2024-03-14 09:38:40.221 [update notice] sent poll-read bai WP QQ=31: on
2024-03-14 09:38:43.346 [update notice] sent write bai SetModeOverride QQ=31: 0;40.5;-;-;-;0;0;0;-;0;0;0
2024-03-14 09:38:43.347 [mqtt notice] write bai SetModeOverride: 0;40.4;-;-;-;0;0;0;-;0;0;0
2024-03-14 09:38:46.206 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 0
2024-03-14 09:38:52.218 [update notice] sent poll-read bai Flame QQ=31: off
2024-03-14 09:38:58.196 [update notice] sent poll-read bai Modulation QQ=31: 0.0
2024-03-14 09:39:04.250 [update notice] sent poll-read bai Statenumber QQ=31: 2
2024-03-14 09:39:10.195 [update notice] sent poll-read bai Status01 QQ=31: 35.0;35.5;-;-;-;off
2024-03-14 09:39:16.209 [update notice] sent poll-read bai FlowTempDesired QQ=31: 40.50
2024-03-14 09:39:22.268 [update notice] sent poll-read bai ACRoomthermostat QQ=31: on
2024-03-14 09:39:28.199 [update notice] sent poll-read bai WP QQ=31: on
2024-03-14 09:39:34.210 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 0

DavidR1953, 2024-03-24:
I cannot really be sure that the mqtt broker (mosquitto version 2.0.11) doesn't loose messages.
ebusd and the broker are on the same machine (RPi4 with 4G memory). It's running openHAB and is fairly lightly loaded, normally 2% ish.
With MQTT logging set to debug, I can estimate there are about 30 publish messages per minute, so not a lot. Of course there could be peaks.
I've also set ebusd log 'other' to debug, so I'll leave it running and see if I can catch when the problem occurs.

DavidR1953, 2024-03-24:
Here is a capture of the problem:
This is filtered to just show it more easily. I've attached the whole log.
[ebusd.txt](https://github.com/john30/ebusd/files/14735869/ebusd.txt)

At time 13:38:08.197, the value changes from 14 to 13, but not published.
At time 13:43:02.193, the value changes from 9 to 8, but not published.
At time 13:44:38.183, the value changes from 7 to 6, but not published.
At time 13:51:08.202, the value changes from 1 to 0, but not published.

	Line 16626: 2024-03-24 13:31:38.219 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 0
	Line 16640: 2024-03-24 13:32:26.190 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 0
	Line 16698: 2024-03-24 13:33:14.196 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 0
	Line 16742: 2024-03-24 13:34:02.194 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 0
	Line 16821: 2024-03-24 13:34:56.244 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 16
	Line 16822: 2024-03-24 13:34:58.496 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 16824: 2024-03-24 13:34:58.497 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 16891: 2024-03-24 13:35:44.208 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 15
	Line 16892: 2024-03-24 13:35:48.544 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 16894: 2024-03-24 13:35:48.544 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 16909: 2024-03-24 13:36:32.245 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 15
	Line 16953: 2024-03-24 13:37:20.247 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 14
	Line 16954: 2024-03-24 13:37:23.499 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 16956: 2024-03-24 13:37:23.499 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17001: 2024-03-24 13:38:08.197 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 13
	Line 17059: 2024-03-24 13:38:56.231 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 12
	Line 17060: 2024-03-24 13:38:58.549 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17062: 2024-03-24 13:38:58.549 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17107: 2024-03-24 13:39:44.272 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 11
	Line 17108: 2024-03-24 13:39:48.481 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17110: 2024-03-24 13:39:48.481 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17129: 2024-03-24 13:40:32.211 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 10
	Line 17187: 2024-03-24 13:40:38.466 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17189: 2024-03-24 13:40:38.466 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17207: 2024-03-24 13:41:20.202 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 10
	Line 17258: 2024-03-24 13:42:14.188 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 9
	Line 17259: 2024-03-24 13:42:18.560 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17261: 2024-03-24 13:42:18.561 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17306: 2024-03-24 13:43:02.193 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 8
	Line 17350: 2024-03-24 13:43:50.215 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 7
	Line 17351: 2024-03-24 13:43:53.465 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17353: 2024-03-24 13:43:53.465 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17398: 2024-03-24 13:44:38.183 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 6
	Line 17426: 2024-03-24 13:45:26.204 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 6
	Line 17470: 2024-03-24 13:46:14.197 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 5
	Line 17471: 2024-03-24 13:46:18.564 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17473: 2024-03-24 13:46:18.564 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17548: 2024-03-24 13:47:02.186 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 4
	Line 17552: 2024-03-24 13:47:08.511 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17554: 2024-03-24 13:47:08.511 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17566: 2024-03-24 13:47:50.199 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 3
	Line 17567: 2024-03-24 13:47:53.560 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17569: 2024-03-24 13:47:53.560 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17644: 2024-03-24 13:48:38.234 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 2
	Line 17661: 2024-03-24 13:49:32.210 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 1
	Line 17693: 2024-03-24 13:49:38.535 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17695: 2024-03-24 13:49:38.535 [mqtt debug] publish ebusd/bai/RemainingBoilerblocktime {
	Line 17710: 2024-03-24 13:50:20.222 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 1
	Line 17768: 2024-03-24 13:51:08.202 [update notice] sent poll-read bai RemainingBoilerblocktime QQ=31: 0
[... cut ...]
r9,,WaterPressure,Water pressure,,,,"0200",,,presssensor,,,CH water pressure,,,
r1,,Flame,Flame,,,,"0500",,,UCH,240=off;15=on,,Flame,,,
r1,,Modulation,Modulation_percent,,,,"2E00",,,SIN,10,%,Modulation as a percentage of boiler maximum,,,
r1,,Statenumber,Statenumber,,,,"AB00",,,UCH,,,Status number,,,
r1,,Status01,FlowTemperature/ReturnRemperature/OutsideTemperature/DHWtemperature/StorageTankTemperature/PumpStatus,,,,01,,,temp1;temp1;temp2;temp1;temp1;pumpstate,,,

```

If can help with testing, please let me know.

john30, 2024-03-29:
thanks for the detailed logging! looking at the handling procedure there might be a chance that the mqtt messages are sent while the message decoding is running and so these could be missed.

john30, 2024-03-29:
could you please check with the recent commit @DavidR1953 ?

DavidR1953, 2024-03-29:
Thanks John. I'm away till end of next week. Will check then. Sorry it cannot be sooner.

john30, 2024-03-29:
no worries, thanks for checking then David!

DavidR1953, 2024-04-06:
Here is a log from the most recent commit:
[ebusd.txt](https://github.com/john30/ebusd/files/14893124/ebusd.txt)

The log starts at 2024-04-05 14:35:53.868 but I didn't set other:debug until 2024-04-05 14:48:25.885

The missed changes have definitely improved, but there is still the occasional error.

At 2024-04-05 15:04:37.244, Statenumber changes from 8 to 0, but is not published.
At 2024-04-05 16:10:07.240, Flame changes from off to on, but is not published.
At 2024-04-05 16:24:37.245, RemainingBoilerblocktime changes from 0 to 19, but is not published.
At 2024-04-05 17:27:13.196, Modulation changes from 83.3 to 81.5, but is not published.
At 2024-04-05 19:41:13.220, Statenumber changes from 7 to 8, but is not published.
At 2024-04-05 19:41:43.201, RemainingBoilerblocktime changes from 12 to 11, but is not published.
At 2024-04-05 19:45:43.218, RemainingBoilerblocktime changes from 8 to 7, but is not published.
At 2024-04-05 19:48:13.256, RemainingBoilerblocktime changes from 6 to 5, but is not published.
At 2024-04-05 20:00:13.210, WP changes from off to on, but is not published.
At 2024-04-06 07:25:07.248, Modulation changes from 21.6 to 23.9, but is not published.

There are more, but hopefully that's enough for now.

If I can do more testing, let me know.

john30, 2024-04-10:
thanks for the input @DavidR1953 !
can you please check again with the recent commit?

DavidR1953, 2024-04-10:
The recent commit is looking good.
It's been running for about 3 hours and hasn't missed any changes.
I'll keep checking.
[ebusd.txt](https://github.com/john30/ebusd/files/14934539/ebusd.txt)

Question unrelated to the problem:
During the update check I get
2024-04-10 13:17:57.343 [main notice] update check: revision 23.3 available, device firmware 1[8419] available, ....

Is this because I'm using eBUS adapter 3 which is on Build: 20221215, or something else?
I couldn't find that reference (8419) anywhere.
btw this is not related to the recent commit, I had it before.

john30, 2024-04-11:
great, thanks for checking.

yes, thats because you're not on the newest, see here https://adapter.ebusd.eu/v31/firmware/ChangeLog.html

DavidR1953, 2024-04-11:
It's been running now for about 24 hours with no missed changes. Success. Thank you.

[ebusd.txt](https://github.com/john30/ebusd/files/14946047/ebusd.txt)

As for the device firmware - I confused the ebusd-esp and picloader versions - it's clear now.

john30, 2024-04-27:
ok great

closed 2024-04-27 by john30