john30/ebusd #1238: Conditional message ignored if slave-address unfilled

pvyleta, 2024-03-25:
### Description

If conditional message does not have slave-address filled in. It is silently ignored. I am able to reproduce this with following minimal file `3c.csv`

```
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment
*[fanmodetwo],TEST_DEVICE,FanMode,,,,2
r,TEST_DEVICE,FanMode,FanMode,,,4022,01,,,UIR,0=Holiday;1=Reduced;2=Normal;3=High;4=Auto,,
[fanmodetwo]r,TEST_DEVICE,InletFlow,ActualInletFlow,,3c,4022,0b,,,UIR,,,
```
If the message `InletFlow` does not have the address `3c` filled in, it is completely skipped. If I fill it in, it is correctly polled for and published to MQTT.

### Actual behavior

Conditional message ignored.

### Expected behavior

Conditional message used and polled for.

### ebusd version

23.2

### ebusd arguments

ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=Rae9ooYeekaigi9OoPhoodahCheel7aequaidiis1oocee8iw5ieG3maingietho --scanconfig --mqttjson --mqttlog --configpath=/config/ebusd-configuration/latest/en --mqttint=/config/ebusd/contrib/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.100.60:9999 --log=all:debug --log=bus:notice

### Operating system

other

### CPU architecture

arm64

### Dockerized

other

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

Logs when `3c` not part of message definition:

```2024-03-25 22:25:41.537 [main debug] performing regular tasks
2024-03-25 22:25:41.805 [update info] sent MS cmd: 313c070400 / 0a400812002203ffffffff
2024-03-25 22:25:41.807 [bus notice] scan 3c: ;ENCON;  ;-;-
2024-03-25 22:25:41.807 [update notice] store 3c ident: done
2024-03-25 22:25:41.807 [update notice] sent scan-read scan.3c  QQ=31: ENCON;  ;-;-
2024-03-25 22:25:41.807 [bus notice] scan 3c: ;ENCON;  ;-;-
2024-03-25 22:25:41.807 [main debug] reading directory /config/ebusd-configuration/latest/en/encon/
2024-03-25 22:25:41.809 [main debug] file type of /config/ebusd-configuration/latest/en/encon/3c.csv is file
2024-03-25 22:25:41.809 [main debug] file type of /config/ebusd-configuration/latest/en/encon/scan.csv is file
2024-03-25 22:25:41.809 [main debug] file type of /config/ebusd-configuration/latest/en/encon/Sky300.inc is file
2024-03-25 22:25:41.809 [main debug] found 1 matching scan config files from encon with prefix 3c: done
2024-03-25 22:25:41.809 [main debug] reading directory /config/ebusd-configuration/latest/en/encon/
2024-03-25 22:25:41.809 [main debug] file type of /config/ebusd-configuration/latest/en/encon/3c.csv is file
2024-03-25 22:25:41.809 [main debug] file type of /config/ebusd-configuration/latest/en/encon/scan.csv is file
2024-03-25 22:25:41.809 [main debug] file type of /config/ebusd-configuration/latest/en/encon/Sky300.inc is file
2024-03-25 22:25:41.836 [main notice] read common config file encon/scan.csv
2024-03-25 22:25:41.842 [main notice] read scan config file encon/3c.csv for ID "", SW65535, HW65535
2024-03-25 22:25:41.843 [main notice] found messages: 16 (1 conditional on 1 conditions, 1 poll, 4 update)
[... 305 lines cut ...]
  "unique_id":"ebusd_TEST_DEVICE_3c_FanMode_0",
  "name":"FanMode ",
  "device":{
   "identifiers":"ebusd_TEST_DEVICE_3c",
   "manufacturer":"ebusd.eu",
   "name":"ebusd TEST_DEVICE.3c",
   "via_device":"ebusd",
   "sw_version":"23.2",
   "suggested_area":"Heating"
  },
  "value_template":"{{value_json[\"0\"].value}}",
  "state_topic":"ebusd/TEST_DEVICE.3c/FanMode",
  "device_class":"enum",
  "options":["Holiday","Reduced","Normal","High","Auto"]
 }
2024-03-25 22:29:44.385 [mqtt debug] publish ebusd/global/signal true
2024-03-25 22:29:44.386 [mqtt debug] publish ebusd/TEST_DEVICE.3c/FanMode {
     "0": {"name": "", "value": "Normal"}}
2024-03-25 22:29:45.576 [update info] received MS cmd: 1f3c40cb0401010002 / 020002
2024-03-25 22:29:45.577 [update notice] received unknown MS cmd: 1f3c40cb0401010002 / 020002
2024-03-25 22:29:45.707 [update info] received MS cmd: 1f3c40ca0101 / 06000200552000
2024-03-25 22:29:45.707 [update notice] received unknown MS cmd: 1f3c40ca0101 / 06000200552000
2024-03-25 22:29:46.296 [update info] sent MS cmd: 313c4022010b / 020051
2024-03-25 22:29:46.296 [update notice] sent poll-read TEST_DEVICE InletFlow QQ=31: 81
```

john30, 2024-03-29:
defaults from the filename are only taken into account when there is an appropriate defaults assignment row:
https://github.com/john30/ebusd/wiki/4.2.-Defaults#defaults-extracted-from-the-file-name

closed 2024-03-29 by john30

john30, 2024-04-27:
I've added it