john30/ebusd #1254: Connect to two network adapters using Home Assistant OS and addon

piotrtobolski, 2024-04-09:
### Description

I have 2 Vaillant devices: ecoCOMPACT and recoVAIR. Both are connected to VRC 700 and recoVAIR requires the VR 32 which makes it impossible to use single adapter.

I found in another issue that the solution is to buy another adapter (which I did) and connect it to the second device bus directly.

How can I now launch another daemon on my HA instance (using HA OS) if the addon can be installed only once and allows connecting to just a single ebusd instance?

john30, 2024-04-10:
obiously thats a question to the author of the addon, not me

closed 2024-04-10 by john30

piotrtobolski, 2024-04-10:
I thought that maybe the daemon could allow connecting to more than one adapter or maybe it does but I don't know how?

john30, 2024-04-11:
a single ebusd instance handles a single bus