john30/ebusd #1256: eBUS Adapter v5, Firmware Version 2024-03-30 Kommunikationsprobleme

Norb1204, 2024-04-13:
### Description

Hallo,

mit der aktuellen Firmwareversion 2024-03-30 für den Adapter v5 wird der Scanning Prozess nicht beendet. Die Vorgänger-Firmwareversion 2024-01-06 funktioniert und er erkennt alle Teilnehmer. Ein Neustart vom ebusd service bringt keinen Erfolg.

Firmwareversion 2024-03-30 "1[431e]", funktioniert nicht
[image]

Firmwareversion 2024-01-06 "1[4106]", funktioniert
[image]

### Actual behavior

siehe Description

### Expected behavior

siehe Description

### ebusd version

23.3

### ebusd arguments

EBUSD_OPTS="--scanconfig=full --configpath=/etc/ebusd/ -d ens:/dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_DC:54:75:63:64:C4-if00 -p 8888 -l /var/log/ebusd.log --loglevel=error --logareas=network  --httpport=8889 --htmlpath=/var/ebusd/html --accesslevel=install"

### Operating system

Debian 10 (Buster) / Ubuntu 18-19 / Raspbian 10 / Raspberry Pi OS 10 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via USB

### Logs

   ebusd.log = 0

phillipc, 2024-04-14:
I can't confirm this issue in my environment: **Docker / Adapter Shield v5 via WIFI**

> root@ab92878b10fb:/# ebusctl info
> version: ebusd **23.3.23.3**
> update check: OK
> device: 192.168.0.103:9999, TCP, enhanced, **firmware 1.1[431e].1[431e]**
> access: *
> signal: acquired
> symbol rate: 23
> max symbol rate: 117
> min arbitration micros: 1
> max arbitration micros: 4
> min symbol latency: 5
> max symbol latency: 17
> scan: finished
> reconnects: 0
> masters: 5
> messages: 593
> conditional: 4
> poll: 182
> update: 10
> address 03: master #11
> address 08: slave #11, **scanned "MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "vaillant/bai.0010006101.inc" ([PROD=''])**, "vaillant/08.bai.csv"
> address 10: master #2
> address 15: slave #2, **scanned "MF=Vaillant;ID=47000;SW=0420;HW=1403", loaded "vaillant/15.470.csv"**
> address 31: master #8, ebusd
> address 36: slave #8, ebusd
> address 3f: master #23
> address 50: slave, **scanned "MF=Vaillant;ID=V6100;SW=0418;HW=1902", loaded "vaillant/50.v61.mc.csv"**
> address 7f: master

overlinden, 2024-04-17:
I can confirm that there is something wrong with the most recent firmware version (20240415). The log displays warnings `host: errors: protocol` as well:
[image]

I downgraded back to 20240106.

john30, 2024-04-20:
not an ebusd issue in the first place, so closing this one. please use [ebusd-esp32](https://github.com/john30/ebusd-esp32) instead for such.
anyway, you might want to try the [newest version](https://adapter.ebusd.eu/v5/ChangeLog#20240420) just released

closed 2024-04-20 by john30