john30/ebusd #1257: HomeAssistant integration - ebusd - measurement 'None' which is not a valid unit for the device class ('temperature')

botihu, 2024-04-14:
### Description

During Home Assistant startup warnings are logged regarding to a few variables have 'None' native measurement units.

### Actual behavior

Home assistant log is noisy and the variables with complain are not available in HA

### Expected behavior

Ebusd and mqtt integration should take care of measurement units of their variables. 

### ebusd version

23.3

### ebusd arguments

--scanconfig

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

same as ebusd version

### Hardware interface

Adapter v3 WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

2024-04-14 11:16:56.886 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_deactivationstemplimiter (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['°C', '°F', 'K']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:19:17.064 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_maintenancedata_prenergysumhwc1 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'MWh', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:19:21.109 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_modulationtempdesired (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('temperature') it is using; expected one of ['°C', '°F', 'K']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:19:51.363 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_prenergycounthc1 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'MWh', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:19:56.409 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_prenergycounthc2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'MWh', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:20:06.516 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_prenergycounthc3 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'MWh', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:20:11.563 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_prenergycounthwc1 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'MWh', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:20:16.611 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_prenergycounthwc2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'MWh', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:20:21.659 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_prenergycounthwc3 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'MWh', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:20:26.712 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_prenergysumhc2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'MWh', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:20:36.764 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_prenergysumhc3 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'MWh', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:20:41.819 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_prenergysumhwc1 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'MWh', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:20:46.870 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_prenergysumhwc3 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'GJ', 'MWh', 'kWh', 'Wh']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2024-04-14 11:21:07.034 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_returntemp_tempmirror (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['°C', '°F', 'K']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

botihu, 2024-04-14:
Here are the environment variables from my kubernetes pod rinnung the ebusd container image
    Environment:
      EBUSD_SCANCONFIG:  full
      EBUSD_DEVICE:      enh:192.168.x.x:9999
      EBUSD_MQTTPORT:    1883
      EBUSD_MQTTHOST:    192.168.x.x
      EBUSD_MQTTUSER:    *redacted*
      EBUSD_MQTTPASS:    *redacted*
      EBUSD_MQTTINT:     /etc/ebusd/mqtt-hassio.cfg
      EBUSD_MQTTJSON:

botihu, 2024-04-14:
Somone reported the same error to home assistant. it was closed as it is not part of HA neither mqtt integration. 
https://github.com/home-assistant/core/issues/112642

Koky05, 2024-04-15:
@botihu they have right because it is maintained by ebusd file `mqtt-hassio.cfg` especially by part of type_map. There is defined domain and class, units come from .csv files decoded mostly by _templates.csv

john30, 2024-10-01:
this is a matter of correcting the definition of the message, especially the unit, e.g. in
[bai.0010007508_inc.tsp:463](https://github.com/john30/ebusd-configuration/blob/e257ec9c39b95550ca0415fa6f24e625a8c6ac71/src/vaillant/bai.0010007508_inc.tsp#L463) or [bai.0010007508.inc:96](https://github.com/john30/ebusd-configuration/blob/e257ec9c39b95550ca0415fa6f24e625a8c6ac71/ebusd-2.1.x/en/vaillant/bai.0010007508.inc#L96) and friends

SiFi2010, 2025-06-22:
I found a similar issue in my log file:

> 2025-06-22 11:36:50.004 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.fm5_mc1flowtempdesired_mixermovement (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement '%' which is not a valid unit for the device class ('temperature') it is using; expected one of ['K', '°C', '°F']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-06-22 11:38:40.652 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_ctlv2_z1timer_friday0_slottemp (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['K', '°C', '°F']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-06-22 11:38:55.712 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_ctlv2_z1timer_monday0_slottemp (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['K', '°C', '°F']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-06-22 11:39:14.901 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_ctlv2_z1timer_saturday0_slottemp (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['K', '°C', '°F']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-06-22 11:39:34.990 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_ctlv2_z1timer_sunday0_slottemp (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['K', '°C', '°F']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-06-22 11:39:50.186 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_ctlv2_z1timer_thursday0_slottemp (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['K', '°C', '°F']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-06-22 11:40:15.225 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_ctlv2_z1timer_tuesday0_slottemp (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['K', '°C', '°F']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
2025-06-22 11:40:36.276 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_ctlv2_z1timer_wednesday0_slottemp (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['K', '°C', '°F']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

kabadisha, 2026-05-05:
I'll chip in and confirm that this is still an issue. I will try and make some time to educate myself enough on the codebase to raise a PR, but with a full-time job and two kids, I think I will struggle, so if someone already familiar can easily add the correct units to the fields, that would be much appreciated.

`2026-05-05 07:39:51.776 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.ebusd_bai_deactivationstemplimiter (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['°F', 'K', '°C']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22`