john30/ebusd #1262: All communication stopped

ajvdw, 2024-04-24:
### Description

Running the v5 of the adapter wit latest firmware 20-4-2024 and 23.3 of ebusd my heatpump (Arotherm) stopped working.

### Actual behavior

All communication on the ebus was blocked (adapter?)

### Expected behavior

Ocassionally I get some bus_errors in the log (SYN received, read timeout) .

### ebusd version

23.3

### ebusd arguments

ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=usr --mqttpass=*** --scanconfig --mqttjson --configpath=/config/ebusconf --latency=200 --pollinterval=5 --mqttint=/config/ebusconf/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.1.127:9999 --log=all:notice --log=main:notice --log=bus:error --log=update:error --log=network:error --log=other:error

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

arm64

### Dockerized

devel

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

2024-04-24 07:01:35.057 [bus error] poll vwzio DCFSignalState failed: ERR: SYN received
2024-04-24 07:01:41.066 [bus error] poll vwzio FlowPressure failed: ERR: SYN received
2024-04-24 07:01:47.110 [bus error] poll vwzio HwcTemp failed: ERR: SYN received
2024-04-24 07:01:53.060 [bus error] poll vwzio ImmersionHeaterHours failed: ERR: SYN received
2024-04-24 07:01:59.057 [bus error] poll vwzio ImmersionHeaterPower failed: ERR: SYN received
2024-04-24 07:02:05.056 [bus error] poll vwzio ImmersionHeaterPowerLimit failed: ERR: SYN received
2024-04-24 07:02:11.048 [bus error] poll vwzio ImmersionHeaterStarts failed: ERR: SYN received
2024-04-24 07:02:17.091 [bus error] poll vwzio ImmersionHeaterTemp failed: ERR: SYN received
2024-04-24 07:02:23.065 [bus error] poll vwzio OutdoorTemp failed: ERR: SYN received
2024-04-24 07:02:29.078 [bus error] poll vwzio ReturnTemp failed: ERR: SYN received
2024-04-24 07:02:35.084 [bus error] poll vwzio RunDataBuildingCircuitPumpPower failed: ERR: SYN received
2024-04-24 07:02:41.088 [bus error] poll vwzio RunDataCondensorOutletTemp failed: ERR: SYN received
2024-04-24 07:02:47.057 [bus error] poll vwzio RunDataReturnFlowTemp failed: ERR: SYN received
2024-04-24 07:02:53.067 [bus error] poll vwzio RunDataSupplyFlowTemp failed: ERR: SYN received
2024-04-24 07:02:59.076 [bus error] poll vwzio RunDataSystemPressure failed: ERR: SYN received
2024-04-24 07:03:05.076 [bus error] poll vwzio RunStatsBuildingCircuitPumpHours failed: ERR: SYN received
2024-04-24 07:03:17.413 [bus error] poll vwzio RunStatsHcHours failed: ERR: SYN received
2024-04-24 07:03:28.460 [bus error] poll vwzio RunStatsImmersionHeaterHours failed: ERR: SYN received
2024-04-24 07:03:45.166 [bus error] poll vwzio RunStatsImmersionHeaterStarts failed: ERR: SYN received
2024-04-24 07:03:45.213 [bus error] poll vwzio RunStatsVWZHours failed: ERR: SYN received
2024-04-24 07:03:51.058 [bus error] poll vwzio State00 failed: ERR: SYN received
2024-04-24 07:04:03.082 [bus error] poll vwzio State07 failed: ERR: SYN received
2024-04-24 07:04:09.068 [bus error] poll vwzio Statuscode failed: ERR: SYN received
2024-04-24 07:04:15.063 [bus error] poll vwzio SupplyTemp failed: ERR: SYN received
2024-04-24 07:04:21.068 [bus error] poll vwzio SupplyTempImmersionHeater failed: ERR: SYN received
2024-04-24 07:04:27.103 [bus error] poll vwzio TemperatureSwitchElectricHeater failed: ERR: SYN received
2024-04-24 07:04:33.049 [bus error] poll vwzio TotalEnergyUsageImmersionHeater failed: ERR: SYN received
2024-04-24 07:04:39.085 [bus error] poll ctlv2 AdaptHeatCurve failed: ERR: SYN received
2024-04-24 07:04:45.048 [bus error] poll ctlv2 CoolingOutsideTempThreshold failed: ERR: SYN received
2024-04-24 07:04:51.081 [bus error] poll ctlv2 DisplayedOutsideTemp failed: ERR: SYN received
2024-04-24 07:04:57.084 [bus error] poll ctlv2 Hc1ActualFlowTempDesired failed: ERR: SYN received
2024-04-24 07:05:03.078 [bus error] poll ctlv2 Hc1DesiredSetbackTemp failed: ERR: SYN received
2024-04-24 07:05:09.083 [bus error] poll ctlv2 Hc1DesiredTemp failed: ERR: SYN received
2024-04-24 07:05:15.073 [bus error] poll ctlv2 Hc1ExcessTemp failed: ERR: SYN received
2024-04-24 07:05:21.071 [bus error] poll ctlv2 Hc1FlowTemp failed: ERR: SYN received
2024-04-24 07:05:27.080 [bus error] poll ctlv2 Hc1HeatCurve failed: ERR: SYN received
2024-04-24 07:05:33.085 [bus error] poll ctlv2 Hc1HeatCurveAdaption failed: ERR: SYN received
2024-04-24 07:05:39.077 [bus error] poll ctlv2 Hc1MaxHeatingFlowTempDesired failed: ERR: SYN received
2024-04-24 07:05:45.076 [bus error] poll ctlv2 Hc1MinCoolingFlowTempDesired failed: ERR: SYN received
2024-04-24 07:07:45.935 [bus error] poll ctlv2 Hc1MinHeatingFlowTempDesired failed: ERR: SYN received
2024-04-24 07:07:45.981 [bus error] poll ctlv2 Hc1PumpStatus failed: ERR: SYN received
2024-04-24 07:07:51.080 [bus error] poll ctlv2 Hc1RoomTempMod failed: ERR: SYN received
2024-04-24 07:08:43.650 [bus error] poll ctlv2 Hc1SetBackMode failed: ERR: SYN received
2024-04-24 07:08:46.602 [bus error] poll ctlv2 Hc1Status failed: ERR: SYN received
2024-04-24 07:08:49.064 [bus error] poll ctlv2 Hc1SummerTempLimit failed: ERR: SYN received
2024-04-24 07:08:55.085 [bus error] poll ctlv2 Hc3ActualFlowTempDesired failed: ERR: SYN received
2024-04-24 07:09:25.817 [bus error] poll ctlv2 Hc3DesiredSetbackTemp failed: ERR: SYN received
2024-04-24 07:09:46.597 [bus error] poll ctlv2 Hc3DesiredTemp failed: ERR: SYN received
2024-04-24 07:09:46.642 [bus error] poll ctlv2 Hc3ExcessTemp failed: ERR: SYN received
2024-04-24 07:09:52.049 [bus error] poll ctlv2 Hc3FlowTemp failed: ERR: SYN received
2024-04-24 07:09:58.086 [bus error] poll ctlv2 Hc3HeatCurve failed: ERR: SYN received

john30, 2024-04-27:
if the firmware version is the only thing you changed, then this is not an ebusd issue, so use the appropriate repo for that please: https://github.com/john30/ebusd-esp32
even more so if did work with an earlier firmware version. did it?
either way, some logging from the adapter would be needed in order to see whats going on

ajvdw, 2024-04-28:
> if the firmware version is the only thing you changed, then this is not an ebusd issue, so use the appropriate repo for that please: https://github.com/john30/ebusd-esp32 even more so if did work [...]

I actually changed the ebusd version to 23.3 as well.

john30, 2024-05-01:
you might as well just have several instances of ebusd running

john30, 2025-04-15:
closed for inactivity

closed 2025-04-15 by john30