john30/ebusd #1266: Guidance on debug level messages

roelof4, 2024-04-29:
### Description

Hardware: Raspberry Pi 3, Debian Bullseye 64-bit.
Command options: ebusd -f --scanconfig=full --configpath=/home/pi/ebusd-configuration/ebusd-2.1.x/en -d ens:/dev/ttyAMA0 --latency=60 --loglevel=debug
Connected to a Vaillant Arotherm Plus heatpump with regular controller and room thermostat/control. Using python script I can seen eBus messages on the bus that make sense in terms of what I find documented.

### Actual behavior

024-04-29 20:08:38.678 [bus notice] device status: transport opened
2024-04-29 20:08:38.679 [main info] registering data handlers
2024-04-29 20:08:38.679 [main info] registered data handlers
2024-04-29 20:08:38.679 [main notice] ebusd 23.3.23.3 started with full scan on device: /dev/ttyAMA0, serial high speed, enhanced
2024-04-29 20:08:38.680 [main info] loading configuration files from /home/pi/ebusd-configuration/ebusd-2.1.x/en/
2024-04-29 20:08:38.680 [main debug] reading directory /home/pi/ebusd-configuration/ebusd-2.1.x/en/
2024-04-29 20:08:38.692 [main debug] file type of /home/pi/ebusd-configuration/ebusd-2.1.x/en/broadcast.csv is file
2024-04-29 20:08:38.692 [main debug] file type of /home/pi/ebusd-configuration/ebusd-2.1.x/en/_templates.csv is file
2024-04-29 20:08:38.692 [main debug] file type of /home/pi/ebusd-configuration/ebusd-2.1.x/en/memory.csv is file
2024-04-29 20:08:38.692 [main debug] file type of /home/pi/ebusd-configuration/ebusd-2.1.x/en/vaillant is dir
2024-04-29 20:08:38.693 [main info] reading templates /
2024-04-29 20:08:38.696 [main info] read templates in /
2024-04-29 20:08:38.696 [main info] reading file broadcast.csv
2024-04-29 20:08:38.699 [main info] successfully read file broadcast.csv
2024-04-29 20:08:38.699 [main info] reading file memory.csv
2024-04-29 20:08:38.701 [main info] successfully read file memory.csv
2024-04-29 20:08:38.701 [main info] read config files, got 11 messages
2024-04-29 20:08:38.702 [bus notice] bus started with own address 31/36
2024-04-29 20:08:38.702 [bus notice] signal acquired
2024-04-29 20:08:43.014 [bus notice] max. symbols per second: 178
2024-04-29 20:08:44.039 [bus notice] max. symbols per second: 216
2024-04-29 20:08:48.703 [main debug] performing regular tasks
2024-04-29 20:08:48.703 [main notice] starting initial full scan
2024-04-29 20:08:48.703 [bus info] scan 02 cmd: 3102070400
2024-04-29 20:08:54.021 [bus notice] max. symbols per second: 267
2024-04-29 20:08:58.703 [main debug] performing regular tasks
2024-04-29 20:09:04.045 [bus notice] max. symbols per second: 272
2024-04-29 20:09:08.704 [main debug] performing regular tasks
2024-04-29 20:09:18.704 [main debug] performing regular tasks
2024-04-29 20:09:28.705 [main debug] performing regular tasks
2024-04-29 20:09:38.705 [main debug] performing regular tasks
2024-04-29 20:09:48.706 [main debug] performing regular tasks
2024-04-29 20:09:58.706 [main debug] performing regular tasks
2024-04-29 20:09:58.706 [main notice] scan completed 1 time(s), check again
2024-04-29 20:10:08.706 [main debug] performing regular tasks
2024-04-29 20:10:18.707 [main debug] performing regular tasks
2024-04-29 20:10:24.028 [bus notice] max. symbols per second: 273
2024-04-29 20:10:28.708 [main debug] performing regular tasks
2024-04-29 20:10:38.708 [main debug] performing regular tasks
2024-04-29 20:10:48.709 [main debug] performing regular tasks
2024-04-29 20:10:48.716 [network debug] SSL state 0=before SSL initialization: type 0x10=start, val 1=?
2024-04-29 20:10:48.716 [network debug] SSL state 0=before SSL initialization: type 0x1001=loop,connect, val 1=?
2024-04-29 20:10:48.718 [network debug] SSL BIO state 2 res -1
2024-04-29 20:10:48.775 [network debug] SSL BIO state 3 res 1
2024-04-29 20:10:48.776 [network debug] SSL BIO state 4 res 7
2024-04-29 20:10:48.776 [network debug] SSL BIO state 6 res 0
2024-04-29 20:10:48.776 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2024-04-29 20:10:48.796 [network debug] SSL BIO state 5 res -1
2024-04-29 20:10:48.796 [network debug] SSL BIO state 5 res 1
2024-04-29 20:10:48.796 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2024-04-29 20:10:48.817 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2024-04-29 20:10:48.837 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1001=loop,connect, val 1=?
2024-04-29 20:10:48.837 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2024-04-29 20:10:48.857 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2024-04-29 20:10:48.878 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2024-04-29 20:10:48.898 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1001=loop,connect, val 1=?
2024-04-29 20:10:48.899 [network debug] SSL state 3=SSLv3/TLS read server hello: type 0x1001=loop,connect, val 1=?
2024-04-29 20:10:48.900 [network debug] SSL state 38=TLSv1.3 read encrypted extensions: type 0x1001=loop,connect, val 1=?
2024-04-29 20:10:48.910 [network debug] SSL state 4=SSLv3/TLS read server certificate: type 0x1001=loop,connect, val 1=?
2024-04-29 20:10:48.911 [network debug] SSL state 39=TLSv1.3 read server certificate verify: type 0x1001=loop,connect, val 1=?
2024-04-29 20:10:48.912 [network debug] SSL state 11=SSLv3/TLS read finished: type 0x1001=loop,connect, val 1=?
2024-04-29 20:10:48.912 [network debug] SSL state 16=SSLv3/TLS write change cipher spec: type 0x1001=loop,connect, val 1=?
[... 69 lines cut ...]
-f --scanconfig=full --configpath=/home/pi/ebusd-configuration/ebusd-2.1.x/en -d ens:/dev/ttyAMA0 --latency=60 --loglevel=debug

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter v3 RPi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

see above

roelof4, 2024-05-01:
Ok, some progress to report!

After changing device name to /dev/ttyAMA0 (without ens: or enh:) 
I am getting scan results that seem to make sense. I also went for a longer connecting cable rather than the sort 30cm lead from the VWZ controller.
Now onto trying to get data out and onto my mqtt broker ....

john30, 2024-05-01:
the device string to ebusd depends on the settings in the device, so on a v3 it depends on how it is configured with the jumpers and/or the ebuspicloader, see the docs [here](https://adapter.ebusd.eu/v31/index.en.html) or [here](https://adapter.ebusd.eu/v30/index.en.html)

closed 2024-05-01 by john30