john30/ebusd #1278: Entities in Home Assistant have double names

TomStarren, 2024-05-20:
### Description

I'm currently using the Home Assistant Add-on to get the ebus values in MQTT and HA. This works great, but the entities in HA aren't named correct

### Actual behavior

Entities are names as shown below.
[image]

### Expected behavior

Entities shouldn't have the ebusd_hmu_ prefix

### ebusd version

23.3

### ebusd arguments

ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser= --mqttpass= --scanconfig --mqttjson --configpath=/config/ebusd-config/latest/en --latency=20 --pollinterval=5 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttvar=topic=%prefix/%circuit/%name --mqtttopic=ebusd --lograwdatafile=/config/ebusd-logs/raw.log --lograwdatasize=100 --device=enh:192.168.40.163:3335 --log=all:error | s6-log 1 n5 s1000000 /config/ebusd-logs

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

MQTT generic, MQTT Home Assistant via mqtt-hassio.cfg

### Logs

.

Koky05, 2024-05-21:
Name prefix is defined in these file:
`/etc/ebusd/mqtt-hassio.cfg`

closed 2024-05-22 by TomStarren

wrongisthenewright, 2024-05-28:
I see this as closed and it may be so but due to a forced HA update due to container corruption to 2024.5.X  I had the same problem, even with an unchanged mqtt-hassio.cfg all ebusd discovered entities had doubled the prefix.
The only way I found to (almost) fix the issue was a python script to bulk rename all the entities. I lost the history, but at least I kept my dashboard running.