john30/ebusd #1301: Template variable warning: 'value_json' is undefined

wiedwo, 2024-07-10:
### Description

2024-07-10 20:42:43.773 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'value_json' is undefined when rendering '{% set my_new = value_json|truncate(255)|regex_replace(find='^[^,]*|, device firmware |,.*| available',replace='') %}{% set my_cur = 'old' %}{% if my_new == '' %}{% set my_new = 'current' %}{% set my_cur = 'current' %}{% endif %}{{ {'installed_version':my_cur,'latest_version':my_new,'entity_picture':'https://adapter.ebusd.eu/favicon.ico','release_url':'https://adapter.ebusd.eu/firmware/ChangeLog'} | tojson }}'

2024-07-10 20:42:43.775 WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'value_json' is undefined when rendering '{% set my_new = value_json|truncate(255)|regex_replace(find=',.*| available|revision v|version v|OK',replace='') %}{% if my_new == '' %}{% set my_new = '23.2' %}{% endif %}{{ {'installed_version':'23.2','latest_version':my_new,'entity_picture':'https://ebusd.eu/logo-32x32.png','release_url':'https://github.com/john30/ebusd/releases/latest'} | tojson }}'

### Actual behavior

none

### Expected behavior

startup error

### ebusd version

23.2

### ebusd arguments

none

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via Ethernet

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

none

john30, 2024-07-13:
what are the arguments given to ebusd? maybe `--mqttjson` is missing?

wiedwo, 2024-07-16:
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=Noh3ahth2helo5AoX7CheRiaFoo8in7uv1ohyeiphaith6ohquiedaewookie4Ai --scanconfig --mqttjson --configpath=/config/ebusd-configuration/ochsner/1524849 --latency=40 --accesslevel=* --pollinterval=10 --mqttint=/config/ebusd-configuration/mqtt-hassio.cfg --mqttvar="filter-name=" --mqtttopic=ebusd --device=ens:192.168.178.68:9999 --log=all:info

2024-07-16 12:49:11.007 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.178.68:9999, enhanced

john30, 2024-09-05:
seems to be some issue with HA not providing the message value as "value_json" variable. did you change the mqtt-hassio.cfg file? or somehow disable mqtt json support in HA?

john30, 2024-10-26:
if this is still an issue, then please post here the value actually sent to HA in the MQTT integration for the global/updatecheck topic

AlexandreA74, 2024-11-16:
I have the same issue on a new installation, with an adapter I ordered last year.

Should I update the firmware of the adapter ?  I was thinking it would not matter, it was working fine in the past, and this warning started to come a few months ago.  I had to reinstall everything for other reasons.

[image]

john30, 2024-11-26:
again the question if --mqttjson was passed to ebusd as I doubt that

AlexandreA74, 2024-11-26:
Yes, ebusd runs with the --mqttjson option set

john30, 2024-11-26:
and was the updatecheck topic last pushed to recently, i.e. with the mqttjson set?

AlexandreA74, 2024-11-26:
I'm running 20231015 on the adapter - would updating it help ?

john30, 2024-12-01:
I don't think so. it anyway depends on when the message was last received, so please just check in HA when the last message was received. i suspect that to be too old

AlexandreA74, 2024-12-01:
You are right: the message is not appearing anymore.

john30, 2024-12-14:
ok, closing