john30/ebusd #1302: Field name containing slash (e.g. "service/time") leads to [mqtt error] received unmatchable topic

SmartLessing, 2024-07-10:
### Description

I am using a configuration with slashes in field names to group them in mqtt for an Ochsner Heatpump:

```
r,,service/time,02-072 Set Time,,,,02c80040,,,param,,,,,,MIN,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
w,,service/time,02-072 Set Time,,,,02c80040,,,MIN,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
```

writing to the field via ebusctl works as expected:
```
ebusctl write -c 24849 service/time 21:00
done
```

### Actual behavior

writing via mqtt to the field **service/time** leads to an error message:

`mosquitto_pub -h localhost -t ebusd/24849/service/time/set -m '21:00' -d`

`2024-07-10 21:40:07.983 [mqtt error] received unmatchable topic ebusd/24849/service/time/set`

changing the field name to **service.time** works as expected:

`mosquitto_pub -h localhost -t ebusd/24849/service.time/set -m '21:00' -d`

`2024-07-10 21:40:07.983 [mqtt notice] write 24849 service.time: 21:00`

### Expected behavior

writing via mqtt to the field **service/time** writes the message to ebus:

`mosquitto_pub -h localhost -t ebusd/24849/service/time/set -m '21:00' -d`

`[mqtt notice] write 24849 service/time: 21:00`

### ebusd version

23.3

### ebusd arguments

EBUSD_OPTS="--scanconfig --configpath=/etc/ebusd/ -d ens:ebus-air11:9999 --latency=30 --pollinterval=5 --accesslevel=* --httpport=8889 --htmlpath=/va$ --mqtthost=raspberrypi --mqttport=1883 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttjson"

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via Ethernet

### Related integration

MQTT generic

### Logs

2024-07-10 21:40:07.983 [mqtt error] received unmatchable topic ebusd/24849/service/time/set
2024-07-10 21:49:12.530 [mqtt notice] write 24849 service.time: 21:00

john30, 2024-07-13:
field names, message and circuit name are supposed to be identifiers similar to those of almost every programming language and as such all character other than alphanumeric and underscore should not be used in the first place.
I'll harden ebusd to decline use of such in future

closed 2024-07-13 by john30

SmartLessing, 2024-07-13:
@john30 thank you very much for your analysis and feedback. Got it: no slashes in field names.
Is there another way via config file to group the field names in a folder structure in mqtt without breaking mqtt?

[image]

john30, 2024-07-13:
the logical way of doing this would be to use an appropriate circuit for those.
since the single address/device 24849 seems to contain all aspects of your heating system, it seems right to split the messages accordingly. so maybe there are like sections of messages (those sharing a common pbsbid prefix) that all address a logical unit, e.g heater, hwc and so on? then it would make sense this way I guess

commit e703e9b7 (2024-07-13): assert names are valid identifier with little exceptions only (closes #1302)