john30/ebusd #1304: raspi:8889/data: SyntaxError: JSON.parse: bad control character in string literal at line 5636 column 19 of the JSON data

WSchimi, 2024-07-11:
### Description

ioBroker (and browser) cannot interpret json-data from ebusd and shows no actual data since 10.07.2024 08:48.
The problem is:
"global": {
  "version": "23.3.23.3",
  "updatecheck": "
OK, vaillant/08.bai.csv: different version available, vaillant/bai.0010010674.inc: different version available",

The problem is the LINEFEED <0x0a> befor OK, vaillant/... in JSON-data of "updatecheck":

### Actual behavior

After restart of ebusd it is working for ~3 minutes till the "updatecheck": is in JSON context with the wrong data (linefeed) befor the string "OK" (see Description)

### Expected behavior

working as 10.07.2024 befor 10:58 MESZ

### ebusd version

23.3

### ebusd arguments

EBUSD_OPTS="--scanconfig --device=ens:192.168.2.78:9999 --configpath=/home/pi/ebusd-configuration/ebusd-2.1.x/de --log=all:error --httpport=8889"

### Operating system

Debian 10 (Buster) / Ubuntu 18-19 / Raspbian 10 / Raspberry Pi OS 10 (including lite)

### CPU architecture

armv7l

### Dockerized

None

### Hardware interface

Adapter Shield v5 via WiFi

### Logs

there are no relevant logs for this problem

WSchimi, 2024-07-11:
adding ebusd argument "--updatecheck=off" fixed this problem.

dibman, 2024-07-11:
got the same problem, thx for this hint/workaround -> works
in result its an issue with the update check of ebusd?

WSchimi, 2024-07-11:
I think so. But I can't find the reason where the line feed is placed before the "OK". The Version of my ebusd executable is unchanged.

commit ad364c68 (2024-07-11): fix extra trailing newline (fixes #1304)