john30/ebusd #1307: Duplictae entry in csv not reported correctly

thomaskilian, 2024-07-12:
### Description

I have created a new csv for my WTSol and trying to add more lines I see this message:
```
2024-07-12 16:33:35.988 [main error] error reading scan config file tem/fc..sc.csv for ID "wrsol", SW2422, HW1112: ERR: duplicate 
entry, tem/fc..sc.csv:5: ERR: duplicate entry, duplicate ID
```
Well, please, which one is duplicate? I almost started and have only 3 lines which makes guessing not so difficult. But clearly an error message should indicate the error and not just croak "something is wrong".

### Actual behavior

see above. incomplete error message

### Expected behavior

Indication of which ID exactly is duplicate

### ebusd version

23.2

### ebusd arguments

ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=mqtt --mqttpass=c00kies --scanconfig --mqttjson --configpath=/config/ebusd/config --accesslevel=* --mqttint=/config/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.2.136:9999 --log=all:notice --enablehex | s6-log 1 n5 s1000000 /config/ebusd/logs 2

### Operating system

other

### CPU architecture

arm64

### Dockerized

same as ebusd version

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

not relevant

john30, 2024-07-13:
whats so difficult about loooking at line 5 of the definitions as reported? every message has an id field, so this would be a good start to check. and the name+circuit need to be unique, which should be obvious as well.
anyway, if you shared the definitions, it would be easy to help. this way it isn't.

thomaskilian, 2024-07-13:
If you KNOW that the 5 indicates the line, it's easier. of course. But an attitude about creating meaningful error messages should be planted in any coder's heart. What's so difficult in changing the code to printing "duplicate ID <ident>"?

john30, 2024-07-13:
colon followed by line number does not really seem unusual at all to me - basically nearly every code editor does it this way, so for a coder not a problem at all.

> ...What's so difficult in changing the code to printing "duplicate ID "?

already the case, at least according to your own comment above:

> ```
> 2024-07-12 16:33:35.988 [main error] error reading scan config file tem/fc..sc.csv for ID "wrsol", SW2422, HW1112: ERR: duplicate 
> entry, tem/fc..sc.csv:5: ERR: duplicate entry, **duplicate ID**
> ```

closed 2024-07-13 by john30