john30/ebusd #1311: RecoVAIR and Hmu on same address 08 are not scanable

rICTx-T1D, 2024-07-16:
### Description

Hi!
The problem was reported some years ago. When you have a heating pump on 08 and a recoVAIR on same, it is not possible to scan recoVAIR. 

Heat pump, recoVAIR and Vrc720 are on same ebus and connected to ebus adapter shield 5 with last ebusd software. 

Please review the problem an looking for a solution to discover on 38 or both under 08.

Have many thanks 
rict

### Actual behavior

On scan full

### Expected behavior

RecoVAIR was not discovered 

### ebusd version

23.3

### ebusd arguments

.

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

devel

### Hardware interface

Adapter Shield v5 via Ethernet

### Related integration

TCP (cmdline client like ebusctl or netcat), HTTP

### Logs

.

JonesGW-MGD, 2024-07-17:
Hi there,

as you can imagine, it's not possible to have two heat sources on the same bus address.  What Valliant implements in such cases is a VR32 Bus Coupler.  This 'fronts' one of your heat sources, in your case the recoVAIR and provides it with a different address - in your case 38.

Of course, there is no way for ebusd to 'know' what type of device sits behind the bus coupler - it could equally be another heat pump in cascade mode, a back-up gas boiler - in your case it just happens to be the recoVAIR.

So there can never be a VR32 configuration file that will work in all these circumstances.  (also the VR32 can present on different addresses to 38 too.)

There are actually a number of posts in the discussion forum on how to address this (and I've posted or commented on a few).  You may wish to start here and follow some of the referenced links too:  

https://github.com/john30/ebusd/discussions/1223#discussioncomment-8749250

But in broad terms - you need to i) host the configuration files locally, ii) 'fool' ebusd by renaming in your case 08.recov.csv to 38.vr32.csv,  iii) adjust the latency parameter on daemon start-up since the VR32 doesn't play well with some of the bus standards.

Good luck

rICTx-T1D, 2024-07-17:
Thanks for your detailed answer. I will try it as in your summary describe.  Give me some days for feedback pls.

rICTx-T1D, 2024-07-18:
Hi @JonesGW-MGD,
have many thank for your last posting. Now my ebusd is founding some values from "recov"
```
docker exec -i ebusd ebusctl f -c recov
recov BypassPosition = 100.0
recov FlowActual = 171
recov FrostProtection = off
recov HumiWasteAir = 46.94
recov PreferredTemp = 23.00
recov TempInletAir = 24.88;ok
recov TempOutgoingAir = 27.75;ok
recov TempOutsideAir = 24.25;ok
recov TempWasteAir = 27.69;ok
recov VentHeatRecovery = auto
recov VentIntensityDay = 3
recov VentIntensityNight = 6
```

But when I try to write / change on this element, ebusd answer with "ERR: element not found"
```
X-OPENHAB:/home/ebusd # docker exec -i ebusd ebusctl w -c recov BypassPosition 70
ERR: element not found

X-OPENHAB:/home/ebusd # docker exec -i ebusd ebusctl w -c v32 BypassPosition 70
ERR: element not found
```

Did you have an idea how I con fix it?

JonesGW-MGD, 2024-07-19:
Both of those two messages you are trying to write to are defined in the configuration file as read messages:

`r9 | recov | BypassPosition | Position of Bypass Flap |   |   |   | 290b0e |   |   | UIN | 10 |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --`

`r | recov | PreferredTemp | Desired Temperature |   |   |   | 29140e |   |   | temp |   |  -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- '| --

I don't know enough about the recoVAIR as to whether you could amend these to be writable - i.e. `r;w, .......` or whether those messages are simply not-writable.  You may cause issues with your devices if you simply try things randomly, so I suggest you open a new thread on the discussion forum about writing RecoVAIR messages (rather than an 'issue') and you may find other users can advise you better?

I also suggest that, since your original post wasn't really an issue with the ebusd software, you change the issue into a discussion item.  I believe you should have an option to convert it.  It might also be helpful to others if you re-title it too - something like 'RecoVAIR - accessing data behind a VR32'. as it may contribute to the knowledge pool.

john30, 2024-09-05:
closed as not an issue in the first place

closed 2024-10-26 by john30