john30/ebusd #1326: HomeAssistant & WRSOL 2.0 & EBUS to WiFi Adapter Module V6.1

telemake69, 2024-08-16:
### Description

Hello
I am new on this forum and also technically new in term of Home Assistant using.
I have some issues with my Wrsol thermal system, reason for I want to monitor it by a digital way.
To do that, I bought a V6.1 adapter and connected it to my WRSOL 2.0 and I have installed ebusd in my HA setup on my raspberry.

I understand that, before accessing Wrsol data in HA, I need to connect sucessfully my adapter, but it seems too complex for me.

Ebusd configuration
 scanconfig: true
 loglevel_all: debug
 mqtttopic: ebusd
 mqttint: /etc/ebusd/mqtt-hassio.cfg
 mqttjson: true
 network_device: 192.168.1.51:3334
 latency: 100

mqtt-hassio.cfg is the one I picked from John configuration files

I put no additionnal lines in my YAML file, just use ebusd like that

As you can seen on the logs, it seems not enouh.

Can you please help the newbee I am ?

Thanks

### Actual behavior

The connection with the adapter is not done
A least two issues identified in the logs
 scan config fc: ERR: arbitration lost
 device invalid

### Expected behavior

Just want to connect my physical adapter successfully, then correctly configure it to access to my Wrsol data in HA...

### ebusd version

23.2

### ebusd arguments

 ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=tahXidoow4Ohgoh8ChaeghaepheibeuX0ahpheiquo5wor7ieDai0lee8quo4Aip --scanconfig --mqttjson --latency=100 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=192.168.1.51:3334 --log=all:notice 2024-08-17 00:37:13.228 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.1.51:3334

### Operating system

other

### CPU architecture

x64

### Dockerized

None

### Hardware interface

other

### Logs

2024-08-17 00:48:27.894 [main notice] SIGTERM received
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=tahXidoow4Ohgoh8ChaeghaepheibeuX0ahpheiquo5wor7ieDai0lee8quo4Aip --scanconfig --mqttjson --latency=100 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=192.168.1.51:3334 --log=all:notice
2024-08-17 00:48:46.800 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.1.51:3334
2024-08-17 00:48:48.112 [bus notice] bus started with own address 31/36
2024-08-17 00:48:48.306 [bus notice] signal acquired
2024-08-17 00:48:48.525 [mqtt notice] connection established
2024-08-17 00:49:06.172 [bus error] signal lost
2024-08-17 00:49:06.184 [bus notice] signal acquired
2024-08-17 00:49:26.113 [bus error] signal lost
2024-08-17 00:49:26.308 [bus notice] signal acquired
2024-08-17 00:50:02.042 [bus error] signal lost
2024-08-17 00:50:02.306 [bus notice] signal acquired
2024-08-17 00:50:11.911 [bus notice] new master f7, master count 2
2024-08-17 00:50:19.941 [main error] scan config fc: ERR: arbitration lost
2024-08-17 00:50:33.402 [main error] scan config fc: ERR: arbitration lost
2024-08-17 00:50:42.343 [bus notice] device invalid
2024-08-17 00:50:47.575 [bus notice] re-opened 192.168.1.51:3334
2024-08-17 00:50:48.626 [main error] scan config fc: ERR: arbitration lost
2024-08-17 00:50:50.878 [main notice] update check: version 23.3 available
2024-08-17 00:51:02.239 [main error] scan config fc: ERR: arbitration lost
2024-08-17 00:51:10.599 [bus notice] device invalid
2024-08-17 00:51:15.837 [bus notice] re-opened 192.168.1.51:3334
2024-08-17 00:51:16.775 [main error] scan config fc: ERR: arbitration lost

telemake69, 2024-08-16:
Activating Debug

2024-08-17 01:01:02.963 [bus info] scan fc cmd: 31fc070400
2024-08-17 01:01:05.100 [bus notice] re-opened 192.168.1.51:3334
2024-08-17 01:01:05.100 [bus debug] start request 31
2024-08-17 01:01:05.100 [bus debug] arbitration start with 31
2024-08-17 01:01:05.291 [bus debug] ERR: read timeout during skip, switching to no signal
2024-08-17 01:01:05.291 [main error] scan config fc: ERR: no signal
2024-08-17 01:01:05.292 [bus error] signal lost
2024-08-17 01:01:05.292 [mqtt debug] publish ebusd/global/scan "running"
2024-08-17 01:01:05.377 [bus debug] ERR: SYN received during no signal, switching to ready
2024-08-17 01:01:05.377 [bus notice] signal acquired
2024-08-17 01:01:05.630 [bus debug] arbitration lost
2024-08-17 01:01:08.297 [mqtt debug] publish ebusd/global/uptime 400
2024-08-17 01:01:13.007 [bus debug] ERR: read timeout during skip, switching to no signal
2024-08-17 01:01:13.007 [bus error] signal lost
2024-08-17 01:01:16.540 [bus notice] device invalid
2024-08-17 01:01:21.681 [bus notice] re-opened 192.168.1.51:3334
2024-08-17 01:01:21.763 [bus debug] ERR: SYN received during no signal, switching to ready
2024-08-17 01:01:21.763 [bus notice] signal acquired
2024-08-17 01:01:24.314 [mqtt debug] publish ebusd/global/uptime 416
2024-08-17 01:01:25.292 [main debug] performing regular tasks
2024-08-17 01:01:25.292 [mqtt debug] publish ebusd/global/scan "finished"

john30, 2024-09-05:
ask the device manufacturer (not me). does not look like an ebusd issue and as such closed

closed 2024-09-05 by john30