john30/ebusd #1348: devel - HA Entity type incorrect / nested values ?

Xyaren, 2024-10-04:
### Description

I switched to the latest devel tag and I am experiencing some unexpected changes.

Docker compose: 

```
services:
  ebusd:
    image: john30/ebusd:devel
    restart: unless-stopped
    environment:
      EBUSD_SCANCONFIG: ''
      EBUSD_DEVICE: 'ens:/dev/ttyAMA0'
      EBUSD_MQTTHOST: 192.168.2.4
      EBUSD_MQTTPORT: 1883
      EBUSD_MQTTUSER: "ebusd"
      EBUSD_MQTTPASS: "XXXXXXXXXXXXXXXX"
      EBUSD_MQTTINT: /etc/ebusd/mqtt-hassio.cfg
      EBUSD_MQTTJSON: ''
      EBUSD_MQTTLOG: ''
      EBUSD_POLLINTERVAL: '2'
      EBUSD_MQTTVAR: "filter-direction=r|u|^w,filter-name=^Flame$|^HcPumpMode$|^ReturnTemp$|^StorageTemp$|^FlowTemp$|^FlowTempDesired$|^Statenumber$|^Hc1ActualFlowTempDesired$|^Hc1HeatCurve$|^Hc1ManualOPRoomTempDesired$|^ActualRoomTemp>
#^Statenumber$"
#Flame$|HcPumpMode$|ReturnTemp$|StorageTemp$|FlowTemp$|FlowTempDesired$|Statenumber$|Hc1ActualFlowTempDesired$|Hc1HeatCurve$|Hc1ManualOPRoomTempDesired$|ActualRoomTempDesiredHc1$|Hc1MinimalFlowTempDesired$|Hc1NightTemp$|Hc1OPMode$|Hc1Q>
      EBUSD_MQTTCLIENTID: "ebusd-rpi"
    devices:
      - /dev/ttyAMA0
    logging:
      driver: "journald"
```

### Actual behavior

Heat curve is no longer of type "number" and is therefore not editable within HA:

[image]

Aso you can see the actual published value is double-wrapped in a "value" attribute:
[image]

### Expected behavior

I would not have expected the change in mqtt format which causes existing HA entities to break.

### ebusd version

current source from git

### ebusd arguments

(see docker compose)

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

devel

### Hardware interface

Adapter Shield v5 via Raspberry GPIO

### Related integration
[... 102 lines cut ...]
2024-10-04 22:27:14.226 [mqtt notice] read bai WaterPressure: ?7
2024-10-04 22:27:14.351 [update notice] sent poll-read bai Gasvalve QQ=31: off
2024-10-04 22:27:14.351 [mqtt notice] read bai GasValve: ?4
2024-10-04 22:27:14.495 [update notice] sent poll-read bai HwcTemp QQ=31: 31.94;ok
2024-10-04 22:27:14.495 [mqtt notice] read bai HwcTemp: ?4
2024-10-04 22:27:14.624 [update notice] sent poll-read bai HwcWaterflow QQ=31: 0.19
2024-10-04 22:27:14.624 [mqtt notice] read bai HwcWaterflow: ?4
2024-10-04 22:27:14.750 [update notice] sent poll-read bai PositionValveSet QQ=31: 0
2024-10-04 22:27:14.751 [mqtt notice] read bai PositionValveSet: ?4
2024-10-04 22:27:14.881 [update notice] sent poll-read bai Statenumber QQ=31: 97
2024-10-04 22:27:17.119 [update notice] sent poll-read f43 Hc1ManualOPRoomTempDesired QQ=31: 21.0
2024-10-04 22:27:17.902 [update notice] received read bai Status01 QQ=10: 33.0;33.0;-;31.0;34.0;off
2024-10-04 22:27:19.860 [update notice] received update-read Broadcast Vdatetime QQ=10: 00:27:20;05.10.2024
2024-10-04 22:27:20.114 [update notice] received update-write bai StatusCirPump QQ=10: off
2024-10-04 22:27:20.185 [update notice] sent poll-read f43 ActualRoomTempDesiredHc1 QQ=31: 21.00
2024-10-04 22:27:21.872 [update notice] received update-write bai SetMode QQ=10: auto;39.0;60.0;-;0;0;1;0;0;0
2024-10-04 22:27:23.157 [update notice] sent poll-read f43 Hc1OPMode QQ=31: manual
2024-10-04 22:27:25.879 [update notice] received read bai Status01 QQ=10: 33.0;33.0;-;31.0;34.0;off
2024-10-04 22:27:26.128 [update notice] sent poll-read f43 HwcOPMode QQ=31: auto
2024-10-04 22:27:27.893 [update notice] received read bai Status02 QQ=10: auto;60;75.0;70;60.0
2024-10-04 22:27:29.156 [update notice] sent poll-read bai Statenumber QQ=31: 1
2024-10-04 22:27:31.880 [update notice] received update-write bai SetMode QQ=10: auto;39.0;60.0;-;0;0;1;0;0;0
2024-10-04 22:27:32.131 [update notice] sent poll-read f43 Hc1ManualOPRoomTempDesired QQ=31: 21.0
2024-10-04 22:27:35.132 [update notice] sent poll-read f43 ActualRoomTempDesiredHc1 QQ=31: 21.00
2024-10-04 22:27:35.915 [update notice] received read bai Status01 QQ=10: 36.0;33.0;-;31.0;34.0;on

john30, 2024-10-05:
this is not related to the changes in [mqtt-hassio.cfg](https://github.com/john30/ebusd/commits/master/contrib/etc/ebusd/mqtt-hassio.cfg) at all but comes with the switch to the new CSV delivery via https://ebus.github.io/ serving files generated from new TypeSpec message definition sources, [see the Changelog](https://github.com/john30/ebusd/blob/master/ChangeLog.md#breaking-changes)

john30, 2024-10-05:
should be better with the [latest adjustment](https://github.com/john30/ebusd/commit/254517f9db70e16dd851a0951e770bea7fdbcf8e)

Kolbi, 2024-10-09:
Seems to be still not fixed in latest devel. Tried it with HASSIO MQTT.

Xyaren, 2024-10-10:
I can edit the curve now.
The nested "value" json is still there but shouldn't be a big issue. It's just weird.

john30, 2024-10-25:
it is necessary like that for messages having more than one field, so working as intended now

closed 2024-10-25 by john30