john30/ebusd #1349: Wolf CGS-2 detected but connectivity lost after ~1 hour

akallabeth, 2024-10-06:
### Description

I am using a `ebus` adapter shield v5 and a `wolf cgs-2` with a `BM-2` with `ebusd` (that is queried by `openhab`)

the detection works and I can read the values from the device at first.
After a few minutes / an hour the whole thing locks up and no more data can be read.

* I have updated the adapter shield firmware just yesterday to be sure not to run into any bug with older firmware
* same is true for `Wolf BM-2`

### Actual behavior

device locks up, no more data can be read

### Expected behavior

well, the `ebusd` works stable and data can be read

### ebusd version

23.3

### ebusd arguments

EBUSD_OPTS="-d ens:/dev/ttyAMA0 --latency=50 --enablehex -l /var/log/ebusd.log -p 8888 --localhost"

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via Raspberry GPIO

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

```
2024-10-06 14:08:45.280 [bus notice] device status: transport opened
2024-10-06 14:08:45.281 [main notice] ebusd 23.3.23.3 started with broadcast scan on device: /dev/ttyAMA0, serial high speed, enhanced
2024-10-06 14:08:45.942 [bus notice] bus started with own address 31/36
2024-10-06 14:08:45.942 [bus notice] device status: reset, supports info
2024-10-06 14:08:45.942 [bus notice] signal acquired
2024-10-06 14:08:45.944 [bus notice] device status: extra info: firmware 1.1[4819].1[4819], jumpers 0x03
2024-10-06 14:08:46.370 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:46.460 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:46.597 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:46.698 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:46.788 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:46.983 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:46.992 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:47.009 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:48.014 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:48.099 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:48.185 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:48.270 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:48.355 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:48.441 [bus error] device status: unexpected enhanced byte 2
2024-10-06 14:08:48.489 [bus error] device status: missing enhanced byte 2
[... 671 lines cut ...]
2024-10-06 14:10:35.356 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:35.385 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:35.413 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:35.764 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:35.816 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:36.177 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:36.207 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:36.595 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:36.998 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:37.021 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:37.371 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:37.388 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:37.568 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:37.910 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:38.337 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:38.849 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:39.191 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:39.959 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:40.301 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:40.320 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:40.628 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:40.727 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:40.767 [bus error] device status: missing enhanced byte 2
2024-10-06 14:10:40.954 [bus error] device status: missing enhanced byte 2
```

john30, 2024-10-25:
not an ebusd issue but rather ebusd-esp32 related, thus closing.
anyway, please check again with current firmware [2024-10-08](https://adapter.ebusd.eu/v5/ChangeLog#20241008) and report on [ebusd-esp32](https://github.com/john30/ebusd-esp32/issues) if it still occurs

closed 2024-10-25 by john30

akallabeth, 2024-11-03:
@john30 ok, finally found some time to update this, but can´t find the firmware files anywhere? (am I blind? only the [changelog[(https://github.com/john30/ebusd-esp32/blob/main/CHANGELOG.md) is easy to find)
don´t have network where the adapter is so kind of hard to install :grin:

akallabeth, 2024-11-03:
@john30 ok, extracted the update link from the previous firmware, now running current version and that looks stable so far (well, as stable as I can tell after an hour)

akallabeth, 2024-11-05:
@john30 ok, thank you.
now already stable connected for 2 days.