john30/ebusd #1357: vr_70/Mc2FlowTempDesired missing in HomeAssistant

jan666, 2024-10-14:
### Description

I just got all this running and added ebusd to Home Assistant via MQTT Auto Discovery. 

```
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0902;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV3;SW=0709;HW=3704", loaded "vaillant/15.ctlv3.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0902;HW=5103", loaded "vaillant/76.vwzio.csv"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX3;SW=0121;HW=0404", loaded "vaillant/f6.netx3.csv"
```

I wanted to check all available data in Home Assistant. Now I found out that the Mixer Status is missing. I can see it in MQTT (using mosquitto_sub):

```
ebusd/vr_70/SensorData {
     "S1": {"value": 31.50},
     "S2": {"value": null},
     "S3": {"value": null},
     "S4": {"value": null},
     "S5": {"value": 33.25},
     "S6": {"value": 31.12},
     "ignore": {"value": "c8 7c 00"}}
ebusd/vr_70/SetActorState {
     "R1": {"value": "off"},
     "R2": {"value": null},
     "R3": {"value": "off"},
     "R4": {"value": "on"},
     "R5": {"value": null},
     "R6": {"value": null},
     "S7": {"value": "off"}}
ebusd/vr_70/Mc2FlowTempDesired {
     "FTStatus": {"value": "on"},
     "FTDesired": {"value": 31.0},
     "MixerStatus": {"value": "on"},
     "MixerMovement": {"value": 0}}

```

but it's not in Home Assistant. And I don't really underatand why it's there at all because when I use ebusctl find I can only see the Sensor Data S1-S6:

```
 $ ebusctl find  | grep -i vr_70
scan.52  = Vaillant;VR_70;0109;2903
vr_70 currenterror = no data stored
vr_70 errorhistory = no data stored
vr_70 SensorData = 31.44;-;-;-;33.25;31.06;c8 7c 00
```

Is there something I miss here?

### Actual behavior

ebusd/vr_70/Mc2FlowTempDesired is missing in home assistant

[image]

### Expected behavior

ebusd/vr_70/Mc2FlowTempDesired should be in home assistant

### ebusd version

23.3

### ebusd arguments

--scanconfig -d ens:192.168.42.8:9999 -c /home/pi/git/ebusd-configuration/ebusd-2.1.x/de/ --log=all:error --httpport=8080 --mqtthost 192.168.42.254 --mqttport 1883 --mqttuser ebusd --mqttpass ebusd --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttjson

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

...

Koky05, 2024-10-14:
Try to change in `mqtt-hassio.cfg` filter parts or run it without filter at all.
```
filter-direction=r|u|^w
#filter-name=
```

jan666, 2024-10-14:
I commented out all filter- entries but VR 70 remains unchanged. I still wondering why I cant see the values via ebusctl find tho

Koky05, 2024-10-14:
If commenting out filter-name did not make any change it could be unsupported type of values.

jan666, 2024-10-14:
MixerMovement is percent:

`52.vr_70.csv:w,,Mc2FlowTempDesired,,,,,"0201",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,`

but I think FTStatus, FTDesired and MixerStatus are instertesting, too.

jan666, 2024-10-16:
I removed everything from HASS yesterday and re-added it. I changed

`filter-name = status|temp|humidity|yield|count|energy|power|runtime|hours|starts|mode|curve|^load$|^party$|sensor|timer|mixer|actor`

to add mixer and actor. Now I have hc2mixermovement from the hmu:

[image]

but not as percent. And this is not in the csv file:

```
 $ grep -i mixermove *
15.700.csv:r,,Hc1MixerMovement,MixerMovement Heizkreis 1,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.700.csv:r,,Hc2MixerMovement,MixerMovement Heizkreis 2,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.700.csv:r,,Hc3MixerMovement,MixerMovement Heizkreis 3,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.720.csv:r,,Hc1MixerMovement,MixerMovement Heizkreis 1,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.720.csv:r,,Hc2MixerMovement,MixerMovement Heizkreis 2,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.720.csv:r,,Hc3MixerMovement,MixerMovement Heizkreis 3,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.b7v.csv:r,,Hc1MixerMovement,MixerMovement Heizkreis 1,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.b7v.csv:r,,Hc2MixerMovement,MixerMovement Heizkreis 2,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.b7v.csv:r,,Hc3MixerMovement,MixerMovement Heizkreis 3,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.bass.csv:r,,Hc1MixerMovement,MixerMovement Heizkreis 1,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.bass.csv:r,,Hc2MixerMovement,MixerMovement Heizkreis 2,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.bass.csv:r,,Hc3MixerMovement,MixerMovement Heizkreis 3,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.basv.csv:r,,Hc1MixerMovement,MixerMovement Heizkreis 1,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.basv.csv:r,,Hc2MixerMovement,MixerMovement Heizkreis 2,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.basv.csv:r,,Hc3MixerMovement,MixerMovement Heizkreis 3,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.ctlv2.csv:r,,Hc1MixerMovement,MixerMovement Heizkreis 1,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.ctlv2.csv:r,,Hc2MixerMovement,MixerMovement Heizkreis 2,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.ctlv2.csv:r,,Hc3MixerMovement,MixerMovement Heizkreis 3,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.ctlv3.csv:r,,Hc1MixerMovement,MixerMovement Heizkreis 1,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.ctlv3.csv:r,,Hc2MixerMovement,MixerMovement Heizkreis 2,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.ctlv3.csv:r,,Hc3MixerMovement,MixerMovement Heizkreis 3,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.ctlv3.SW0709.HW3704.csv:r,,Hc1MixerMovement,MixerMovement Heizkreis 1,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.ctlv3.SW0709.HW3704.csv:r,,Hc2MixerMovement,MixerMovement Heizkreis 2,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
15.ctlv3.SW0709.HW3704.csv:r,,Hc3MixerMovement,MixerMovement Heizkreis 3,,,,1A00,,,EXP,,,"status of mixer (<0 closing, >0 opening)"
26.vr_71.csv:w,,Mc1FlowTempDesired,,,,,0200,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
26.vr_71.csv:w,,Mc2FlowTempDesired,,,,,0201,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
26.vr_71.csv:w,,Mc3FlowTempDesired,,,,,0202,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
52.vr_70.csv:w,,Mc1FlowTempDesired,,,,,"0200",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
52.vr_70.csv:w,,Mc2FlowTempDesired,,,,,"0201",FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,
```

and it's not updated that often - in MQTT I can see VR 70 already changed MixerMovement to 0:

```
2024-10-16T06:38:00+0200 ebusd/vr_70/Mc2FlowTempDesired 143 '{
     "FTStatus": {"value": "on"},
     "FTDesired": {"value": 33.0},
     "MixerStatus": {"value": "on"},
     "MixerMovement": {"value": 0}}'
```

I don't really understand whats going on. What am I missing?

john30, 2024-10-25:
as the entity ID in HA should reveal, it is part of the ctlv3 and this will only be published to MQTT discovery when the message was actively sent by a unit on the bus if not adjusted otherwise by https://github.com/john30/ebusd/blob/3a9ee44728446cee4607062e36a050658b749515/contrib/etc/ebusd/mqtt-hassio.cfg#L96-L100

jan666, 2024-10-25:
Sorry, I don't really get it :(

I changed it 
`filter-seen = 1`

but I don't see any changes. I can see via mosqutto_sub something is sent to MQTT:

```
2024-10-25T08:03:03+0200 ebusd/vr_70/Mc2FlowTempDesired 144 '{
     "FTStatus": {"value": "on"},
     "FTDesired": {"value": 32.0},
     "MixerStatus": {"value": "on"},
     "MixerMovement": {"value": -5}}'
```

but in HASS i still see

[image]

with an outdated value. It's only updated after I read it via `ebusctl read Hc2MixerMovement`. Why is the newer value in MQTT automatically? And why is this value not (updated?) in HASS?

dylan09, 2024-10-25:
You are looking for two different data points. In MQTT you are watching data from your mixer (vr_70). But in HA you are watching data from your sensoComfort (ctlv3).

Mixer data is permanently send and updated over eBus by mixer. But register from sensoComfort are updated rarely. You have to wait, until they are polled by ebusd.

jan666, 2024-10-25:
> You are looking for two different data points. In MQTT you are watching data from your mixer (vr_70). But in HA you are watching data from your sensoComfort (ctlv3).

Yes, that's what I wanted to say. My question is: how can I get the data from the VR_70 into HASS? Its there in MQTT but not added to HASS. Only the values from sensoComfort are added to HASS.

john30, 2024-10-26:
> 52.vr_70.csv:w,,Mc2FlowTempDesired

you pointed it out yourself: the message you're interested in is a write message and those are not published to discovery unless activated by adding the write direction to filter-direction, see "When appending |^w to the filter-direction..." here: https://github.com/john30/ebusd/wiki/MQTT-integration#home-assistant

closed 2024-10-26 by john30

jan666, 2024-10-28:
I really still don't get what I'm doing wrong :(

`filter-name = status|temp|humidity|yield|count|energy|power|runtime|hours|starts|mode|curve|^load$|^party$|sensor|timer|mixer|actor`

`filter-direction = r|u|^w`

but the values are still missing. Seeing a lot of writable entities now, but not those I'm looking for.

[image]

[image]

jan666, 2024-10-28:
I just found https://github.com/john30/ebusd/issues/1113 and it sounds like the same problem to me. It works for him with 23.2 but not 23.3 (which is the first version I ever used)

john30, 2024-10-28:
these messages might have appeared in an older ebusd version, but were not usable at all since the write message contains more than one field in the master part. consequently, these can't be set via mqtt at all