john30/ebusd #1362: HA Addon ebusd / MQTT Daten werden nicht vollständig angelegt im HA

heasti, 2024-10-18:
### Description

Die Daten welche mittels 15.ctlv2.csv abgefragt werden, werden nicht vollständig im HA per MQTT angezeigt. Es fehlen z.b. z1Name1 sowie die gesamten Zeitprogramme für Warmwasser Zirkulationspumpe und Zone 1-3.

### Actual behavior

Die Daten werden im MQTT-Explorer angezeigt, werden aber nicht automatisch im HA angelegt und angezeigt. 

### Expected behavior

Eine Vollständige Darstellung aller Daten die über MQTT ankommen werden automatisch als Entitäten erstellt.

### ebusd version

23.2

### ebusd arguments

-

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

-

jan666, 2024-10-18:
check the `mqtt-hassio.cfg` file for `filter-...` and adjust for your needs. And maybe see https://github.com/john30/ebusd/issues/1357

heasti, 2024-10-24:
@jan666  Thanks for the hint. After some searching I found the possibility in the EBUSD addon
[image]

closed 2024-10-25 by john30