john30/ebusd #1366: ebusd not scanning devices correctly and no write messages can be sent to bus

pauldtc, 2024-10-26:
### Description

After starting ebusd it acquires correctly the signal to my ebus adapter but devices remain in "scanning" status except ebusd.

version: ebusd 23.3.23.3
device: /dev/ttyACM0, serial high speed, enhanced, firmware 1.1[4a08].1[4a08]
access: *
signal: acquired
symbol rate: 111
max symbol rate: 183
min arbitration micros: 6
max arbitration micros: 39
min symbol latency: 5
max symbol latency: 8
scan: finished
reconnects: 0
masters: 6
messages: 296
conditional: 0
poll: 0
update: 54
address 00: master #1
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 37: master #18
address 3c: slave #18, scanning
address 70: master #4
address 75: slave #4, scanning
address 7f: master #24
address 84: slave #24, scanning
address f0: master #5
address f5: slave #5, scanning

Furthermore when running "ebusctl find" i get "(ERR: invalid position for 313c070400 / 00)"

heatpump silent_mode = no data stored
heatpump water_pump_info = no data stored
ignored ign4 =
ignored ign5 =
ignored ign6 = no data stored
scan.05  = no data stored
scan.3c  =  (ERR: invalid position for 313c070400 / 00)
scan.75  =  (ERR: invalid position for 3175070400 / 00)
scan.84  =  (ERR: invalid position for 3184070400 / 00)
scan.f5  =  (ERR: invalid position for 31f5070400 / 00)
sensys 23_sw_version = no data stored
sensys boiler_sw_version = no data stored

Boiler is a Ariston Alteas One, CSV file is the one from wrongisthenewright's repo (https://github.com/wrongisthenewright/ebusd-configuration-ariston-bridgenet).

Sorry if my report is ambiguous but i'm fairly new to ebusd and i don't know exactly what to look for.

### Actual behavior

Running "ebusctl read -m 0 -c energymgr z1_heat_request" returns "ERR: read timeout"

### Expected behavior

Return actual value ON or OFF

### ebusd version

23.3

### ebusd arguments

--scanconfig --configpath=/etc/ebusd -d ens:/dev/ttyACM0 --mqttport=1883 --mqtthost=172.10.2.101 --mqttuser ebusd --mqttpass Interbelic123! --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd/%circuit/%name --enablehex --accesslevel=* --latency=80000 --receivetimeout=100000  --acquiretimeout=20000  --acquireretries=10  --sendretries=10

### Operating system

[... 496 lines cut ...]
2024-10-26 19:21:50.726 [update notice] received unknown BC cmd: 37fe20100e6d96be0000c69402006e96be0000
2024-10-26 19:21:50.851 [bus error] send to 18: ERR: read timeout, retry
2024-10-26 19:21:50.944 [update notice] received unknown BC cmd: 37fe201010d3460000cf9c0000d9950000da950000
2024-10-26 19:21:51.061 [bus error] send to 18: ERR: read timeout, retry
2024-10-26 19:21:51.155 [update notice] received unknown BC cmd: 37fe201010db950000dc950000dd950000de950000
2024-10-26 19:21:51.274 [bus error] send to 18: ERR: read timeout
2024-10-26 19:21:51.275 [bus error] send message part 0: ERR: read timeout
2024-10-26 19:21:51.367 [update notice] received unknown BC cmd: 37fe201010d9940000da940000db940000dc940000
2024-10-26 19:21:51.492 [update notice] received unknown BC cmd: 37fe201008dd940000de940000
2024-10-26 19:21:51.586 [update notice] received read boiler config_version_counter QQ=70: 7
2024-10-26 19:21:51.680 [update notice] received update-read energymgr hybrid_LWT_setpoint QQ=37: 58.0
2024-10-26 19:21:51.774 [update notice] received read boiler boiler_status QQ=7f: standby
2024-10-26 19:21:51.867 [update notice] received unknown BC cmd: 37fe20100f639744020064974402006597440200
2024-10-26 19:21:51.972 [update notice] received update-read ignored ign5 QQ=7f:
2024-10-26 19:21:52.023 [update notice] received unknown BC cmd: 37fe2010056697440200
2024-10-26 19:21:52.088 [update notice] received unknown BC cmd: 00fe203a0129
2024-10-26 19:21:52.154 [update notice] received unknown BC cmd: 37fe2010096147500000d7460000
2024-10-26 19:21:52.273 [update notice] received update-read ignored ign4 QQ=7f:
2024-10-26 19:21:52.275 [main error] unable to load scan config f5: slave part too short (0)
2024-10-26 19:21:52.276 [main error] scan config f5: empty
2024-10-26 19:21:52.324 [update notice] received unknown BC cmd: 37fe2010057310690000
2024-10-26 19:21:52.469 [update notice] received unknown BC cmd: 37fe20100d7647690000d1940200d2940200
2024-10-26 19:21:52.627 [update notice] received unknown BC cmd: 37fe201010d3940200d4940200d5940200d6940200
2024-10-26 19:21:52.785 [update notice] received unknown BC cmd: 37fe2010100a1901000c1900000d1900000e190000
2024-10-26 19:21:52.977 [update notice] received unknown MS cmd: 3784200008c0f21421c5f00921 / 050f01000001

john30, 2024-10-27:
looks like the devices don't respond to the scan message (e.g. "empty" or "slave part too short") and as a consequence, you can't use the scanconfig feature at all. instead you'd have to manually pick CSVs that match your devices, put them in a folder and point ebusd to it

closed 2024-10-27 by john30

pauldtc, 2024-10-27:
Hi John, thank you for your quick response.

I removed the --scanconfig parameter from the the ebusd configuration file and tried again to read the parameters.

It now returns ERR: SYN received / ERR: wrong symbol received:

root@DESKTOP-475AH0D:/home/paul# ebusctl read -m 0 -c energymgr z1_heat_request
ERR: SYN received

root@DESKTOP-475AH0D:/home/paul# ebusctl read -m 0 -c energymgr z2_heat_request
ERR: SYN received

This is what i found in the logs:

2024-10-27 12:48:32.487 [update notice] received unknown BC cmd: 37fe2010056697440200
2024-10-27 12:48:32.560 [update notice] received unknown BC cmd: 7ffe2020030c9000
2024-10-27 12:48:32.665 [update notice] received unknown BC cmd: 37fe2010096147500000d7460000
2024-10-27 12:48:32.761 [update notice] received unknown BC cmd: 7ffe207309000000000000000004
2024-10-27 12:48:32.935 [update notice] received unknown BC cmd: 37fe2001026572
2024-10-27 12:48:33.027 [update notice] received unknown BC cmd: 37fe200e0865725e015e013403
2024-10-27 12:48:33.099 [bus error] send to 18: ERR: SYN received, retry
2024-10-27 12:48:33.118 [bus error] send to 18: ERR: wrong symbol received, retry
2024-10-27 12:48:33.320 [bus error] send to 18: ERR: SYN received, retry
2024-10-27 12:48:33.338 [bus error] send to 18: ERR: wrong symbol received, retry
2024-10-27 12:48:33.510 [update notice] received read boiler config_version_counter QQ=70: 3            2024-10-27 12:48:33.589 [update notice] received update-read boiler boiler_status QQ=37: heating        2024-10-27 12:48:33.662 [bus error] send to 18: ERR: SYN received, retry                                2024-10-27 12:48:33.681 [bus error] send to 18: ERR: wrong symbol received, retry                       2024-10-27 12:48:33.807 [update notice] received unknown BC cmd: 37fe20101001910000029101000391000004910000                                                                                                     2024-10-27 12:48:33.930 [update notice] received unknown BC cmd: 37fe2010080591000006910000             2024-10-27 12:48:34.066 [update notice] received unknown BC cmd: 37fe200002c67e                         2024-10-27 12:48:34.141 [update notice] received unknown BC cmd: 37fe200f04c67e0200                     2024-10-27 12:48:34.213 [bus error] send to 18: ERR: SYN received                                       2024-10-27 12:48:34.214 [bus error] send message part 0: ERR: SYN received                              2024-10-27 12:48:34.220 [bus notice] arbitration won in invalid state ready                             2024-10-27 12:48:34.612 [update notice] received unknown BC cmd: 00fe203a0129                           2024-10-27 12:48:34.814 [update notice] received update-read broadcast bus_control QQ=7f:               2024-10-27 12:48:34.851 [update notice] received unknown BC cmd: 37fe2001026573                         2024-10-27 12:48:34.943 [update notice] received unknown BC cmd: 37fe200e0865735e015e013403  

This is the new config for ebusd: 

EBUSD_OPTS="--configpath=/etc/ebusd -d ens:/dev/ttyACM0 --mqttport=1883 --mqtthost=172.10.2.101 --mqttuser ebusd --mqttpass Interbelic123! --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd/%circuit/%name --enablehex --accesslevel=* --latency=80000 --receivetimeout=100000  --acquiretimeout=20000  --acquireretries=10  --sendretries=10 --latency=20"

The CSV file is located under /etc/ebusd.

Kind Regards.

john30, 2024-10-28:
please post some raw logging with "--lograwdata=bytes" inluding at least one wrong symbol or syn received message.
and maybe you want to update the device to the lastet firmware first