john30/ebusd #1367: ERR: end of input reached, vaillant/15.basv2.csv:1: ERR: end of input reached, missing message type/name/pbsb

boskape, 2024-10-26:
### Description

error reading scan config file vaillant/15.basv2.csv for ID "basv2", SW0507, HW1704: ERR: end of input reached, vaillant/15.basv2.csv:1: ERR: end of input reached, missing message type/name/pbsb

since one week i get this error
already replaced csv file
but no change

### Actual behavior

basv device not scanned

### Expected behavior

that all ebusd devices could scan correct

### ebusd version

23.2

### ebusd arguments

 ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=ooxohHei4ta2quee9ke9ahw9Deu1iedoo1iidachei1eepooMefie2ue8AhT8at9 --scanconfig --mqttjson --configpath=/config/ebusd/ --accesslevel=* --mqttint=/config/mqtt-hassio.cfg --mqtttopic=ebusd --device=enh:192.168.1.237:9999 --log=all:notice --mqttvar=filter-direction=r|u|^w --mqttvar=filter-name=

### Operating system

other

### CPU architecture

x64

### Dockerized

other

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

error reading scan config file vaillant/15.basv2.csv for ID "basv2", SW0507, HW1704: ERR: end of input reached, vaillant/15.basv2.csv:1: ERR: end of input reached, missing message type/name/pbsb

john30, 2024-10-27:
maybe a simple access rights issue. check the folder with the csvs (/config/ebusd) and also check the size of the 15.basv2.csv file in there

boskape, 2024-10-27:
when i use as config Path https://cfg.ebusd.eu/
then i works

but i want to use the local copy from JonasPD
and they already working
but since a week i have this strange issue
maybe something on Home Assistant has changed

i check the permissions
all files has the same permissions
and 08.bai.csv is also used without problems

im a little bit lost

chrizzzp, 2024-10-29:
symlink-related issue, workaround in https://github.com/jonesPD/ebusd-configuration/pull/5#issuecomment-2440150854

john30, 2024-10-30:
closing as unrelated