john30/ebusd #1370: MQTT JSON Data is changed

HeikoGr, 2024-10-28:
### Description

I Use ebusd v24.1 in docker with mqtt and JSON content activated.
since the new ebusd Release, the JSON content is different

in FHEM (autcreate complex) there are now new readings:
old: 
`"Flame_0_value"`
new: 
`"Flame_value_value"`

### Actual behavior

new:
```
{
  "value": 
    {"value": "on"}
}
```

### Expected behavior

old:
```
{
  "0": 
    {"name": "","value": "on"}
}
```

### ebusd version

24.1

### ebusd arguments

see logs (it's a multiline docker compose.yml)

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

same as ebusd version

### Hardware interface

Adapter Shield v5 via Ethernet

### Related integration

MQTT generic

### Logs

      EBUSD_MQTTHOST: "192.168.178.221"
      EBUSD_MQTTPORT: 1883
      EBUSD_MQTTCLIENTID: "ebusd"
      EBUSD_MQTTTOPIC: "ebusd/%circuit/%name"
      EBUSD_MQTTGLOBAL: "ebusd/global/"
      EBUSD_MQTTJSON: ""

HeikoGr, 2024-10-28:
don't really know if it's a bug or a change i didn't realized in the changelog.

pascalwinters, 2024-10-28:
For me it's also broken since the update

Kolbi, 2024-10-28:
According to https://github.com/john30/ebusd/issues/1348 it is a change...

pascalwinters, 2024-10-28:
> According to #1348 it is a change...

But that will mean that all existing HA entities are not updated anymore?

pascalwinters, 2024-10-28:
> Exactly and new ones are created... I'm also struggeling with it and also that some (a lot of) entities are not appearing anymore.

Do you mean something like this:
[image]

It's creating a lot of duplicate entities with `_2` in the end.

john30, 2024-10-28:
it is noted as [breaking change in the changelog](https://github.com/john30/ebusd/blob/master/ChangeLog.md#241-2024-10-27) and also noted in the [release notes](https://github.com/john30/ebusd/releases/tag/24.1) that the default for config files is adjusted to the new [reworked typespec based message definitions](https://github.com/john30/ebusd-configuration/tree/rework-normalize/src).
as part of that, the existing definitions went through some necessary normalization and that process is not even finished yet.
btw: not all messages are affected, only those having a single field only as well those that had the weird numbered field names instead of something useful (like the given example of "Flame_0_value")

HeikoGr, 2024-10-28:
thank you for your clarification!
i thought this was a changed url in the first place, since there was no link to the reworked message definitions.

closed 2024-10-28 by HeikoGr

john30, 2024-10-28:
it's there when you follow the links:
[Changelog](https://github.com/john30/ebusd/blob/master/ChangeLog.md#241-2024-10-27) => [ebus.github.io](https://ebus.github.io/) => ["src folder of ebusd-configuration"](https://github.com/john30/ebusd-configuration/tree/master/src)

pixeldoc2000, 2024-11-20:
If you want revert to the "old" json data / behavior, one can change docker tag from `latest` to `john30/ebusd:v23.1` (pin version).