john30/ebusd #1373: Not all messages are parsed.

ulrichsc, 2024-10-30:
### Description

Hi John,
thank you for this awesome projekt.
I try to parse messages for an elco Aerotop Split heatpump. However, some messages are not parsed, even if they are described in the CSV.

e.g.
`0x13fe200e086271e10064002c01`
which should translate to
`QQ:13 ZZ:fe PBSB:200e ID:6271 Data:22.5°C 10°C 30°C`
is defined by
`b,energymgr,elco_z1_daytemp_settings,Day Temp Settings,,fe,200e,6271,daytemp_setting,s,SIN,10,°C,,min_value,s,SIN,10,°C,,max_value,s,SIN,10,°C`
in the CSV.

same for
`0x13fe200e086126f4015e012602`
`QQ:13 ZZ:fe PBSB:200e ID:6126 Data:50°C 35°C 55°C`

No error in the ebusd-logfile.

Am I missing something?

### Actual behavior

Not all messages are parsed as described in the CSV

### Expected behavior

Messages should be parsed according to CSV-File.

### ebusd version

23.3

### ebusd arguments

EBUSD_OPTS="--device ens:172.20.20.159:9999 --configpath=/etc/ebusd/ --mqttport=1883 --mqtttopic=ebusd/%circuit/%name --mqttuser=ebusd --mqttpass=awkelu34 --mqtthost=localhost --loglevel=debug"

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via USB

### Logs

2024-10-30 08:31:42.250 [update info] received BC cmd: 13fe200e086271e10064002c01
2024-10-30 08:31:42.250 [update notice] received unknown BC cmd: 13fe200e086271e10064002c01

john30, 2024-10-30:
maybe check if the message is really loaded by using `ebusctl f -a`

john30, 2024-11-12:
wrong definition in this case as ZZ=broadcast does not work together with fields put into the response of the target (as it will never respond). correct defintion would be this:
```csv
b,energymgr,elco_z1_daytemp_settings,Day Temp Settings,,fe,200e,6271,daytemp_setting,,SIN,10,°C,,min_value,,SIN,10,°C,,max_value,,SIN,10,°C
```

closed 2024-11-12 by john30