john30/ebusd #1376: 24.1 error reading config files

MrBlue90, 2024-11-02:
### Description

Version 24.1.24.1 has issues reading config files.
Reverting to ebusd 23.3.23.3 fixed the issue.

### Actual behavior

Last week I noticed some values were not being received anymore, and I couldn't set values either.

I noticed this in the logs:
`2024-11-02 14:41:47.607 [mqtt error] write message kwl FanSpeed not found`

`ebusctl find` only showed a small list of values, without FanSpeed

Starting ebusd showed the following error message:
`2024-11-02 14:41:47.566 [main error] error reading config files from /config/: ERR: invalid argument, last error: 3c.renovent-excellent-300-english.csv:44: ERR: invalid argument, name EWTT-`

### Expected behavior

No error message about config file while starting.
No error message when setting FanSpeed.
FanSpeed should be listed in `ebusctl find`

```
ebusctl find
kwl ActualSpeedOfExhaustAir = 834
kwl ActualSpeedOfSupplyAir = 882
kwl AirFlowFilterThreshold = 324
kwl AirFlowLevel0 = 50;0;50;50;50
kwl AirFlowLevel1 = 80;50;300;5;100
kwl AirFlowLevel2 = 125;50;300;5;150
kwl AirFlowLevel3 = 250;50;300;5;225
kwl AirflowTotal = 3539000;32768;32767;1;0
kwl AirflowWithFilter = 0
kwl BypassFlow = 0
kwl BypassHysteresis = 2.0;0.0;5.0;0.5;2.0
kwl BypassOperation = no data stored
kwl BypassStatus = Closed
kwl BypassTemperature = 21.0;15.0;35.0;0.5;22.0
kwl CentralHeatingWRG = no data stored
kwl CO2Sensor1Status = 1
kwl CO2Sensor1Value = 0
kwl CO2Sensor2Status = 1
kwl CO2Sensor2Value = 0
kwl CO2Sensor3Status = 1
kwl CO2Sensor3Value = 0
kwl CO2Sensor4Status = 1
kwl CO2Sensor4Value = 0
kwl Contact1Position = 0
kwl Contact2Position = 0
kwl DaysWithFilter = 0
kwl DipswitchValue = 23
kwl eBusSyncError = 180
kwl Errors = 3;104;104;104;104;104;104;104;104;104;104
kwl EWTStatus = Disabled
kwl EWTT+ = 25.0;15.0;40.0;0.5;25.0
kwl EWTT- = 5.0;0.0;10.0;0.5;5.0
kwl ExhaustAirPressure = 8.0
kwl ExhaustAirTemperature = 20.0
kwl ExhaustAirVolume = 80
kwl FanSpeed = Low
kwl FanStatus = Const. Flow
kwl FilterStatus = Clean
kwl FixedPressureImbalance = 0;-100;100;1;0
kwl FrostStatus = No Frost
kwl Humidity = 0.0
kwl IntakeAirPressure = 13.0
kwl IntakeAirVolume = 80
kwl LightingDisplay = no data stored
kwl MoistureIncrease = Sensor Not Active
kwl OperatingHoursTotal = 53955;32768;32767;1;0
kwl OptionTemperature = 0
kwl OutsideAirTemperature = 13.0
kwl PerilexPosition = 1
kwl PostheaterPower = 0
kwl PostheaterStatus = Disabled
kwl PreheaterPower = 0
kwl PreheaterStatus = Disabled
kwl PreheatRegister_inst = -
kwl PressureImbalanceAllowed = no data stored
kwl RHSensorPresent = no data stored
kwl RHSensorSensitivity = 0;-2;2;1;0
kwl SoftwareVersion = S1.06.05 0009
kwl SwitchPosition = 1
kwl TargetExhaustAirVolume = 80
kwl TargetIntakeAirVolume = 80
scan.3c  = ENCON;  ;-;-
```

### ebusd version

24.1

### ebusd arguments

``ebusd -d ens:192.168.20.13:3335 --scanconfig --latency=200000 --mqtthost=172.30.0.2 --mqttport=1883

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

other

### Related integration

MQTT generic

### Logs

2024-11-02 14:41:47.566 [main error] error reading config files from /config/: ERR: invalid argument, last error: 3c.renovent-excellent-300-english.csv:44: ERR: invalid argument, name EWTT-

MrBlue90, 2024-11-02:
@
[3c.renovent-excellent-300-english.csv](https://github.com/user-attachments/files/17607394/3c.renovent-excellent-300-english.csv)

This is the config file which is being used

rastaralle, 2024-11-03:
+1
Same issue here unsing very similar (same?) config file.

jan666, 2024-11-04:
See line 44:

```
w,,EWTT-,408001,,,4080,19,,,UIR,10,°C
w,,EWTT+,408001,,,4080,1A,,,UIR,10,°C
```

I think - and + in names are no longer supported. I had a similar issue with 4Valve... renamed it to FourValve... and it worked again. Try changing it to

```
w,,EWTTPlus,408001,,,4080,19,,,UIR,10,°C
w,,EWTTMinus,408001,,,4080,1A,,,UIR,10,°C
```

rastaralle, 2024-11-04:
Thank you for the hint, Jan! 
I changed the file as proposed and was able to update to dockerversion 24.1 without limitations.

jan666, 2024-11-04:
> Thank you for the hint, Jan! I changed the file as proposed and was able to update to dockerversion 24.1 without limitations.

I made a mistake btw. It should be EWTT**Minus**,408001,,,4080,**19** and w,,EWTT**Plus**,408001,,,4080,**1A**,,,UIR,10,°C

Foradh, 2024-11-06:
Dear all, I have the same problem after switching to version 24. Is there a list of supported characters in csv file?

john30, 2024-11-11:
this is due to the [breaking change listed in the release notes: "change validation of identifiers to no longer accept unusual characters"](https://github.com/john30/ebusd/blob/master/ChangeLog.md#breaking-changes)

closed 2024-11-11 by john30