john30/ebusd #1380: Docker-compose file has no grab command

BelgianGuyWithALaser, 2024-11-06:
### Description

The docker compose file does nog have the grab command. ( entered the container with exec -it ebusd sh )
so sniffing messages to make a new / adjusted CSV is impossible.

running docker compose on a server with ubuntu 24.04.1 LTS

### Actual behavior

issue a grab command so it will logg some unknown messages, this replies sh: grab : not found

### Expected behavior

issue a grab command so it will logg some unknown messages and grab result showing them

### ebusd version

current source from git

### ebusd arguments

.

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

Adapter Shield v5 via Ethernet

### Related integration

other

### Logs

.

jan666, 2024-11-06:
isnt it `ebusctl grab`, not just `grab`?

BelgianGuyWithALaser, 2024-11-06:
You are absolutly right. that worked.

PEBCAK at it's finest! cheers.

closed 2024-11-06 by BelgianGuyWithALaser