john30/ebusd #1381: A better understandable update process

derjohn, 2024-11-06:
### Description

in the Web-UI my system shows:
Build: [20241027](https://adapter.ebusd.eu/v5/ChangeLog#20241027) (on swap partition)

And home assistant says "update avaiable for ebusd".

Is it normal that I  run on the swap partition? If I switch back on the main (?) partition I have 202408xx running, if I press the update button, it fetches the update and goes back to the swap partition. On the main partition is still the old build. How can I bring the lastest release to the non-swap partition ?

wrongisthenewright, 2024-11-09:
> That's totally fine. It switch partition on every update, so if new update fails you can still go to previous version just by switching partition. It was added for better security.

Is it documented on how to swap partition? the adapter does it automatically in the event i can't boot on the new FW? 
Sorry bur after a cursory search I haven't found the info.

derjohn, 2024-11-10:
It's at least confusing on the status page. It says:

Build: [20241027]  (on swap partition)

Does it mean "There is build 20241027 on the swap partiton" ? Or does it mean "You are currently running that build from the swap partion" ? Is that fine or do I have to switch it back to "normal mode" somehow?

IMVHO it should say something like:
```
Version 24.1
main partition (/dev/whatsoever1p1), build 20240899 
swap partition (/dev/whatsoever1p2), build 20241027 (currently active)
```

BTW: I still don't get why I can't swap back to the main partition (old version) and update that to 20241027, too. That's not intended to be done, so keep the user always safe with a rollback?

rgds,
derjohn

john30, 2024-11-11:
its pretty logical IMO:
the app runs in one of two partitions and reports the version currently running as well as the location (swap partition or not) in the status page.
with the two partitions you can swap running to either one (by using the "swap firmware" button in the UI and of course when currently running the main partition, you can only update the swap partition and vice versa.
after each update, the newly flashed partition is started and once it reaches a validation point, it marks itself as such so it will stick to that partition in future (unless the other partition is updated so the process starts fromscratch)

not an ebusd issue anyway, so closing

closed 2024-11-11 by john30

derjohn, 2024-11-11:
Yes it's logical, but confusing as it says nowhere "currently running", instead it says "[found] build 12324 on swap partition". Just Saying .....I personally know now with my Fachwissen how to handle it, others dont.

john30, 2024-11-13:
I don't see where it says "found". you're basically looking at the status page and as such it reports the status of the device, so it reveals the build that is currently running and whether it is on swap or not. since you're the first one to complain about it, I guess the rest is fine with it