john30/ebusd #1382: Home Assistant discovery - entity identifiers changed after update

piotrtobolski, 2024-11-07:
### Description

Casing of device and entity identifiers has changed in ebusd that resulted in:
- duplicated devices and entities in home assistant (old unavailable after mosquitto restart)
- lost configuration applied to old entities (and there is ton of entities tbh)
- lost long term statistics

Environment:
- official Docker image
- very simple config using `/etc/ebusd/mqtt-hassio.cfg`

[image]
[image]
[image]

### Actual behavior

identifiers differ from 23.3 and previous ebusd versions

### Expected behavior

identifiers the same as in previous ebusd versions

### ebusd version

24.1

### ebusd arguments

ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=redacted --scanconfig --mqttjson --latency=10 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:redacted:9999 --log=all:notice

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

same as ebusd version

### Hardware interface

Adapter v3 Ethernet

### Logs

2024-11-07 11:52:16.191 [device notice] transport opened
2024-11-07 11:52:16.192 [main notice] ebusd 24.1.24.1 started with broadcast scan on device: redacted:9999, TCP, enhanced
2024-11-07 11:52:16.595 [bus notice] bus started with own address 31/36
2024-11-07 11:52:16.595 [bus notice] signal acquired
2024-11-07 11:52:16.677 [device notice] reset, supports info
2024-11-07 11:52:16.683 [device notice] extra info: firmware 1.1[8419].2[847f], jumpers 0x37
2024-11-07 11:52:16.701 [mqtt notice] connection established
2024-11-07 11:52:19.773 [bus notice] new master 10, master count 2
2024-11-07 11:52:19.836 [bus notice] new master 03, master count 3
2024-11-07 11:52:19.836 [update notice] received unknown MS cmd: 1008b5110101 / 09363820074153000037
2024-11-07 11:52:20.102 [update notice] received unknown MS cmd: 1008b5100900003affffff000000 / 0101
2024-11-07 11:52:26.601 [main notice] starting initial broadcast scan
2024-11-07 11:52:26.708 [update notice] sent write Broadcast Queryexistence QQ=31: 
2024-11-07 11:52:29.868 [update notice] received unknown MS cmd: 1008b5110101 / 09363820074153000037
2024-11-07 11:52:30.135 [update notice] received unknown MS cmd: 1008b5100900003affffff000000 / 0101
2024-11-07 11:52:30.392 [update notice] received unknown MS cmd: 1008b5110102 / 06033c5a46825a
2024-11-07 11:52:36.897 [bus notice] scan 08: ;Vaillant;BAI00;0205;9602
2024-11-07 11:52:36.897 [update notice] store 08 ident: done
2024-11-07 11:52:36.897 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0205;9602
2024-11-07 11:52:36.897 [bus notice] scan 08: ;Vaillant;BAI00;0205;9602
2024-11-07 11:52:37.314 [main notice] read common config file vaillant/broadcast.csv
[... 136 lines cut ...]
2024-11-07 11:56:30.220 [update notice] sent poll-read 700 Hc3HeatCurve QQ=31: 1.2
2024-11-07 11:56:30.839 [update notice] received read bai Status01 QQ=10: 27.0;28.0;7.125;32.5;41.5;off
2024-11-07 11:56:31.108 [update notice] received update-write bai SetMode QQ=10: auto;29.0;-;-;0;0;0;0;0;0
2024-11-07 11:56:31.359 [update notice] received poll-read bai Status02 QQ=10: auto;60;45.0;70;65.0
2024-11-07 11:56:36.220 [update notice] sent poll-read 700 Hc3HeatCurveAdaption QQ=31: 0.0
2024-11-07 11:56:40.893 [update notice] received read bai Status01 QQ=10: 27.0;28.0;7.125;32.5;41.5;off
2024-11-07 11:56:41.161 [update notice] received update-write bai SetMode QQ=10: auto;29.0;-;-;0;0;0;0;0;0
2024-11-07 11:56:42.220 [update notice] sent poll-read 700 Hc3MaxFlowTempDesired QQ=31: 90
2024-11-07 11:56:48.208 [update notice] sent poll-read 700 Hc3MinFlowTempDesired QQ=31: 15
2024-11-07 11:56:50.944 [update notice] received read bai Status01 QQ=10: 27.0;28.0;7.125;32.5;41.5;off
2024-11-07 11:56:51.210 [update notice] received update-write bai SetMode QQ=10: auto;29.0;-;-;0;0;0;0;0;0
2024-11-07 11:56:54.209 [update notice] sent poll-read 700 Hc3PumpStatus QQ=31: 0
2024-11-07 11:57:00.197 [update notice] sent poll-read 700 Hc3RoomTempSwitchOn QQ=31: off
2024-11-07 11:57:00.947 [update notice] received read bai Status01 QQ=10: 27.0;28.0;7.125;32.5;41.5;off
2024-11-07 11:57:01.214 [update notice] received update-write bai SetMode QQ=10: auto;29.0;-;-;0;0;0;0;0;0
2024-11-07 11:57:01.483 [update notice] received read bai DateTime QQ=10: valid;11:57:03;07.11.2024;7.125
2024-11-07 11:57:01.738 [update notice] received poll-read bai Status02 QQ=10: auto;60;45.0;70;65.0
2024-11-07 11:57:01.974 [update notice] received update-read Broadcast Vdatetime QQ=10: 11:57:01;07.11.2024
2024-11-07 11:57:02.238 [update notice] received unknown MS cmd: 1008b5110100 / 08b601110000000081
2024-11-07 11:57:02.453 [update notice] received update-read Broadcast Outsidetemp QQ=10: 7.125
2024-11-07 11:57:06.230 [update notice] sent poll-read 700 Hc3Status QQ=31: 0
2024-11-07 11:57:10.992 [update notice] received read bai Status01 QQ=10: 27.0;28.0;7.125;32.5;41.5;off
2024-11-07 11:57:11.260 [update notice] received update-write bai SetMode QQ=10: auto;29.0;-;-;0;0;0;0;0;0
2024-11-07 11:57:12.237 [update notice] sent poll-read 700 Hc3SummerTempLimit QQ=31: 21
2024-11-07 11:57:14.188 [main notice] scan completed 4 time(s), check again

Commifreak, 2024-11-10:
Yea, same here:

[image]
[image]

I updated my mqtt-hass config with the recent changes as well.

Not sure if this is a bug, feature or mqtt-hass thing.

john30, 2024-11-11:
this is due to the [breaking change listed in the release notes: "change default config path to https://ebus.github.io/ ..."](https://github.com/john30/ebusd/blob/master/ChangeLog.md#breaking-changes)
this is part of the normalization recently applied to the [ebusd-configuration files](https://github.com/john30/ebusd-configuration/tree/rework-csvgen/src).
if you want the old ones, then switch to local files using the repo folder ebusd-2.1.x directly instead.

closed 2024-11-11 by john30

Commifreak, 2024-11-11:
Is there any option to migrate the HASS history then? I have some historical temps which are now seperated. Yes, thats A hass thing but an info for ebusd/hass users if and how this could be migrated.

piotrtobolski, 2024-11-11:
I think I will embed old configurations in my addon and let users choose the old behavior along with some warning that it won't receive any updates. That should work, right @john30?

john30, 2024-11-11:
> Is there any option to migrate the HASS history then?

no idea but I guess that should be possible somehow

Commifreak, 2024-11-14:
I dont know if this is related to this, but..

Due to the breaking change the entity names changed - yes. I noticed that the suffixes went away (like tempv, press, pressv and so on). but the flowtemp is still displayed as `ebusd_bai_flowtemp_temp` when other suffixes went away. It seems that only the `temp` suffix is still there.

Is this correct?

john30, 2024-11-15:
the extra suffix is only dropped for messages having a single field only as in this case it does not have any benefit at all. if a message has more than one field, the name suffix is kept