john30/ebusd #1383: ebusd service does not start after restart PI 5

Norb1204, 2024-11-08:
### Description

After restarting the raspberry pi 5, ebusd service do not start.
[image]

### Actual behavior

Nothing to preview

### Expected behavior

Nothing to preview

### ebusd version

24.1

### ebusd arguments

EBUSD_OPTS="--scanconfig=full --configpath=/etc/ebusd/ -d ens:/dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_DC:54:75:63:64:C4-if00 -p 8888 -l /var/log/ebusd.log --loglevel=error --logareas=network  --httpport=8889 --htmlpath=/var/ebusd/html --accesslevel=install --updatecheck=off"

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via USB

### Logs

Nothing to preview

Norb1204, 2024-11-08:
I still have an old ebus 3 adapter, this still works.
The Shield v5 adapter may have a problem after I tried to download the latest software version (2024-10-27). Going back to the old software (version 2024-01-06) is now the same problem.
Maybe the adapter is broken now?

Norb1204, 2024-11-09:
I am one step closer to finding the error:
If I delete the characters of the USB path after ...DC:54, I can start the ebusd service.

**ebus service starts:**
EBUSD_OPTS="--scanconfig=full --configpath=/etc/ebusd/ -d ens:/dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_DC:54<s>:75:63:64:C4-if00</s> -p 8888 -l /var/log/ebusd.log --loglevel=error --logareas=network --httpport=8889 --htmlpath=/var/ebusd/html --accesslevel=install --updatecheck=off"

**ebus service cannot be started:**
EBUSD_OPTS="--scanconfig=full --configpath=/etc/ebusd/ -d ens:/dev/serial/by-id/usb-Espressif_USB_JTAG_serial_debug_unit_DC:54:75:63:64:C4-if00  -p 8888 -l /var/log/ebusd.log --loglevel=error --logareas=network --httpport=8889 --htmlpath=/var/ebusd/html --accesslevel=install --updatecheck=off"

john30, 2024-11-11:
duplicate of #1374

closed 2024-11-11 by john30