john30/ebusd #1384: No writable entities in Homeassistant

Pavel-Demidyuk, 2024-11-08:
### Description

I run ebusd as an addon in Supervised HA, and it can discover my Vaillant boiler and bunch of sensors, but there is no writable entities for my boiler, even though I added 
```
--mqttvar=filter-direction=r|u|^w
```
to the config

### Actual behavior

A lot of sensors are successfully displayed on the main page of HA, however there is nothing I can write to.
```
ebusctl find -w
broadcast id = no data stored
broadcast queryexistence = no data stored
broadcast vdatetime = no data stored
```

### Expected behavior

Some writable entities should be presented

### ebusd version

current source from git

### ebusd arguments

--foreground --mqtthost=192.168.50.87 --mqttport=1883 --mqttuser=ebusd --mqttpass=ebusd --scanconfig --mqttjson --configpath=/config/ebusd-configuration/latest/en/ --mqttint=/config/ebusd-configuration/latest/en/mqtt.cfg --mqtttopic=ebusd --device=enh:192.168.50.106:9999 --log=all:notice --log=main:notice --log=bus:notice --log=update:notice --log=network:notice --log=other:notice --mqttvar=filter-direction=r|u|^w

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

same as ebusd version

### Hardware interface

Adapter Shield v5 via Ethernet

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=192.168.50.87 --mqttport=1883 --mqttuser=ebusd --mqttpass=ebusd --scanconfig --mqttjson --configpath=/config/ebusd-configuration/latest/en/ --mqttint=/config/ebusd-configuration/latest/en/mqtt.cfg --mqtttopic=ebusd --device=enh:192.168.50.106:9999 --log=all:notice --log=main:notice --log=bus:notice --log=update:notice --log=network:notice --log=other:notice --mqttvar=filter-direction=r|u|^w
2024-11-08 18:38:36.023 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.50.106:9999, enhanced
2024-11-08 18:38:36.066 [bus notice] device status: resetting
2024-11-08 18:38:36.067 [bus notice] bus started with own address 31/36
2024-11-08 18:38:36.069 [bus notice] device status: reset, supports info
2024-11-08 18:38:36.070 [bus notice] device status: extra info: firmware 1.1[4505].1[4505], jumpers 0x07
2024-11-08 18:38:36.087 [bus notice] signal acquired
2024-11-08 18:38:36.140 [mqtt notice] connection established
2024-11-08 18:38:36.187 [mqtt error] write message basv StorageTempDesired not found
2024-11-08 18:38:36.187 [mqtt error] write message bai StorageTempDesired not found
2024-11-08 18:39:39.579 [bus notice] new master 03, master count 2
2024-11-08 18:39:46.246 [bus notice] scan 08: ;Vaillant;BAI00;0109;7503
2024-11-08 18:39:46.246 [update notice] store 08 ident: done
2024-11-08 18:39:46.246 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0109;7503
2024-11-08 18:39:46.246 [bus notice] scan 08: ;Vaillant;BAI00;0109;7503
2024-11-08 18:39:46.281 [main notice] read common config file vaillant/broadcast.csv
2024-11-08 18:39:46.284 [main notice] read common config file vaillant/general.csv
2024-11-08 18:39:46.288 [main notice] read common config file vaillant/scan.csv
2024-11-08 18:39:46.295 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0109, HW7503
2024-11-08 18:39:46.444 [update notice] sent scan-read scan.08 id QQ=31: 
2024-11-08 18:39:46.599 [update notice] sent scan-read scan.08 id QQ=31: 
2024-11-08 18:39:46.749 [update notice] sent scan-read scan.08 id QQ=31: 
2024-11-08 18:39:46.902 [update notice] sent scan-read scan.08 id QQ=31: 21;22;48;0010023658;3100;005560;N7
2024-11-08 18:39:46.952 [main notice] found messages: 224 (3 conditional on 28 conditions, 0 poll, 10 update)
2024-11-08 18:39:47.038 [bus notice] max. symbols per second: 116
2024-11-08 18:39:47.143 [update notice] sent scan-read scan.08 id QQ=31: 21;22;48;0010023658;3100;005560;N7
2024-11-08 18:39:47.300 [update notice] sent scan-read scan.08 id QQ=31: 21;22;48;0010023658;3100;005560;N7
2024-11-08 18:39:47.452 [update notice] sent scan-read scan.08 id QQ=31: 21;22;48;0010023658;3100;005560;N7
2024-11-08 18:39:47.604 [update notice] sent scan-read scan.08 id QQ=31: 21;22;48;0010023658;3100;005560;N7
2024-11-08 18:39:47.604 [bus notice] scan 08: ;21;22;48;0010023658;3100;005560;N7
2024-11-08 18:39:56.780 [update notice] sent poll-read bai CounterStartattempts1 QQ=31: 0
2024-11-08 18:40:02.146 [update notice] sent poll-read bai CounterStartattempts2 QQ=31: 0
2024-11-08 18:40:08.168 [update notice] sent poll-read bai CounterStartAttempts3 QQ=31: 0
2024-11-08 18:40:14.143 [update notice] sent poll-read bai CounterStartAttempts4 QQ=31: 0
2024-11-08 18:40:20.162 [update notice] sent poll-read bai DeactivationsTemplimiter QQ=31: 0
2024-11-08 18:40:26.141 [update notice] sent poll-read bai DisplayMode QQ=31: 2
2024-11-08 18:40:32.183 [update notice] sent poll-read bai expertlevel_ReturnTemp QQ=31: -1.81;cutoff
2024-11-08 18:40:38.163 [update error] unable to parse poll-read bai externalFlowTempDesired from 3108b509030d2500 / 00: ERR: invalid position
2024-11-08 18:40:39.874 [mqtt error] decode bai externalFlowTempDesired: ERR: invalid position
2024-11-08 18:40:44.136 [update error] unable to parse poll-read bai ExtFlowTempDesiredMin from 3108b509030d6e04 / 00: ERR: invalid position
2024-11-08 18:40:49.887 [mqtt error] decode bai ExtFlowTempDesiredMin: ERR: invalid position
2024-11-08 18:40:50.159 [update notice] sent poll-read bai FanHours QQ=31: -
2024-11-08 18:40:56.146 [update notice] sent poll-read bai FanStarts QQ=31: -
2024-11-08 18:41:02.175 [update notice] sent poll-read bai FlowTemp QQ=31: 35.75;ok
2024-11-08 18:41:08.156 [update notice] sent poll-read bai FlowTempDesired QQ=31: 37.00
2024-11-08 18:41:14.139 [update notice] sent poll-read bai FlowTempMax QQ=31: 0.00
2024-11-08 18:41:20.166 [update notice] sent poll-read bai HcHours QQ=31: 0
2024-11-08 18:41:26.141 [update error] unable to parse poll-read bai HcPumpMode from 3108b509030d4904 / 00: ERR: invalid position
2024-11-08 18:41:29.942 [mqtt error] decode bai HcPumpMode: ERR: invalid position
2024-11-08 18:41:32.167 [update notice] sent poll-read bai HcPumpStarts QQ=31: -
2024-11-08 18:41:38.149 [update notice] sent poll-read bai HcStarts QQ=31: 0
2024-11-08 18:41:44.169 [update notice] sent poll-read bai HcUnderHundredStarts QQ=31: 0
2024-11-08 18:41:50.152 [update error] unable to parse poll-read bai HoursTillService from 3108b509030d2004 / 00: ERR: invalid position
2024-11-08 18:41:50.606 [main notice] update check: version 24.1 available, device firmware 1[4a1b] available, broadcast.csv: different version available, memory.csv: different version available, vaillant/bai.308523.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: different version available
2024-11-08 18:41:55.615 [mqtt error] decode bai HoursTillService: ERR: invalid position
2024-11-08 18:41:56.173 [update notice] sent poll-read bai HwcHours QQ=31: 0
2024-11-08 18:42:02.148 [update error] unable to parse poll-read bai HwcPostrunTime from 3108b509030d1104 / 00: ERR: invalid position
2024-11-08 18:42:05.628 [mqtt error] decode bai HwcPostrunTime: ERR: invalid position
2024-11-08 18:42:08.136 [update notice] sent poll-read bai HwcStarts QQ=31: 0
2024-11-08 18:42:14.144 [update notice] sent poll-read bai HwcTemp QQ=31: 0.00;ok
2024-11-08 18:42:20.470 [update error] unable to parse poll-read bai HwcTempDesired from 3108b509030dea03 / 00: ERR: invalid position
2024-11-08 18:42:25.655 [mqtt error] decode bai HwcTempDesired: ERR: invalid position
2024-11-08 18:42:26.160 [update error] unable to parse poll-read bai HwcTempMax from 3108b509030d4304 / 00: ERR: invalid position
2024-11-08 18:42:30.660 [mqtt error] decode bai HwcTempMax: ERR: invalid position
2024-11-08 18:42:32.140 [update notice] sent poll-read bai HwcUnderHundredStarts QQ=31: 0
2024-11-08 18:42:38.161 [update notice] sent poll-read bai maintenancedata_HwcTempMax QQ=31: 0.00
2024-11-08 18:42:44.139 [update notice] sent poll-read bai ModulationTempDesired QQ=31: 100.0

john30, 2024-11-11:
check the discovery messages sent to HA using e.g. `mosquitto_sub -v -t 'homeassistant/#'` and check if there is really no write message in it

mansellrace, 2024-11-12:
Same for me, I have a vaillant that is recognised as bai 08.
I have configured the ability to write, but on home assistant I only see entities that cannot be edited

[image]

john30, 2024-11-13:
did you add the writable direction to the filter? and was that escaped correctly (as the pipe symbol is special at least on the command line)?

mansellrace, 2024-11-13:
> did you add the writable direction to the filter? and was that escaped correctly (as the pipe symbol is special at least on the command line)?

Hi! 
Yesterday I studied the way configuration files work and I found that for my model there is no specific configuration file, and the generic one is loaded that it has no writable entities

I have a 0010019985 Vaillant

As soon as I have some free time I would like to try to understand if one of the specific ones for other models is suitable for my boiler
Is this the best way to proceed? Is there any command or procedure to understand what data can I extract from the boiler? Or is it just attempts?

Thanks for your project!

Kolbi, 2024-11-13:
You are some steps ahead of me (using a 10021924 ) :) I'm currently at the same point not exactly knowing how to determine infos for creating a special file.

JonesGW-MGD, 2024-11-13:
@mansellrace - many us are able to write values even using the default bai file (bai308523.inc) and the associated controller devices - so I suspect it's not an actual 'issue' with the ebusd application or the configuration files, but perhaps your set-up?  You might find if you raise a discussion item rather than an issue you get some help - I'd be happy to share my experience for example.  (Some of which I documented [here](https://github.com/john30/ebusd/discussions/1177)  to write values as mundane as MaintenanceDate and as useful as HeatCurve using ebusd and HA 'out of the box'

Pavel-Demidyuk, 2024-11-30:
> check the discovery messages sent to HA using e.g. `mosquitto_sub -v -t 'homeassistant/#'` and check if there is really no write message in it

Seems like there are only sensors and binary sensors, se below. I'm really stuck here, arguments seem to be valid, which is reflected in startup logs
```
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=192.168.50.87 --mqttport=1883 --mqttuser=ebusd --mqttpass=ebusd --scanconfig --mqttjson --configpath=/config/ebusd-configuration/latest/en/ --mqttint=/config/ebusd-configuration/latest/en/mqtt.cfg --mqtttopic=ebusd --device=enh:192.168.50.106:9999 --log=all:notice --log=main:notice --log=bus:notice --log=update:notice --log=network:notice --log=other:notice --mqttvar=filter-direction=r|u|^w
```

I've also cloned config locally and uncommented 
```
# include only messages having the specified direction ("r", "w", "u", or "uw". partial match, alternatives and wildcard supported).
# HA integration: writable messages excluded for now
#filter-direction = r|u
# HA integration: for including writable messages, use this line or overwrite with '--mqttvar=filter-direction=r|u|^w'.
filter-direction = r|u|^w
```

but still no luck! here is mqtt output

```
$ mosquitto_sub -v -t 'homeassistant/#'

homeassistant/binary_sensor/ebusd_global_running/config {
  "unique_id":"ebusd_global_running",
  "device":{
   "identifiers":"ebusd",
   "manufacturer":"ebusd.eu",
   "name":"ebusd",
   "sw_version":"23.2",
   "suggested_area":"Heating"
  },
  "state_topic":"ebusd/global/running",
  "name":"global running",
  "device_class":"running",
  "payload_on":"true",
  "payload_off":"false"
 }
homeassistant/binary_sensor/ebusd_global_signal/config {
  "unique_id":"ebusd_global_signal",
  "device":{
   "identifiers":"ebusd",
   "manufacturer":"ebusd.eu",
   "name":"ebusd",
   "sw_version":"23.2",
   "suggested_area":"Heating"
  },
  "state_topic":"ebusd/global/signal",
  "name":"global signal",
  "device_class":"connectivity",
  "payload_on":"true",
  "payload_off":"false"
 }
homeassistant/sensor/ebusd_global_uptime/config {
  "unique_id":"ebusd_global_uptime",
  "device":{
   "identifiers":"ebusd",
   "manufacturer":"ebusd.eu",
   "name":"ebusd",
   "sw_version":"23.2",
   "suggested_area":"Heating"
  },
  "state_topic":"ebusd/global/uptime",
  "name":"global uptime",
  "device_class":"duration",
  "state_class":"total_increasing",
  "unit_of_measurement":"s"
 }
homeassistant/update/ebusd_global_updatecheck/config {
  "unique_id":"ebusd_global_updatecheck",
  "device":{
[... 307 lines cut ...]
   "suggested_area":"Heating"
  },
  "value_template":"{{value_json[\"hoursum2\"].value}}",
  "state_topic":"ebusd/bai/HcHours",
  "unit_of_measurement":"h",
  "state_class":"total_increasing"
 }
homeassistant/sensor/ebusd_bai_HcPumpMode_0/config {
  "unique_id":"ebusd_bai_HcPumpMode_0",
  "name":"HcPumpMode ",
  "device":{
   "identifiers":"ebusd_bai",
   "manufacturer":"ebusd.eu",
   "name":"ebusd bai",
   "via_device":"ebusd",
   "sw_version":"23.2",
   "suggested_area":"Heating"
  },
  "value_template":"{{value_json[\"0\"].value}}",
  "state_topic":"ebusd/bai/HcPumpMode",
  "device_class":"enum",
  "options":["post run","permanent","winter"]
 }

```

john30, 2024-12-01:
if you're using supervised addon, why do you start ebusd in foreground in the first place? this would interfere with another running instance i guess, so please check that

john30, 2024-12-14:
did you change anything else in the mqtt.cfg file used? maybe just post it here, so we're on the same page

tomasz-kos, 2025-01-06:
Hello, I have exactly the same situation, tried different things: filter-direction in HA UI, copy and edit mqtt-hassio.cfg nothing helped. Last time I tried in my bai config file for few line change type from "wi" to "w" and that line with "w" type was discovered by HA as writable and writing functionality is working. I need to search more about difference in type definition as I'm completely beginner in ebus.

Koky05, 2025-01-09:
If change from "wi" to "w" works for you you need to add `--accesslevel="*"` into ebusd start command it start to listen also installer values.

tomasz-kos, 2025-01-09:
My ebusd already have that:
Log from HA addon
`
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=#### --scanconfig --mqttjson --mqttlog --configpath=/config/ebusd-configuration-master/ebusd-2.1.x/en --accesslevel=* --pollinterval=1 --mqttint=/config/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.122.223:9999 --log=all:notice --log=main:notice --log=bus:notice --log=update:notice --log=network:notice --log=other:notice
2025-01-09 16:29:49.132 [device notice] transport opened
`
and it doesn't work.
Both "w" and "wi" are visible in HA but only "w" as writable and "wi" as read only.

tomasz-kos, 2025-01-10:
I did all that already even completely remove ebusd devices from HA and reinstalling ebusd addon. Nothing helps. MQTT Explorer shows that HA showing correctly what MQTT config messages are send by ebusd. In my system config messages for "wi" are created differently than for "w" so I'm trying to understand source code which doing that.

tomasz-kos, 2025-01-11:
After some investigation it is even more complicated. If i have message defined as "wi" non of them is shown as writable in HA. When I changed 2 the most interesting for me from "wi" to "w" they both shows as writable (based on that were my last comments). But when I change all "wi" to "w" only 2 more become writable (from 11 "w" messages). All are visible in HA but only 4 writable

My bai config
```
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment
#,BAI00,Home Regular,308523 174,,,,,,,,,,
*r,,,,,,"B509","0D",,,,,,
*w,,,,,,"B509","0E",,,,,,
*wi,#install,,,,,"B509","0E",,,,,,
*ws,#service,,,,,"B509","0E",,,,,,
*[SW],scan,,,SW,,,,,,,,,
# ##### dia level 1 #####,,,,,,,,,,,,,
r;w,,BlockTimeHcMax,d.02 Max blocking time CH,,,,"2100",,,minutes0,,,max. burner anti cycling period at 20°C Flow temperature setpoint
r1,,ACRoomthermostat,d.08 Room thermostat 230 V,,,,"2A00",,,onoff,,,External controls heat demand (Clamp 3-4)
r;w,,HwcTempMax,d.20 Maximum DHW temperature,,,,"D900",,,temp,,,Adjustment of maximum tank setpoint range (maximum endposition of potmeter)
r1,,FlowTemp,d.40 TFT_DK,,,,"1800",,,tempsensor,,,flow temperature
r1,,ReturnTemp,d.41 Temp heating return,,,,"9800",,,tempmirrorsensor,,,return temperature
r;w,,HeatCurve,d.43 Heating curve,,,,"3500",,,curve,,,heating curve
r;w,,FanSpeedOffsetMin,d.50 MinFanSpeedOffset,,,,"A700",,,SIN,,1/min,Offset for minimum fan speed
r;w,,FanSpeedOffsetMax,d.51 MaxFanSpeedOffset,,,,"A800",,,SIN,,1/min,Offset for maximum fan speed
r;w,,FlowsetHcMax,d.71 Max value flow temp,,,,"A500",,,temp,,,Adjustment of maximum central heating setpoint range (maximum endposition of potmeter)
r;w,,FlowsetHwcMax,d.78 Max value flow temp storage,,,,"A600",,,temp,,,storage charging temperature limit (target flow temperature in storage mode)
r1,,ModulationTempDesired,Modulationsetpoint_DK,,,,"2E00",,,SIN,10,%,Modulation set point
r1,,Flame,Flame,,,,"0500",,,UCH,240=off;15=on,,flame
r;w,,PrEnergySumHwc1,PrEnergySumDHW1_DK,,,,"C500",,,ULG,,,Predictive Maintenance data
r;w,,PrEnergyCountHwc1,PrEnergyCountDHW1_DK,,,,"C600",,,ULG,,,Predictive Maintenance data
r;w,,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F500",,,ULG,,,Predictive Maintenance data
r;w,,PrEnergyCountHc1,PrEnergyCountCH1_DK,,,,"F600",,,ULG,,,Predictive Maintenance data
!include,errors.inc,,,,,,,,,,,,
```
Writable are HwcTempMax, HeatCurve, FlowsetHcMax, FlowsetHwcMax
Maybe it is connected with unit and parsing that?

tomasz-kos, 2025-01-11:
It is connected with unit
When I change:
`r;w,,PrEnergyCountHwc1,PrEnergyCountDHW1_DK,,,,"C600",,,ULG,,,Predictive Maintenance data`
to 
`r;w,,PrEnergyCountHwc1,PrEnergyCountDHW1_DK,,,,"C600",,,temp,,,Predictive Maintenance data`
After restart of ebusd PrEnergyCountHwc1 shows up in HA - with wrong value of course.

john30, 2025-01-21:
then your filters in mqtt-hassio.cfg exclude it. when you want to have messages with access level available, you need to set up ACL

tomasz-kos, 2025-01-22:
About filter in mqtt-hassio.cfg. I assume the problem is with ` type_switch-w-number =`. For `type_switch-number =` is line `sensor,,total_increasing = poweron|count,|countvalue,` but for ` type_switch-w-number =` there isn't. I tried to copy and modified, but so far without success.

For the comment about ACL. I assume it is about "w" and "wi" difference. I was assuming `--accesslevel=*` should solve that. In some comment I seen that ` --accesslevel=install` did the work, but not for me. If i need something else could you point where should I search for information.

john30, 2025-01-25:
did you try with accesslevel "*"?

tomasz-kos, 2025-01-27:
As above log shows I'm starting it with ` --accesslevel=* ` is it what you mean or should I try set accesslevel "*" somewhere else to try?

AbsurdSyssie, 2025-02-10:
I have a smiliar issue with 08.bai - able to see many sensors but no controls.
I am using a shield C6
Docker containers for MQTT, HA, ebusd - they can all see each other
I have the filters and access level as described above
I am able to get and see a writable command flowtempdesired, but when I try to use MQTT to set it
```
topic: ebusd/bai/flowtempdesired
payload: 60.00
```
(i have tried variations on the payload such as in a json)

 ebusd tells me:

```
2025-02-10 17:01:48.847 [mqtt error] write message bai FlowTempDesired not found
```
I followed [JonesGW-MGD](https://github.com/JonesGW-MGD) guide as closely as I could in a docker instance for HA but got nowhere.

I appreciate any feedback on this as I cannot seem to get anywhere! TIA 

Info: 

## docker config
ebusd:
    image: john30/ebusd
    container_name: ebusd
    ports:
      - "3005:8888" # Map port 3005 on the host to port 8888 in the container
    command: "--scanconfig -d ens:192.168.1.175:9999 --accesslevel=* --latency=10 --mqtthost=mosquitto --mqttport=1883 --mqtttopic=ebusd --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttjson --mqttvar=filter-direction=r|u|^w"
    restart: unless-stopped # Ensure the container restarts unless manually stopped

## HA config
ebusd:
  host: ebusd
  port: 3005
  circuit: bai

## ebusctl info
version: ebusd 24.1.24.1
device: 192.168.1.175:9999, TCP, enhanced, firmware 1.1[5114].1[5114]
access: *
signal: acquired
symbol rate: 22
max symbol rate: 40
min arbitration micros: 4
max arbitration micros: 4
min symbol latency: 6
max symbol latency: 22
scan: finished
reconnects: 0
masters: 2
messages: 18
conditional: 0
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401"
address 31: master #8, ebusd
address 36: slave #8, ebusd

JonesGW-MGD, 2025-02-10:
Hi @AbsurdSyssie - if you take a couple of steps back - have you tried issuing a write command directly from `ebusctl` on a terminal session and if so does that work?  I've always found (and possibly because I'm not a techy) that if you start at the top of the stream and work back downstream, things become clearer!

It's a slightly different question but what is your use-case for setting `flowtempdesired`?  

ebusd is an immensely powerful tool but you really want to avoid 'fighting Vaillant' - by which I mean (and it will depend on your particular configuration) - if I tried to do this, I'd be overriding the weather-compensation features - My boiler takes the outside temp, feeds that and the desired room / zone temp to a heat curve and determine the flow temp.  I really don't want to be messing with this.  

Even if you don't have weather comp - your controller will likely be telling the boiler what to set the desired flow temp to.  And it will do this all the time - like every so many minutes (opinions vary) by using the 'SetMode' command so if you want to override the controller, you need to 'flood' ebusd more frequently with your settings.  Even if your controller is totally 'dumb' like a non-Vaillant control, (Hive / Nest, etc.) that is basically powering the boiler on/off  - it's still possible that you'll need to continuously 'instruct' the boiler.

JonesGW-MGD, 2025-02-10:
@AbsurdSyssie - when ebusd starts - can you confirm what bai file it loads?  (just run - `ebusctl i`)

AbsurdSyssie, 2025-02-10:
```
ebusctl i
version: ebusd 24.1.24.1
update check: OK, device firmware v5-c6 1[4c15] available
device: 192.168.1.175:9999, TCP, enhanced, invalid
access: *
signal: no signal
reconnects: 1
masters: 3
messages: 234
conditional: 3
poll: 71
update: 16
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.308523.inc" ([id_hw=7401]), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=05000;SW=0113;HW=9601"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

The thermostat is a ```VRT 50```
and the switch is a ```Time Switch 150```

The boiler is a ```ecoTEC pro 28```

JonesGW-MGD, 2025-02-10:
@AbsurdSyssie - so you're loading the boiler message file bai.308523.inc which one could regard as the 'default' file if no other more specific file is discovered based on your boiler product ID, SW and HW versions etc.

If I assume, because you are on version 24.1, you are taking your message definitions from the new repository here - https://github.com/eBUS/ebus.github.io/blob/main/en/vaillant/bai.308523.inc

Then, line 16 of this file is:

`r,,,FlowTempDesired,d.05 flow/return setpoint: Flow temperature target value or return target value when return regulation is set,,,b509,0d3900,value,,D2C,,°C,temperature`

so it's set currently as not writable.

you now have some options depending on your appetite! (but if you replicate the files locally - set a --configpath to point to these, you can then experiment with making them writeable for example.)

AbsurdSyssie, 2025-02-10:
I see, thanks a lot!
So if I set this to writeable, then sent an mqtt message /ebus/bai/flowtempdesired payload 60.00 - this should in theory tell the boiler to aim for that?

I have had a look at bai.308523.inc and I can see a couple of values for AC/DC thermostats.
I suppose I could play with these and see how it behaves. This seems more like the intended use. 

I am assuming I would send an MQTT message /ebusd/bai/DCthermostat payload 1

Will let you know how it goes! I am very appreciative of your help!

AbsurdSyssie, 2025-02-10:
Thanks for all your help again JonesGW.

Could I quickly ask about how one goes about using the setmode command, I can't find any comprehensive guide on it. 

ebusctl logs look like this when the VRT is on;
```
2025-02-07 16:03:06.276 [update notice] received update-write bai SetMode QQ=10: auto;33.5;-;-;0;0;1;0;0;0

2025-02-07 16:03:09.171 [update notice] sent poll-read bai HcUnderHundredStarts QQ=31: 0

2025-02-07 16:03:11.132 [update notice] received poll-read bai Status01 QQ=10: 48.0;47.0;-;-;37.0;off
```

The MQTT broker sees this;

```
Message 106 received on ebusd/bai/SetMode at 16:01:

{

"hcmode": {

"value": "auto"

},

"flowtempdesired": {

"value": 90

},

"hwctempdesired": {

"value": null

},

"hwcflowtempdesired": {

"value": null

},

"disablehc": {

"value": 0

},

"disablehwctapping": {

"value": 0

},

"disablehwcload": {

"value": 1

},

"remotecontrolhcpump": {

"value": 0

},

"releasebackup": {

"value": 0

},

"releasecooling": {

"value": 0

}

}

{

"hwcmode": {

"value": "auto"

},

"temp": {

"value": 60

},

"temp_1": {

"value": 55

},

"temp_2": {

"value": 70

},

"temp_3": {

"value": 62

}

}

```

john30, 2025-04-13:
closing as not related to ebusd but configuration issues

closed 2025-04-13 by john30