john30/ebusd #1388: Some readings from Vaillant dissapeared...

rande83, 2024-11-17:
### Description

I read out my Vaillant heating system with FHEM & ioBroker, log the readings and then visualize the whole thing in graphs.
Now I've had the problem that some of my readings are no longer appearing since November 4th at 5am.

What does not work: the readings for: Kollektortemperatur, Vor- & Rücklauf der Heizung, as well as the Solarspeichertemperatur
What is working: the readings for: Aussentemperatur, Heizkurve, aktuelle Leistung der Pumpe, Wasserdruck & Speichertemperatur

As you can see, not all readings have disappeared. I have not yet found a cause for this. It seems that the missing readings are simply no longer being read through my adapter. I can't get the readings via mqtt either.
However, the hardware of the heating system is OK. I can display all values in the actuator/sensor test.

Does anyone have any idea how I can fix this?

Thanks!

Example: FHEM-Logtext from Kollektortemperatur
`
2024-11-04_04:59:28 Kollektortemperatur Kollektortemperatur:   5.6
2024-11-04_04:59:28 Kollektortemperatur Kollektortemperatur   5.6
2024-11-04_05:01:26 Kollektortemperatur Kollektortemperatur:   0.0
2024-11-04_05:01:26 Kollektortemperatur Kollektortemperatur   0.0
`

### Actual behavior

Above mentioned readings are missing

### Expected behavior

Want the readings back.

### ebusd version

24.1

### ebusd arguments

EBUSD_OPTS="-d mdns:dc5475****** -p 8888 -l /var/log/ebusd.log --scanconfig --latency=20 --httpport=8080 --mqttport=1883 --mqttjson --mqtthost=localhost --mqttuser=pi --mqttpass=************* --mqtttopic=SmartHome/Haus/Hausraum/Vaillant/%circuit/%name"

### Operating system

Debian 11 (Bullseye) / Ubuntu 20-21 / Raspbian 11 / Raspberry Pi OS 11 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

HTTP, other

### Logs

2024-11-17 16:27:42.178 [device notice] transport opened
2024-11-17 16:27:42.181 [main notice] ebusd 24.1.24.1 started with broadcast scan on device: 192.168.178.44, TCP, enhanced
2024-11-17 16:27:42.481 [bus notice] bus started with own address 31/36
2024-11-17 16:27:42.481 [device notice] reset, supports info
2024-11-17 16:27:42.481 [bus notice] signal acquired
2024-11-17 16:27:42.482 [mqtt notice] connection established
2024-11-17 16:27:42.484 [device notice] extra info: firmware 1.1[4a1b].1[4a1b], jumpers 0x0b
2024-11-17 16:27:50.379 [bus notice] new master 10, master count 2
2024-11-17 16:27:50.440 [bus notice] new master 03, master count 3
[... 132 lines cut ...]
2024-11-17 16:29:34.341 [update notice] received unknown MS cmd: 1008b5120204ff / 0101
2024-11-17 16:29:34.584 [update notice] received unknown MS cmd: 1008b513020508 / 00
2024-11-17 16:29:34.796 [update notice] received unknown BC cmd: 10feb510020601
2024-11-17 16:29:41.026 [update notice] received read bai Status01 QQ=10: 27.0;26.5;8.188;-;-;off
2024-11-17 16:29:41.324 [update notice] received read vr_71 SensorData1 QQ=10: -;-;-;-;38.12;18.94;8.38;55 00
2024-11-17 16:29:41.620 [update notice] received read vr_71 SensorData2 QQ=10: 16.88;-;-;-;-895.81;0.00;80 54 04
2024-11-17 16:29:41.891 [update notice] received update-write bai SetMode QQ=10: auto;27.5;-;-;0;0;1;0;0;0
2024-11-17 16:29:42.610 [update notice] received write vr_71 SetActorState QQ=10: -;off;off;off;off;off;-;-;off;off;off;off;00 00
2024-11-17 16:29:42.859 [update notice] received write vr_71 Mc1FlowTempDesired QQ=10: on;27.5;on;-100
2024-11-17 16:29:51.079 [update notice] received read bai Status01 QQ=10: 26.5;26.5;8.188;-;-;off
2024-11-17 16:29:51.378 [update notice] received read vr_71 SensorData1 QQ=10: -;-;-;-;38.12;18.94;8.38;55 00
2024-11-17 16:29:51.672 [update notice] received read vr_71 SensorData2 QQ=10: 16.88;-;-;-;-895.81;0.00;80 54 04
2024-11-17 16:29:51.943 [update notice] received update-write bai SetMode QQ=10: auto;27.5;-;-;0;0;1;0;0;0
2024-11-17 16:29:52.664 [update notice] received write vr_71 SetActorState QQ=10: -;off;off;off;off;off;-;-;off;off;off;off;00 00
2024-11-17 16:29:52.912 [update notice] received write vr_71 Mc1FlowTempDesired QQ=10: on;27.5;on;-100
2024-11-17 16:29:55.865 [update notice] sent read bai OutdoorstempSensor QQ=31: 8.19;ok
2024-11-17 16:29:55.886 [main error] read bai Status01 cached: decode ERR: element not found
2024-11-17 16:29:55.905 [main error] read bai Status01 cached: decode ERR: element not found
2024-11-17 16:29:56.088 [update notice] sent read bai PumpPower QQ=31: 15
2024-11-17 16:29:56.317 [update notice] sent read 700 WaterPressure QQ=31: 1.9
2024-11-17 16:29:56.541 [update notice] sent read 700 HwcStorageTemp QQ=31: 38.125
2024-11-17 16:29:56.770 [update notice] sent read 700 Hc1HeatCurve QQ=31: 0.35
2024-11-17 16:29:57.004 [update notice] sent read vr_71 SensorData1 QQ=31: -;-;-;-;38.12;18.94;8.38;55 00
2024-11-17 16:29:57.004 [main error] read vr_71 SensorData1: decode ERR: element not found
2024-11-17 16:29:57.057 [bus notice] max. symbols per second: 125

HeikoGr, 2024-11-18:
i think this is related to #1370.
so, the readings just have new names.

rande83, 2024-11-19:
Oh, sorry. I must have missed that.
I'll take a closer look and try to rewrite it so that it works again with the new names.

Thanks for the info.

pixeldoc2000, 2024-11-20:
On can roll back to docker tag `john30/ebusd:v23.2`. That will not fix the double entries in HA, but it will make your Automation and Dashboard work again.

This change is a bit PITA.

Maybe I will hold of with updating, untill refactory is done.

rande83, 2024-11-20:
Thanks for the advice.
I'll probably go back to the old version for the time being.

john30, 2024-11-24:
that won't work on the long run as the old webservice with the old csvs will be shutdown soon. instead you'd have to go with a localc checkout of the old CSVs.

closed 2024-11-24 by john30