john30/ebusd #1392: Updating to version 24.1 results in malfunction.

firewiremb, 2024-11-22:
### Description

The update from version 23.3 to 24.1 means that in my installation no data can be read from the bus anymore.
After I went back to version 23.3, everything works as before.

The line "error reading config files" that appears in the log file does not occur with version 23.3.
Therefor I added the config File in this Bug report.

[7c.renovent-excellent-400.csv](https://github.com/user-attachments/files/17866795/7c.renovent-excellent-400.csv)

### Actual behavior

```
pi@mypi:~ $ ebusctl read -f Aussenlufttemperatur
ERR: element not found
```

### Expected behavior

```
pi@mypi:~ $ ebusctl read -f Aussenlufttemperatur
1.6

```

### ebusd version

24.1

### ebusd arguments

--updatecheck=off --device=enh:192.168.188.34:9999 --configpath=/home/pi/Pi-ebus/ --latency=100 --logfile=/home/pi/Pi-ebus/log.txt --log=all:error

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

Adapter v3 WiFi

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

2024-11-22 08:45:03.483 [main error] error reading config files from /home/pi/Pi-ebus/: ERR: invalid argument, last error: 7c.renovent-excellent-400.csv:44: ERR: invalid argument, name EWTT-
2024-11-22 08:45:05.750 [device error] eBUS comm error: framing
2024-11-22 08:45:05.758 [device error] eBUS comm error: framing
2024-11-22 08:45:15.777 [device error] eBUS comm error: framing
2024-11-22 08:45:15.785 [device error] eBUS comm error: framing
2024-11-22 08:45:25.784 [device error] eBUS comm error: framing
2024-11-22 08:45:25.793 [device error] eBUS comm error: framing
2024-11-22 08:45:35.806 [device error] eBUS comm error: framing
2024-11-22 08:45:35.813 [device error] eBUS comm error: framing
2024-11-22 08:45:55.839 [device error] eBUS comm error: framing
2024-11-22 08:45:55.847 [device error] eBUS comm error: framing
2024-11-22 08:46:05.860 [device error] eBUS comm error: framing
2024-11-22 08:46:05.869 [device error] eBUS comm error: framing
2024-11-22 08:46:15.888 [device error] eBUS comm error: framing
2024-11-22 08:46:15.896 [device error] eBUS comm error: framing
2024-11-22 08:46:25.919 [device error] eBUS comm error: framing
2024-11-22 08:46:25.928 [device error] eBUS comm error: framing
2024-11-22 08:46:40.919 [device error] eBUS comm error: framing
2024-11-22 08:46:40.928 [device error] eBUS comm error: framing

jan666, 2024-11-23:
See https://github.com/john30/ebusd/issues/1376 for a Solution

john30, 2024-11-24:
special chars in message name are no longer allowed since 24.1 see the breaking changes in the [changelog](https://github.com/john30/ebusd/blob/master/ChangeLog.md)

closed 2024-11-24 by john30