john30/ebusd #1393: 1-write Sensor conencted to ebus adapter is not included in the MQTT

mintgroen, 2024-11-22:
### Description

I have connected a 1-wire DS18B20 to pin D7, 3.3v and gnd and the value shows in the status column:

[image]

Enabled MQTT server:

[image]

MQTT reports connected:

[image]

### Actual behavior

[image]

Defined topic pops up in the MQTT server with some info but the sensor value is not included

### Expected behavior

Sensor value transmitted to MQTT  server

### ebusd version

24.1

### ebusd arguments

None

### Operating system

Debian 10 (Buster) / Ubuntu 18-19 / Raspbian 10 / Raspberry Pi OS 10 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via WiFi

### Logs

Cant find any related log info in that file.

djungelorm, 2024-11-22:
I am also seeing this behaviour on the C6. +1 to get this feature added.

The sensor values do show up on the HTTP API though, via: `http://<ip-address>/api/v1/pin`

john30, 2024-11-24:
not an ebusd issue, so closing here.
use https://github.com/john30/ebusd-esp32 instead.
anyway, duplicate of https://github.com/john30/ebusd-esp32/issues/68

closed 2024-11-24 by john30

mintgroen, 2024-11-25:
> I am also seeing this behaviour on the C6. +1 to get this feature added.
> 
> The sensor values do show up on the HTTP API though, via: `http://<ip-address>/api/v1/pin`

Ok, easy to fix that in Node red and relay the info to MQTT.