john30/ebusd #1413: 24.1 container EBUSD_FOREGROUND

SzosszeNET, 2024-12-13:
### Description

The 24.1 or latest container image does accept the "EBUSD_FOREGROUND" environment variable anymore. Without the variable the container does not run - but exitst with a 0 statuscode

### Actual behavior

The container stats and stops

### Expected behavior

The container keeps running

### ebusd version

24.1

### ebusd arguments

EBUSD_VERSION 24.1, EBUSD_DEVICE ens:IP:9999, EBUSD_LATENCY 10, EBUSD_POLLINTERVAL 10

### Operating system

other

### CPU architecture

arm64

### Dockerized

latest

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

date	stream	content
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.911 [main info] configPath URL is valid
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.910 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x4008=write,alert, val 256=close notify
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.870 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1002=exit,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.870 [network debug] SSL state 9=SSLv3/TLS read server session ticket: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.870 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.870 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.870 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x1002=exit,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.870 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x20=done, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.870 [network debug] SSL state 18=SSLv3/TLS write finished: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.870 [network debug] SSL state 16=SSLv3/TLS write change cipher spec: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.870 [network debug] SSL state 11=SSLv3/TLS read finished: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.870 [network debug] SSL state 39=TLSv1.3 read server certificate verify: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.870 [network debug] SSL state 4=SSLv3/TLS read server certificate: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.869 [network debug] SSL state 38=TLSv1.3 read encrypted extensions: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.869 [network debug] SSL state 3=SSLv3/TLS read server hello: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.869 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.849 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.849 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.849 [network debug] SSL BIO state 5 res 1
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.849 [network debug] SSL BIO state 5 res -1
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.828 [network debug] SSL state 12=SSLv3/TLS write client hello: type 0x1002=exit,connect, val -1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.828 [network debug] SSL BIO state 6 res 0
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.827 [network debug] SSL BIO state 4 res 3
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.827 [network debug] SSL BIO state 3 res 1
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.826 [network debug] SSL BIO state 2 res -1
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.826 [network debug] SSL state 0=before SSL initialization: type 0x1001=loop,connect, val 1=?
2024/12/13 19:55:25	stdout	2024-12-13 19:55:25.826 [network debug] SSL state 0=before SSL initialization: type 0x10=start, val 1=?
2024/12/13 19:54:17	stdout	2024-12-13 19:54:17.123 [network debug] SSL state 1=SSL negotiation finished successfully: type 0x4008=write,alert, val 256=close notif

john30, 2024-12-14:
which image are you using? tried both devel and v24.1 and both behave as desired.
maybe in "EBUSD_DEVICE ens:IP:9999" the "IP" part needs to be corrected

closed 2024-12-14 by john30

SzosszeNET, 2024-12-14:
v24.1
obviously the address was a valid address, but regardless if that's valid or not the service runs.

Might be a Synology-related something - what I noticed that when I create the container there is a default empty "EBUSD_FOREGROUND" environment variable

[image]

Which unfortunately means that further settings cannot be made till I delete it.

Leaving the variable there looks like allows the container to run as expected

[image]

If the empty env variable is deleted

[image]

After the container is started is stops

[image]

with a 0 status

[image]

This wasn't the case with the 23.x images but seemingly an issue due to the "EBUSD_FOREGROUND" change in the 24.x images

SzosszeNET, 2024-12-14:
@john30 also would appear that EBUSD_MQTTJSON is not needed either nor accepted as an environment variable.

But EBUSD_FOREGROUND does appear is still needed...

john30, 2025-01-10:
EBUSD_FOREGROUND is part of the docker image itself, so there is no need to have it set explicitly