john30/ebusd #1418: Value not writeable via MQTT

ulrichsc, 2024-12-16:
### Description

I have a message defined as:
```
b,energymgr,z1_heat_mode,Z1 Heating Mode,,fe,2020,c679,,,UCH
w,energymgr,z1_heat_mode,Z1 Heating Mode,,fe,2020,c679,,,UCH
```

When the message is broadcasted on the bus, it is parsed and published via MQTT (I can see it passing with mosquitto_sub as `ebusd/energymgr/z1_heat_mode`).
Setting the value via ebusd works,e.g. `ebusctl write -c energymgr z1_heat_mode 2`

Setting the value via MQTT (e.g. mosquitto_sub) is ignored.
`mosquitto_pub -h 172.20.20.155 -p 1883 -u "ebusd" -P "<secret>" -t "ebusd/energymgr/z1_heat_mode" -m 3`

Is setting values via MQTT not supported or am I missing something?

### Actual behavior

Setting the value via MQTT (e.g. mosquitto_sub) is ignored.

### Expected behavior

When setting the value via MQTT it is passed to the bus.

### ebusd version

24.1

### ebusd arguments

EBUSD_OPTS="--device ens:172.20.20.165:9999 --configpath=/etc/ebusd/ --mqttport=1883 --mqtttopic=ebusd/%circuit/%name --mqttuser=ebusd --mqttpass=<password> --mqtthost=localhost --loglevel=debug"

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via Ethernet

### Related integration

MQTT generic

### Logs

The value set via MQTT is not showing in the logs.

wrongisthenewright, 2024-12-26:
Jumping in here because I made the same mistake: for writing to a topic you should use the /set sub-topic

In your case you should publish to the topic ebusd/energymgr/z1_heat_mode/set

ulrichsc, 2025-01-06:
Oh dang... you are right.
`https://github.com/john30/ebusd/wiki/3.3.-MQTT-client`

closed 2025-01-06 by ulrichsc