john30/ebusd #1420: Missing new datapoints with version 24.1

topa-LE, 2024-12-24:
### Description

**scan.15  = Vaillant;CTLV3;0808;8004
scan.76  = Vaillant;VWZIO;0902;5103
scan.f6  = Vaillant;NETX3;0124;0404**

Are there already new current tsp Are there already new current .tsp in src ?

Thank you. Best Regards.

topa-LE, 2024-12-29:
Done ;-) Thank You John for best Works!

use clone:
`https://github.com/eBUS/ebus.github.io/tree/main/next/de`
and symlink 
`ln -s /root/ebus.github.io/next/en /etc/ebusd/config`

Config: 
`--configpath=/etc/ebusd/config/`

address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0902;HW=5103", loaded "vaillant/08.hmu.csv"
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV3;SW=0808;HW=8004", loaded "vaillant/15.ctlv3.csv"
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0902;HW=5103", loaded "vaillant/76.vwzio.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0808;HW=8004", loaded "vaillant/ec.sol.sc.csv"

address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX3;SW=0124;HW=0404"

The only thing missing at the moment: 
NETX3;SW=0124;HW=0404"
as CSV

Regards

john30, 2025-01-10:
closing as not ebusd in the first place and solved apparently

closed 2025-01-10 by john30