john30/ebusd #1424: Segmentation fault (core dumped) - when ssh into docker container

D058548, 2024-12-30:
### Description

Executing ebusctl inside the docker container leads to error message "Segmentation fault (core dumped)". 

Im following the steps described in the ebusd addon description in chapter "Running ebusd and ebusctl directly". It neither works via putty nor the terminal (ssh addon) in home assistant. My goal is to execute ebusctl write commands towards my vaillant heap pump. 

➜  ~ docker exec -it `docker ps | grep ebusd | awk '{print $1}'` /bin/bash
2ad9b828-ebusd:/# ebusctl
Segmentation fault (core dumped)
2ad9b828-ebusd:/# 

Apart from that the addon works with the following config: 

scanconfig: true
loglevel_all: notice
mqtttopic: ebusd
mqttint: /etc/ebusd/mqtt-hassio.cfg
mqttjson: true
mode: ens
network_device: 192.168.178.125:9999
latency: 10
configpath: /etc/ebusd/config/de

Im having almost all data inside MQTT. Now I would like to write values to the Vaillant Heap Pump (AroTherm 55). 

I tried to reinstall the addon without success. 

Patch Level: 

Home Assistant: Core: 2024.12.5, Supervisor: 2024.12.0
Supervised HA installation
ebusd:  23.2.6
eBUS Adapter Shield C6
OS: Debian GNU/Linux 12 (bookworm)
Docker version: Docker version 27.4.1, build b9d17ea

### Actual behavior

Segmentation fault (core dumped)

### Expected behavior

Reading results from the vaillants heat pump ebus. 

### ebusd version

current source from git

### ebusd arguments

ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=secret --scanconfig --mqttjson --configpath=/etc/ebusd/config/de --latency=10 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.178.125:9999 --log=all:notice

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via USB

### Logs

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=sohgiewie7Vo3saipah2ooquahshemieCiu4phohzee6Lurae2Eed9eeShohkah9 --scanconfig --mqttjson --configpath=/etc/ebusd/config/de --latency=10 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=ens:192.168.178.125:9999 --log=all:notice
2024-12-30 14:21:56.358 [device notice] transport opened
2024-12-30 14:21:56.361 [main notice] ebusd 24.1.p20241110 started with broadcast scan on device: 192.168.178.125:9999, TCP, enhanced
2024-12-30 14:21:56.364 [bus notice] bus started with own address 31/36
2024-12-30 14:21:56.364 [device notice] reset, supports info
2024-12-30 14:21:56.365 [mqtt notice] connection established
2024-12-30 14:21:56.366 [device notice] extra info: firmware 1.1[4c15].1[4c15], jumpers 0x0b
2024-12-30 14:21:56.392 [bus notice] signal acquired
2024-12-30 14:21:56.817 [bus notice] new master 10, master count 2
2024-12-30 14:21:56.875 [bus notice] new master 03, master count 3
2024-12-30 14:21:56.875 [update notice] received unknown MS cmd: 1008b5110101 / 093f360080ffff0100ff
2024-12-30 14:21:57.139 [bus notice] new master 71, master count 4
2024-12-30 14:21:57.140 [update notice] received unknown MS cmd: 1076b5110101 / 093effc300ffff0000ff
2024-12-30 14:21:57.404 [update notice] received unknown MS cmd: 1076b512030f0001 / 07008000f20111ff
2024-12-30 14:21:57.627 [update notice] received unknown MS cmd: 7108b5110107 / 0a2b920181083300000000
2024-12-30 14:21:57.665 [update notice] received unknown BC cmd: 10feb505025c00
2024-12-30 14:21:57.930 [update notice] received unknown MS cmd: 1008b51009000042ffffff060000 / 0101
2024-12-30 14:21:58.195 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2024-12-30 14:21:58.463 [update notice] received unknown MS cmd: 1076b5040100 / 0a0300221430120124c300
2024-12-30 14:21:58.702 [update notice] received unknown MS cmd: 1008b507020942 / 026401
2024-12-30 14:21:58.941 [update notice] received unknown BC cmd: 10feb516080057211430120124
2024-12-30 14:21:59.206 [update notice] received unknown MS cmd: 1008b5110100 / 09fa0111740409000058
2024-12-30 14:21:59.449 [update notice] received unknown MS cmd: 1076b51303040d00 / 020d00
2024-12-30 14:21:59.666 [update notice] received unknown BC cmd: 10feb5160301b4fe
2024-12-30 14:21:59.922 [update notice] received unknown MS cmd: 0376b512061300115c0300 / 0200ff
2024-12-30 14:22:01.065 [update notice] received unknown MS cmd: 7108b5110107 / 0a2c920181083300000000
2024-12-30 14:22:04.364 [bus notice] new master f1, master count 5
2024-12-30 14:22:04.407 [update notice] received unknown MS cmd: f115b52406020003001b00 / 0601031b000400
2024-12-30 14:22:04.499 [update notice] received unknown MS cmd: 7108b5110107 / 0a2c920181083300000000
2024-12-30 14:22:06.365 [main notice] starting initial broadcast scan
2024-12-30 14:22:06.466 [update notice] sent write broadcast queryexistence QQ=31: 
2024-12-30 14:22:06.911 [update notice] received unknown MS cmd: 1008b5110101 / 093f360080ffff0100ff
2024-12-30 14:22:07.175 [update notice] received unknown MS cmd: 1076b5110101 / 093effc300ffff0000ff
2024-12-30 14:22:07.440 [update notice] received unknown MS cmd: 1076b512030f0001 / 07008000f10111ff
2024-12-30 14:22:07.704 [update notice] received unknown MS cmd: 1008b51009000042ffffff060000 / 0101
2024-12-30 14:22:07.870 [update notice] received unknown MS cmd: 0376b512061300115d0300 / 0200ff
2024-12-30 14:22:07.964 [update notice] received unknown MS cmd: 7108b5110107 / 0a2c920181083300000000

D058548, 2025-01-09:
Also after the update to Version 24.1.0 Im still getting "Segmentation fault (core dumped)" when executing ebusctl within the docker container.

cociweb, 2025-01-10:
both of the above issue was rooted in the latest ha-addon version - the fix is already released: Please update your instance. There is nothing to do by @john30  in his repos, so please, close the ticket.

john30, 2025-01-10:
I thought so, thanks for clarifiying. wondering why people still use the old (and no longer maintained) ebusctl instead of netcat or telnet and friends...

closed 2025-01-10 by john30