john30/ebusd #1445: Missing WPPWMPower (`0x7300`) since v24.1

git-developer, 2025-01-13:
### Description

After update from v23.3 to v24.1 (Docker), the value WPPWMPower (`0x7300`) is missing (Vaillant / BAI).

First, I was thinking this might be related to the major configuration rework (typespec) or some other kind of configuraton issue (cf. https://github.com/john30/ebusd-configuration/issues/450). But now I found out that even a request for the value by hex command has different behavior, so I guess this is no config issue.

Downgrading to v23.3 still provides the WPPWMPower value, so the hardware / heating side seems to be OK.

### Actual behavior

```
read -h 08b509030d1700
03cb0200

read -h 08b509030d7300
ERR: element not found
```

### Expected behavior

```
read -h 08b509030d1700
03cb0200

read -h 08b509030d7300
0100
```

### ebusd version

24.1

### ebusd arguments

-f --device=/dev/ttyUSB_ebus --scanconfig --logareas=main,network,bus --loglevel=notice --mqtt...

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

armv7l

### Dockerized

same as ebusd version

### Hardware interface

other

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

### v24.1
```
ebusd  | 2025-01-13 08:42:47.774 [main notice] ebusd 24.1.24.1 started with broadcast scan on device: /dev/ttyUSB_ebus, serial
ebusd  | 2025-01-13 08:42:48.057 [bus notice] bus started with own address 31/36
ebusd  | 2025-01-13 08:42:48.058 [bus notice] signal acquired
ebusd  | 2025-01-13 08:42:48.975 [bus notice] new master 10, master count 2
ebusd  | 2025-01-13 08:42:49.045 [bus notice] new master 03, master count 3
ebusd  | 2025-01-13 08:42:58.058 [main notice] starting initial broadcast scan
ebusd  | 2025-01-13 08:43:07.882 [bus notice] scan 08: ;Vaillant;BAI00;0414;7401
ebusd  | 2025-01-13 08:43:07.882 [bus notice] scan 08: ;Vaillant;BAI00;0414;7401
ebusd  | 2025-01-13 08:43:08.266 [main notice] read common config file vaillant/broadcast.csv
ebusd  | 2025-01-13 08:43:08.360 [main notice] read common config file vaillant/general.csv
ebusd  | 2025-01-13 08:43:08.453 [main notice] read common config file vaillant/scan.csv
ebusd  | 2025-01-13 08:43:08.554 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0414, HW7401
ebusd  | 2025-01-13 08:43:08.685 [main notice] found messages: 229 (3 conditional on 30 conditions, 0 poll, 16 update)
ebusd  | 2025-01-13 08:43:09.372 [bus notice] scan 08: ;21;09;12;0010003862;3100;005747;N1
ebusd  | 2025-01-13 08:43:11.535 [bus notice] scan 15: ;Vaillant;43000;0127;2002
ebusd  | 2025-01-13 08:43:11.535 [bus notice] scan 15: ;Vaillant;43000;0127;2002
ebusd  | 2025-01-13 08:43:12.331 [bus notice] scan 15: ;21;07;16;0020028515;0008;013271;N2
ebusd  | 2025-01-13 08:43:12.672 [main notice] read scan config file vaillant/15.430.csv for ID "43000", SW0127, HW2002
ebusd  | 2025-01-13 08:43:12.672 [main notice] found messages: 450 (19 conditional on 38 conditions, 0 poll, 16 update)
ebusd  | 2025-01-13 08:43:32.195 [bus error] poll 430 AdpPreHCurrentRoomTemp failed: ERR: read timeout
ebusd  | 2025-01-13 08:43:38.232 [bus error] poll 430 AdpPreHOutdoorTemp failed: ERR: read timeout
ebusd  | 2025-01-13 08:43:38.700 [main notice] read hex cmd: 3108b509030d1700
ebusd  | 2025-01-13 08:43:44.222 [bus error] poll 430 AdpPreHOutdoorTempStart failed: ERR: read timeout
ebusd  | 2025-01-13 08:43:44.468 [main notice] read hex cmd: 3108b509030d7300
ebusd  | 2025-01-13 08:43:50.212 [bus error] poll 430 AdpPreHRoomTempDesired failed: ERR: read timeout
ebusd  | 2025-01-13 08:43:56.196 [bus error] poll 430 AdpPreHRoomTempStart failed: ERR: read timeout
```

### v23.3
```
ebusd  | 2025-01-13 08:39:46.210 [bus notice] device status: transport opened
ebusd  | 2025-01-13 08:39:46.210 [main notice] ebusd 23.3.23.3 started with broadcast scan on device: /dev/ttyUSB_ebus, serial
ebusd  | 2025-01-13 08:39:46.771 [bus notice] bus started with own address 31/36
ebusd  | 2025-01-13 08:39:46.771 [bus notice] signal acquired
ebusd  | 2025-01-13 08:39:49.620 [bus notice] new master 10, master count 2
ebusd  | 2025-01-13 08:39:49.687 [bus notice] new master 03, master count 3
ebusd  | 2025-01-13 08:39:56.772 [main notice] starting initial broadcast scan
ebusd  | 2025-01-13 08:40:07.073 [bus notice] scan 08: ;Vaillant;BAI00;0414;7401
ebusd  | 2025-01-13 08:40:07.073 [bus notice] scan 08: ;Vaillant;BAI00;0414;7401
ebusd  | 2025-01-13 08:40:07.662 [main notice] read common config file vaillant/scan.csv
ebusd  | 2025-01-13 08:40:07.806 [main notice] read common config file vaillant/broadcast.csv
ebusd  | 2025-01-13 08:40:07.947 [main notice] read common config file vaillant/general.csv
ebusd  | 2025-01-13 08:40:08.088 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0414, HW7401
ebusd  | 2025-01-13 08:40:09.241 [main notice] found messages: 202 (0 conditional on 26 conditions, 0 poll, 10 update)
ebusd  | 2025-01-13 08:40:09.979 [bus notice] scan 08: ;21;09;12;0010003862;3100;005747;N1
ebusd  | 2025-01-13 08:40:10.031 [bus notice] max. symbols per second: 114
ebusd  | 2025-01-13 08:40:12.125 [bus notice] scan 15: ;Vaillant;43000;0127;2002
ebusd  | 2025-01-13 08:40:12.125 [bus notice] scan 15: ;Vaillant;43000;0127;2002
ebusd  | 2025-01-13 08:40:12.918 [bus notice] scan 15: ;21;07;16;0020028515;0008;013271;N2
ebusd  | 2025-01-13 08:40:13.208 [main notice] read scan config file vaillant/15.430.csv for ID "43000", SW0127, HW2002
ebusd  | 2025-01-13 08:40:13.891 [main notice] found messages: 422 (16 conditional on 34 conditions, 0 poll, 10 update)
ebusd  | 2025-01-13 08:40:36.898 [main notice] read hex cmd: 3108b509030d1700
ebusd  | 2025-01-13 08:40:53.476 [main notice] read hex cmd: 3108b509030d7300
ebusd  | 2025-01-13 08:41:08.627 [main notice] scan completed 1 time(s), check again
```

john30, 2025-01-21:
please post the result of `find -f -c bai` on both versions or at least on 24.1

git-developer, 2025-01-21:
### v24.1
```
i
version: ebusd 24.1.24.1
device: /dev/ttyUSB_ebus, serial
signal: acquired
symbol rate: 35
max symbol rate: 112
min arbitration micros: 3
max arbitration micros: 9
min symbol latency: 5
max symbol latency: 7
scan: finished
reconnects: 0
masters: 3
messages: 450
conditional: 19
poll: 135
update: 16
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0414;HW=7401", loaded "vaillant/bai.308523.inc" ([id_hw=7401]), "vaillant/08.bai.cs"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=43000;SW=0127;HW=2002", loaded "vaillant/15.430.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

find -f -c bai
r,bai,AccessoriesOne,d.27 Zubehörrelais 1,,08,b509,0d4b04,value,s,UCH,1=circulationpump;2=extheatingpump;3=storagechargingpump;4=fluegasfla1
r,bai,AccessoriesTwo,d.28 Zubehörrelais 2,,08,b509,0d4c04,value,s,UCH,1=circulationpump;2=extheatingpump;3=storagechargingpump;4=fluegasfla2
r,bai,ACRoomthermostat,d.08 Raumthermostat: Status des extrenen Raumthermostat an Klemme 3/4,,08,b509,0d2a00,value,s,UCH,0=off;1=on,,
r5,bai,APCComStatus,d.92 APC_ComStatus_DK: actoSTORE communication status,,08,b509,0d6200,value,s,UCH,,,
r,bai,APCLegioProtection,d.74 APC_LegioProtection: Legionella protection for actoSTORE,,08,b509,0d9704,value,s,UCH,,,
r,bai,AverageIgnitiontime,d.64 Mittlere Zündzeit,,08,b509,0d2d00,value,s,UCH,10,s,Mittlere Zündzeit
r,bai,BlockTimeHcMax,d.02 Maximale Brennersperrzeit: Max. Brennersperrzeit bei einem Vorlaufsollwert von 20°C,,08,b509,0d0904,value,s,UCH,,n
r,bai,BoilerType,BoilerType_DK: Boiler typ of the bmu,,08,b509,0d5e00,value,s,UCH,,,
r,bai,ChangesDSN,Anzhl der DSN Änderungen: Anzahl der DSN (Gerätekennung) Änderungen,,08,b509,0d0c00,value,s,UCH,,,
r,bai,CirPump,d.13 Zirkulationspumpe: Status Zirkulationspumpe (über ein externes Modul ansteuerbar),,08,b509,0d7b00,value,s,UCH,0=off;1=on,
r5,bai,CounterStartattempts1,d.68 Zündfehler 1. Versuch: Anzahl der erfolglosen Zündversuche (im 1. Versuch),,08,b509,0d6e00,value,s,UCH,,°r
r5,bai,CounterStartattempts2,d.69 Zündfehler 2. Versuch: Anzahl der erfolglosen Zündversuche (im 2. Versuch),,08,b509,0d6f00,value,s,UCH,,°r
r5,bai,CounterStartAttempts3,Zündfehler 3. Versuch: Anzahl der erfolglosen Zündversuche (im 3. Versuch),,08,b509,0d8100,value,s,UCH,,°C,Temr
r5,bai,CounterStartAttempts4,Zündfehler 4. Versuch: Anzahl der erfolglosen Zündversuche (im 4. Versuch),,08,b509,0d8200,value,s,UCH,,°C,Temr
r,bai,Currenterror,Aktuelle Fehler,,08,b503,0001,error,s,UIN,,,Fehlernummer,error_1,s,UIN,,,Fehlernummer,error_2,s,UIN,,,Fehlernummer,errorr
r,bai,DateTime,Datum Uhrzeit,,08,b504,00,dcfstate,s,UCH,0=nosignal;1=ok;2=sync;3=valid,,DCF Empfängerstatus,btime,s,BTI,,,Uhrzeit,bdate,s,Br
r,bai,DcfState,d.91 DCF Status,,08,b509,0d6900,value,s,UCH,0=nosignal;1=ok;2=sync;3=valid,,DCF Empfängerstatus
r,bai,DCFTimeDate,DCF Datum/Uhrzeit,,08,b509,0de500,btime,s,BTI,,,Uhrzeit,bdate,s,BDA,,,Datum
r,bai,DCRoomthermostat,d.16 Raumthermostat: Wärmeanforderung vom externen Regler (Klemme 3-4),,08,b509,0d0e00,value,s,UCH,0=off;1=on,,
r,bai,DeactivationsIFC,d.61 Anzahl Zündfehler: Anzahl der Zündfehler (nicht erfolgreiche Züdung im letzten Versuch oder fehlerhaftes Flamme,
r5,bai,DeactivationsTemplimiter,d.60 STB Abschaltungen: Anzahl der Abschaltungen durch den Sicherheitstemperaturbgrenzers,,08,b509,0d2000,v,
r,bai,DeltaFlowReturnMax,MaxDeltaFlowReturn_DK: Wartungsdaten,,08,b509,0d3d00,value,s,D2C,16,°C,Temperatur
r5,bai,DisplayMode,DisplayMode_DK: Display mode of the aplliance,,08,b509,0dda00,value,s,UCH,,,
r,bai,DSN,DSN: Device Specific number,,08,b509,0d9a00,value,s,UIN,,,
r,bai,DSNOffset,d.93 Gerätekennung: Gerätekennung (DSN),,08,b509,0d3004,value,s,UCH,,,
r,bai,DSNStart,DSN Startadresse,,08,b509,0d3104,value,s,UIN,,,
r,bai,EBusHeatcontrol,d.90 Digitaler Regler: Gibt an ob ein digitaler Regler erkannt wurde,,08,b509,0d0004,value,s,UCH,0=no;1=yes,,
r,bai,EbusSourceOn,eBUS Spannungsversorgung: Aktivierung der eBUS Speisung,,08,b509,0d3400,value,s,UCH,0=off;1=on,,
r,bai,EbusVoltage,Rückmeldung der eBUS Spannung,,08,b509,0d7f00,value,s,UCH,0=off;1=on,,
r,bai,Errorhistory,Fehlerhistorie,,08,b503,0101,index,m,UCH,,,,status,s,UCH,,,Status,time,s,VTM,,,Uhrzeit,date,s,HDA:3,,,Datum,error,s,UIN,r
r5,bai,Expertlevel_ReturnTemp,Externer Rücklauftemperatursensor,,08,b509,0d6b00,temp,s,D2C,16,°C,Temperatur,sensor,s,UCH,0=ok;85=circuit;17s
r,bai,ExternalFaultmessage,Ext. Störmeldung: Signal für die externe Störmeldeeinrichtung,,08,b509,0d3e00,value,s,UCH,0=off;1=on,,
r5,bai,ExternalFlowTempDesired,ext. Vorlaufsollwert: Vorlaufsollwert von einem externen Regler an Klemme 7-8-9,,08,b509,0d2500,value,s,D2C,r
r,bai,ExternalHwcSwitch,Wasserschalter: Speicheranforderung eines externen Speichers über den Speicherkontakt,,08,b509,0d0000,value,s,UCH,0,
r,bai,ExternGasvalve,Ext. Gasventil: Externes Magnetventil,,08,b509,0de400,value,s,UCH,0=off;1=on,,
r5,bai,ExtFlowTempDesiredMin,d.09 Min. ext. Vorlaufsollwert: minimum out of Kl.7 and eBus flow setpoint,,08,b509,0d6e04,value,s,D2C,16,°C,Tr
r,bai,ExtStorageModulCon,VR65 angeschlossen: Externes Speichermodul (VR65) angeschlosssen,,08,b509,0dbf00,value,s,UCH,0=no;1=yes,,
r,bai,ExtWP,d.11 ext. Heizungspumpe: Externe Heizungspumpe,,08,b509,0d3f00,value,s,UCH,0=off;1=on,,
r5,bai,FanHours,Betriebsstunden Lüfter: Betriebsstunden des Lüfters,,08,b509,0d1b00,value,s,UIN,,h,Stunden
r,bai,FanMaxSpeedOperation,Max. Lüfterdrehzahl,,08,b509,0de000,value,s,UIN,,1/min,Lüfter Maximaldrehzahl
[... cut ...]
r,bai,WaterpressureVariantSum,WaterpressureVariantSum_DK,,08,b509,0df000,pressm2,s,UIN,,mbar,Wartungsdaten
r,bai,WP,d.10 Wasserpumpe,,08,b509,0d4400,onoff,s,UCH,0=off;1=on,,Interne Heizungspumpe
r,bai,WPPostrunTime,d.01 Pumpennachlaufzeit,,08,b509,0df703,minutes0,s,UCH,,min,Wasserpumpennachlaufzeit nach Heizbetrieb
r,bai,WPPWMPower,d.15 Pumpendrehzahl Istwert,,08,b509,0d7300,percent0,s,UCH,,%,Output of pwm pump
r,bai,WPSecondStage,WP_SecondStage_DK,,08,b509,0ded00,,s,UCH,15=on;240=off,,Second stage of the pump activated
```

john30, 2025-01-26:
can't reproduce. please provide the logs again with additional "--lograwdata"

git-developer, 2025-01-26:
### v24.1
[v24.1.log](https://github.com/user-attachments/files/18549468/v24.1.log)
[v24.1-raw.log](https://github.com/user-attachments/files/18549470/v24.1-raw.log)

### v23.3
[v23.3.log](https://github.com/user-attachments/files/18549469/v23.3.log)
[v23.3-raw.log](https://github.com/user-attachments/files/18549471/v23.3-raw.log)

git-developer, 2025-01-26:
Yesterday, the system was rebooted. The problem no longer occurs since this reboot.

This is really surprising to me: before the reboot, the value `WPPWMPower` was provided from v23.3 but not from v24.1.

Anyway, not reproducible, no longer occurs => case closed.

closed 2025-01-26 by git-developer