john30/ebusd #1464: Support BIT data types for 2 data bytes and reusing of message id

J11522, 2025-01-21:
### Description

I'm currently trying to integrate my Wolf CHA 07 into HomeAssistant. 
I've taken the inspiration from https://github.com/puni2k/ebusd-configuration-wolf and https://github.com/zivillian/ism7mqtt to auto generate the messages.

Doing this I stumbled across two problems.
In the Wolf description files, two bytes of a message are used for multiple parameters, where each bit controls a single binary param.
I tried encoding this using BI0:1...BI7:1 as type, but this doesn't allow me to use values from the second byte.

Furtheremore, I get a duplicate ID error.

Am I missing something or are that features currently not supported by ebusd?
I can split this into two issues for better readability.

john30, 2025-01-26:
it works and is also part of the test cases. so please share your definition.
anyway, if you want to use bits from a following byte, you need to ensure that the first bit in the def addressing that byte uses a  smaller bit index than the last from the previous. i.e. BI0;BI7;BI0 splits into two bytes with BI0;BI7 and BI0
for the duplicate Id: well a duplicate id is an issue, isn't it? how would you then distinguish two messages.
due to lack of input, i can't tell you more

J11522, 2025-01-27:
Thanks for the answer. Maybe the practical example makes it more clear.

r | cha | heizkreispumpe | Heizkreispumpe | 8 | 5022 | CC7930 | heizkreispumpe | BI0:1 | 0X00=Aus;0X01=Ein
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
r | cha | zubringer_heizkreispumpe | Zubringer-/Heizkreispumpe | 8 | 5022 | CC7930 | zubringer_heizkreispumpe | BI1:1 | 0X00=Aus;0X01=Ein
r | cha | drei_wege_umschaltventil_hz_ww | 3-Wege-Umschaltventil HZ/WW | 8 | 5022 | CC7930 | drei_wege_umschaltventil_hz_ww | BI2:1 | 0X00=Heizen;0X01=Warmwasser
r | cha | drei_wege_umschaltventil_hz_k | 3-Wege-Umschaltventil HZ/K | 8 | 5022 | CC7930 | drei_wege_umschaltventil_hz_k | BI3:1 | 0X00=Heizen;0X01=Kühlen
r | cha | ausgang_a1 | Ausgang A1 |   | 8 | 5022 | CC7930 | ausgang_a1 |   | BI4:1 | 0X00=Aus;0X01=Ein
r | cha | e_heizung | E-Heizung |   | 8 | 5022 | CC7930 | e_heizung |   | BI5:1 | 0X00=Aus;0X01=Ein
r | cha | verdichter | Verdichter |   | 8 | 5022 | CC7930 | verdichter |   | BI6:1 | 0X00=Aus;0X01=Ein
# r | cha | ausgang_a3 | Ausgang A3 |   | 8 | 5022 | CC7930 | ausgang_a3 |   | BI8:1 | 0X00=Aus;0X01=Ein
# r | cha | ausgang_a4 | Ausgang A4 |   | 8 | 5022 | CC7930 | ausgang_a4 |   | BI9:1 | 0X00=Aus;0X01=Ein
# r | cha | e_heizung_phase_1 | E-Heizung Phase 1 | 8 | 5022 | CC7930 | e_heizung_phase_1 | BI10:1 | 0X00=Aus;0X01=Ein
# r | cha | e_heizung_phase_2_und_3 | E-Heizung Phase 2 und 3 | 8 | 5022 | CC7930 | e_heizung_phase_2_und_3 | BI11:1 | 0X00=Aus;0X01=Ein

Here, we have multiple binary values encoded in the bits of a single message. 
TBH. I am not sure of the correct definition of the BI0...7 as the docs state the length can be 0-7, so I figured the length indicator of :1 is required.

Can you suggest on how to properly encode this?

john30, 2025-01-29:
if there is an extra byte that needs to be skipped, then just insert an ignored field using the IGN data type

closed 2025-02-03 by john30