john30/ebusd #1472: No status in Ui since new firmware

pascalwinters, 2025-01-26:
### Description

Since I updated to the last firmware 2025-01-20 I have issues in the UI. Hardware v5.  See screenshot:

[image]

### Actual behavior

See screenshot 

### Expected behavior

Actual status

### ebusd version

24.1

### ebusd arguments

Na

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via USB

### Logs

Na

Koky05, 2025-01-27:
Same behavior in my case Adapter Shield v5 rev3 via WiFi.

pascalwinters, 2025-01-27:
> Same behavior in my case Adapter Shield v5 rev3 via WiFi. But it is only GUI problem because adapter works correctly on background and communicate with heater without any problem.

For me exactly the same, only on Ethernet instead of WiFi.

derjohn, 2025-01-27:
Yes, +1 from me. Extactly the same. Also v5 with wifi.

john30, 2025-01-27:
not an ebusd issue, but ebusd-esp32. thus closing here. please reopen there if needed.
anyway, since I can't reproduce and too little info is given here, please provide more details and check if the settings are still on the device when using the EBUS access point (button click+hold while the leds fade 3 times)

closed 2025-01-27 by john30

pascalwinters, 2025-01-28:
https://github.com/john30/ebusd-esp32/issues/93