john30/ebusd #1474: Help Needed. Can't find config for Vaillant Ecotec Plus VCW 296 3/5

jonasv05, 2025-01-27:
### Description

First off, i'm sorry for making a bugreport but I didn't know how to get one labeled "help wanted".

My setup is as followed:

1. eBUS Adapter Shield c6 (WiFi)
```
Hostname: ebus-95c660
Up time: 13943
Free heap: 181404 / 365192
ebusd connected: yes
eBUS signal: acquired
MQTT connected: yes
```
2. ebusd in docker from john30/ebusd:latest via docker-compose

  ebusd: #om signaal van boiler van ebus adapter (10.0.50.8) om te zetten naar iets bruikbaars
    image: john30/ebusd:latest
    container_name: ebusd
    hostname: ebusd
    environment:
      EBUSD_MQTTPORT: 1883
      EBUSD_MQTTHOST: 10.x.x.x
      EBUSD_MQTTCLIENTID: "********************"
      EBUSD_MQTTUSER: "********************" # Connect as USER to MQTT broker (no default)
      EBUSD_MQTTPASS: "********************" # Use MQTT TOPIC (prefix before /%circuit/%name or complete format)
      EBUSD_MQTTTOPIC: "ebusd" # Use TOPIC for global data (default is "global/" suffix to mqtttopic prefix)
      #EBUSD_MQTTRETAIN: "" # Retain all topics instead of only selected global ones
      EBUSD_OPTS: "--scanconfig -d "
      EBUSD_DEVICE: "mdns:e4b32395c660"
      EBUSD_LATENCY: 40 # Extra transfer latency in ms
      EBUSD_CONFIGPATH: "https://ebus.github.io/" #Path to local config files
      EBUSD_SCANCONFIG: "full"
      EBUSD_HTMLPATH: "/var/ebusd/html" # Path for HTML files served by HTTP port
      EBUSD_PORT: 8888 # Listen for command line connections on PORT
      EBUSD_HTTPPORT: 8080 # Listen for HTTP connections on PORT, 0 to disable
      # Log options:
      EBUSD_LOGFILE: "/var/log/ebusd.log" # Write log to FILE (only for daemon, empty string for using syslog)
      EBUSD_LOG: "all:debug" # Only write log for matching AREA(S) up to LEVEL (alternative to EBUSD_LOGAREAS/EBUSD_LOGLEVEL, may be used multiple times)
    depends_on:
      - mqtt5      
    volumes:
       - /mnt/NAS01/docker/casa-vanve-docker01/domotica/ebusd/config:/ebusd/config:rw 
       - /mnt/NAS01/docker/casa-vanve-docker01/domotica/ebusd/html:/var/ebusd/html:rw
       - /mnt/NAS01/docker/casa-vanve-docker01/domotica/ebusd/log/:/var/log/:rw  
    networks:
      VLAN0-MGMT:
        ipv4_address: 10.x.x.x  
    ports:
      - "8888:8888" #
      - "8080:8080" #webport
    restart: unless-stopped

### Actual behavior

Question 1: Configfiles
So i've tried configuring with both local and the online repository. Both give me the following answer on "ebusctl info"
So the question is how can i know that this is the correct config for my boiler? How can i find out which configs should be used for my boiler?
```
version: ebusd 24.1.24.1
device: 10.0.50.8, TCP, enhanced, firmware 1.1[5114].1[5114]
signal: acquired
symbol rate: 22
max symbol rate: 94
min arbitration micros: 3
max arbitration micros: 5
min symbol latency: 5
max symbol latency: 24
scan: finished
reconnects: 0
masters: 2
messages: 250
conditional: 0
poll: 1
update: 16
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.0010006101.inc" ([Scan_id_product='']), "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

Question 2: logfiles
Currently I'm checking what's going on via "sudo docker exec -it <containerid> ebusctl info" or "sudo docker exec -it <containerid> /bin/bash" followed bij "ebusctl" in the root prompt.
However for some reason I can't get the logfile to work the way I've set it up and, to me, this looks correct.

Question 3:
I'm confused as to the HTTP port. I've enabled it, I've set a path to html files (that is blank). Is there a webinterface for ebusd I don't know about or is it to send commands over HTTP?

Thx in advance for all the clarifications, I hope to hear back form you soon!

### Expected behavior

Logs to work
Correct config to get loaded
Clarification on the HTTP-Port usage (webinterface or smth else?)
I do see that MQTT keeps reconnecting, what settings should I tweak to prevent this?

### ebusd version

24.1

### ebusd arguments

--scanconfig -d

### Operating system

other

### CPU architecture

x64

### Dockerized

latest

### Hardware interface

other

### Logs

as said i don't get logs to work for some reason.

pixeldoc2000, 2025-01-28:
Answer 1)
Looks like ebusd did detect your boilder  "MF=Vaillant;ID=BAI00;SW=0518;HW=7401" and loaded a suitable config "vaillant/bai.0010006101.inc".
So your eBUS, docker and basic setup is working.
Now test if you can read any / correct data from your boilder.

Execute to following in the container:
- `ebusctl find -F name` will give you a list with "know" values.
- `ebusctl read FlowTempDesired` pick some values by names and try to read them like `FlowTempDesired` (this should read the heater temp setting from you boiler)

If you can get most or all value, the config should fit your boiler, otherwise it is possible to map your boilder "SW=0518;HW=7401" to an other config via a local config. Than it should fixed via PR.

Answer 2)
Log does not work for me neither in docker. Maybe a bug or permission error?

Answer 3)
AFAIK there is a http api one can query, it is not a Webinterface you can use with a browser.

john30, 2025-02-03:
thanks for answering @pixeldoc2000 

re 2: the logfile option is only relevant when ebusd is run in daemon mode, which is not the case for docker. instead logging is sent to stdout and it is up to the user to adjust the container/dockerd logging appropriately

re 3: there is a rather old (and unmaintained) web UI for ebusd under https://github.com/john30/ebusd/tree/master/contrib/html which uses the [http api](https://github.com/john30/ebusd/blob/master/contrib/html/openapi.yaml) to retrieve data from ebusd and visualize it

closed 2025-02-03 by john30