john30/ebusd #1477: Ebus adapter latest firmware 2025-01-20

SzosszeNET, 2025-01-29:
### Description

It appears that the lates firmware did soft brick my adapter - version 2025-01-20.

As far as I can see the ebusd can still poll data from the adapter, but the adapters status page and potentially other part of the interface are broken

[image]

### Actual behavior

API retrieval failed for /status: timeout

### Expected behavior

No timeouts

### ebusd version

current source from git

### ebusd arguments

N/A

### Operating system

other

### CPU architecture

x64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via USB

### Logs

14:23:14	
Info	

execute: led 0 255 0 700

14:23:14	
	

0 duty requested to 254

14:23:14	
Info	

execute: led 1 255 0 700

14:23:14	
	

1 duty set to 254

14:23:14	
Info	

execute: led 2 255 0 700

14:23:14	
	

2 duty set to 254

14:23:14	
Info	

execute: sleep 1000

14:23:14	
Info	

leds: duty set to 254

14:23:15	
Info	

execute: if "isio -l 9" "var skipstart 1"

14:23:15	
Info	

execute: isio -l 9

14:23:15	
	

IO 9 level 1 is unmet

14:23:15	
Info	

execute: leds 255 0 700

14:23:15	
Info	

execute: led 0 255 0 700

14:23:15	
	

0 duty requested to 254

14:23:15	
Info	

execute: led 1 255 0 700

14:23:15	
	

1 duty set to 254

14:23:15	
Info	

execute: led 2 255 0 700

14:23:15	
	

2 duty set to 254

14:23:15	
Info	

execute: sleep 1000

14:23:15	
Info	

leds: duty set to 254

14:23:16	
Info	

execute: if "isio -l 9" "var skipstart 1"

14:23:16	
Info	

execute: isio -l 9

14:23:16	
	

IO 9 level 1 is unmet

14:23:16	
Info	

execute: var rpi 0

14:23:16	
Info	

execute: if "isio -l 6" "var rpi 1"

14:23:16	
Info	

execute: isio -l 6

14:23:16	
	

IO 6 level 1 is unmet

14:23:16	
Info	

execute: var isnew 1

14:23:16	
Info	

execute: if "§isnvs ebus.host" "var isnew 0"

14:23:16	
Info	

execute: isnvs ebus.host

14:23:16	
Info	

execute: var isnew 0

14:23:16	
	

variable isnew = 0

14:23:16	
Info	

execute: if "§isnvs sta.on" "var isnew 0"

14:23:16	
Info	

execute: isnvs sta.on

14:23:16	
Info	

execute: var isnew 0

14:23:16	
Info	

execute: if "§isnvs eth.on" "var isnew 0"

14:23:16	
Info	

execute: isnvs eth.on

14:23:16	
Info	

execute: var isnew 0

14:23:16	
Info	

execute: on -lo iol 9 "leds 255" "sleep 500" restart

14:23:16	
	

io trigger installed

14:23:16	
Info	

execute: on iol 9 "ledboot 50 0"

14:23:16	
	

io trigger installed

14:23:16	
Info	

execute: function start "ledg 50 0 3000" server mqtt

14:23:16	
	

function 'start' installed

14:23:16	
Info	

execute: on -o wifi 1 start

14:23:16	
	

network trigger installed

14:23:16	
Info	

execute: on -o eth 1 start

14:23:16	
	

network trigger installed

14:23:16	
Info	

execute: on wifi 0 "ledb 0"

14:23:16	
	

network trigger installed

14:23:16	
Info	

execute: on eth 0 "ledb 0"

14:23:16	
	

network trigger installed

14:23:16	
Info	

execute: on -o server 2 "after -o -m 500 0 restart"

14:23:16	
	

server trigger installed

14:23:16	
Info	

execute: if "var skipstart" "ledb 255 0 5000" "ledboot 50" "ap EBUS" start exit

14:23:16	
Info	

execute: var skipstart

14:23:16	
	

variable skipstart = 0

14:23:16	
Info	

execute: if "isnvs sta.on" "ledb 10 255 5000" "sleep 100" sta

14:23:16	
Info	

execute: isnvs sta.on

14:23:16	
Info	

execute: ledb 10 255 5000

14:23:16	
Info	

execute: led 0 10 255 5000

14:23:16	
	

0 duty requested to 10

14:23:16	
Info	

execute: sleep 100

14:23:16	
Info	

leds: duty set to 10

14:23:16	
Info	

execute: sta

14:23:16	
	

sta connecting to 'PLUSNET-G9FJ'

14:23:16	
Warning	

wifi:

14:23:16	
	

Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2

14:23:19	
Info	

cmd_wifi: sta connect

14:23:20	
Info	

esp_netif_handlers: sta ip: 192.168.1.136, mask: 255.255.255.0, gw: 192.168.1.1

14:23:20	
Info	

cmd_wifi: sta got ip

14:23:20	
Info	

repl_event_task: sta event: 1

14:23:20	
Info	

execute: start

14:23:20	
Info	

execute: ledg 50 0 3000

14:23:20	
Info	

execute: led 1 50 0 3000

14:23:20	
	

1 duty set to 50

14:23:20	
Info	

execute: server

14:23:20	
Info	

execute: if "var isnew" "ap EBUS" "sta -w" else "if \"isnvs ap.on\" ap"

14:23:20	
Info	

execute: var isnew

14:23:20	
	

variable isnew = 0

14:23:20	
Info	

execute: if "isnvs ap.on" ap

14:23:20	
Info	

execute: isnvs ap.on

14:23:20	
Info	

execute: var eth 0

14:23:20	
Info	

execute: if "eth -t" "var eth 1" else "if \"eth -wt\" \"var eth 2\""

14:23:20	
Info	

execute: eth -t

14:23:20	
Info	

initialise_mdns: mdns configured

14:23:20	
Info	

esp-rest: Starting HTTP Server

14:23:20	
	

server started

14:23:20	
Info	

execute: mqtt

14:23:21	
	

Ethernet on USR-ES1 not available

14:23:21	
Warning	

execute: command failed: eth -t = 1

14:23:21	
Info	

execute: if "eth -wt" "var eth 2"

14:23:21	
Info	

execute: eth -wt

14:23:21	
Info	

sntpTimeCb: got NTP time: 2025-01-29 14:23:21

14:23:22	
	

Ethernet on W5500 module not available

14:23:22	
Warning	

execute: command failed: eth -wt = 1

14:23:22	
Info	

execute: io -a

14:23:22	
Info	

execute: function chkethcfg "if \"eth -c\" \"var eth 1\""

14:23:22	
	

function 'chkethcfg' installed

14:23:22	
Info	

execute: if "!var isnew" "if \"!var eth\" chkethcfg"

14:23:22	
Info	

execute: var isnew

14:23:22	
	

variable isnew = 0

14:23:22	
Info	

execute: if "!var eth" chkethcfg

14:23:22	
Info	

execute: var eth

14:23:22	
	

variable eth = 0

14:23:22	
Info	

execute: chkethcfg

14:23:22	
Info	

execute: if "eth -c" "var eth 1"

14:23:22	
Info	

execute: eth -c

14:23:22	
Warning	

execute: command failed: eth -c = 1

14:23:22	
Info	

execute: function enet eth "leds 50"

14:23:22	
	

function 'enet' installed

14:23:22	
Info	

execute: if "isnvs eth.on" "if \"var eth\" enet"

14:23:22	
Info	

execute: isnvs eth.on

14:23:22	
Info	

execute: if "isnvs ebus.host" ebus exit

14:23:22	
Info	

execute: isnvs ebus.host

14:23:22	
Info	

execute: ebus

14:23:22	
Info	

init_ebus: switching to enhanced eBUS mode on TCP port (with bitrate deviation 0 and enhanced arbitration delay 200 us)

14:23:22	
Info	

setup_ebus_host_tcp: listening on port 9999

14:23:22	
Info	

setup_uart: eBUS UART 1 on TX=10(inv) RX=7(inv) with 2400 Bd

14:23:22	
Info	

uart: queue free spaces: 16

14:23:22	
Info	

addMdnsService: mDNS service added

14:23:22	
Info	

execute: exit

14:23:22	
Info	

execute: command exit

14:23:22	
Info	

execute: command exit

14:23:22	
Info	

replExecuteFromFile: successfully executed 35/35 commands from '/data/main.eas'

14:23:22	
	

Welcome to eBUS Adapter Shield Interface!

14:23:23	
Info	

wait_host_tcp: accepted: 192.168.1.243

14:23:23	
Info	

loop: host reset

SzosszeNET, 2025-01-29:
Not sure what went wrong initially.

Anyhow was able to recover via CLI.
Run "ota --run" that updated again and this time around the board got updated and seems to be back to normal

[image]

closed 2025-01-29 by SzosszeNET

john30, 2025-01-30:
duplicate of #1472 and https://github.com/john30/ebusd-esp32/issues/93