john30/ebusd #1480: Vaillant missing Heating Circuits from autoconfig

peedign, 2025-02-02:
### Description

autoconfig is not picking up 1 out of 2 VR70 and none of the VR92. Occassinally the scan picks up some signal under address ec and assigns a solar slave device, which is actually not existing.

### Actual behavior

some slave devices are not getting automatically scanned

### Expected behavior

all slave devices are scanned properly

### ebusd version

current source from git

### ebusd arguments

N/A

### Operating system

other

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

> max symbol rate: 228
min arbitration micros: 10
max arbitration micros: 988
min symbol latency: 5
max symbol latency: 60
scan: finished, some messages pending
reconnects: 0
masters: 7
messages: 707
conditional: 0
poll: 276
update: 29
address 03: master https://github.com/john30/ebusd/issues/11
address 08: slave https://github.com/john30/ebusd/issues/11, scanned "MF=Vaillant;ID=HMU00;SW=0522;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master https://github.com/john30/ebusd/issues/2
address 15: slave https://github.com/john30/ebusd/issues/2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0100;HW=5904", loaded "vaillant/26.vr_71.csv"
address 30: master https://github.com/john30/ebusd/issues/3
address 31: master https://github.com/john30/ebusd/issues/8, ebusd
address 35: slave https://github.com/john30/ebusd/issues/3, scanned "MF=Vaillant;ID=VR_92;SW=0514;HW=1204"
address 36: slave https://github.com/john30/ebusd/issues/8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0101;HW=5804", loaded "vaillant/52.vr_70.csv"
address 53: slave, scanned "MF=Vaillant;ID=VR_70;SW=0101;HW=5804"
address 70: master https://github.com/john30/ebusd/issues/4
address 71: master https://github.com/john30/ebusd/pull/9
address 75: slave https://github.com/john30/ebusd/issues/4, scanned "MF=Vaillant;ID=VR_92;SW=0514;HW=1204"
address 76: slave https://github.com/john30/ebusd/pull/9, scanned "MF=Vaillant;ID=VWZ00;SW=0522;HW=5103", loaded "vaillant/76.vwz.csv"

[image]

Discussed in https://github.com/john30/ebusd/discussions/1454

jan666, 2025-02-03:
for VR70 maybe have a look at https://github.com/john30/ebusd/issues/1113#issuecomment-2510897198

peedign, 2025-02-08:
unfortunately, that did not help. I think ebusd ignores the second VR_70 completely

JonesGW-MGD, 2025-02-08:
Did you try creating a copy of 52.vr_70.csv as 53.vr_70.csv as suggested here: https://github.com/john30/ebusd/discussions/1454#discussioncomment-11903059

peedign, 2025-02-08:
Indeed I tried a number of times, various ways, following your hints.

 I think something is not quite right with my setup as no csv was loaded at all after local scans (beside vr_70s). Now I'm reinstalling HA ebusd add-on. Though, I am still hoping for a fix for the online config at some point.

JonesGW-MGD, 2025-02-10:
Cool - well, when you get all the config files loaded locally that you had previously when you used the repository service, maybe resurrect your previous 'discussion' thread with some updates and the community may be able to help get the wiring centre and controllers recognised too.

The fact that ebusd 'found' a solar element that isn't in your set up would worry me though, although 'one problem at a time'!

john30, 2025-04-13:
not an ebusd issue but ebusd-configuration, so please post there: https://github.com/john30/ebusd-configuration/issues/

closed 2025-04-13 by john30