john30/ebusd #1487: mqtt hassio related errors

garageboxnv, 2025-02-15:
### Description

I have ebusd running with home assistant through mqtt discovery and am really happy but some warnings pop up related with the autodiscovered sensors, not sure though if I should put them here or in the configuration repository:
-  Invalid value for number.ebusd_700_outsidetempavg: -0.1875 (range 0.0 - 100.0) 
- sensor.ebusd_hmu_state_energy (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('energy') it is using; expected one of ['Gcal', 'Wh', 'TWh', 'cal', 'GJ', 'GWh', 'kJ', 'kcal', 'kWh', 'MWh', 'MJ', 'mWh', 'J', 'Mcal']
- sensor.ebusd_pms_internalhctempdesired (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['°C', 'K', '°F']
- sensor.ebusd_pms_internalhwctempdesired (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'None' which is not a valid unit for the device class ('temperature') it is using; expected one of ['°C', 'K', '°F']
- Ignoring invalid option received on topic 'ebusd/hwc/Mode', got '9', allowed: inactive, mixer, fixed, hwc, returnincr, pool, circulation
- Ignoring invalid option received on topic 'ebusd/hmu/RunDataStatuscode', got '0', allowed: Frost_protection, Standby, Heating_Compressor_shutdown, Heating_Compressor_blocked, Heating_Prerun, Heating_Compressor_active, Heating_Overrun, Cooling_Compressor_shutdown, Cooling_Compressor_blocked, Cooling_Prerun, Cooling_Compressor_active, Cooling_Overrun, Heating_Immersion_heater_active, Warm_Water_Compressor_blocked, Warm_Water_Prerun, Warm_Water_Compressor_active, Warm_Water_Immersion_Heater_active, Warm_Water_Overrun, Heating_Immersion_Heater_shutdown, Heating_Immersion_Heater_blocked, Warm_Water_Immersion_Heater_shutdown, Warm_Water_Immersion_Heater_blocked, Air_purging_program_active, Compressor_oil_heating_activated, Fan_1_blocked, Fan_1_air_inlet_too_high, Fan_1_air_inlet_too_low, Fan_2_blocked, Building_circuit_flow_too_low, Building_circuit_pump_fault, Frequency_converter_fault_Compressor, Frequency_converter_fault_Mains_voltage, Frequency_converter_fault_Overheating, Deicing_time_too_long, Deicing_flow_temperature_too_low, Compressor_outlet_temperature_too_high, Hot_gas_temperature_switch_open, Fan_1_Wind, Fan_2_Wind, Current_limt_active, High_pressure_switch_open, Compressor_outlet_temperature_too_high_1, Evaporation_temperature_too_low, Condensation_temperature_too_low, Evaporation_temperature_too_high, Condensation_temperature_too_high, Building_circuit_return_temperature_too_low, Building_circuit_return_temperature_too_high, Immersion_heater_Flow_temperature_too_high, Deicing_active, Frequency_converter_fault, Connection_fault_frequency_converter, _4_port_valve_position_fault

Not sure if this the right place to post it

john30, 2025-04-13:
all besides the first one are related to ebusd-configuration, so please post there instead: https://github.com/john30/ebusd-configuration/issues/

wrt outsidetempavg: please post the definition topic for that sent on mqtt

garageboxnv, 2025-04-17:
The mqtt definition topic is: 
```
unique_id: ebusd_700_OutsideTempAvg_value
device:
  identifiers: ebusd_700
  manufacturer: ebusd.eu
  via_device: ebusd
  sw_version: '24.1'
  suggested_area: Heating
  name: ebusd 700
value_template: '{{value_json["value"].value}}'
state_topic: ebusd/700/OutsideTempAvg
command_topic: ebusd/700/OutsideTempAvg/set
unit_of_measurement: °C
device_class: temperature
name: 'OutsideTempAvg '
``` 
I will repost the others

john30, 2025-04-24:
weird, seems right to me and basically HA should also accept a negative value as temperature. so this needs to be addressed on HA, probably here: https://github.com/home-assistant/core/issues
or did you somehow add a limit or range to that readng?

closed 2025-04-24 by john30