john30/ebusd #1500: eBUSd Not Loading Configuration for Vaillant BAI00  – No Data Stored (Adapter Shield C6)

Hotelk52339, 2025-02-25:
### Description

eBUSd detects my Vaillant BAI00 boiler but **does not load the configuration**. Previously, when running `ebusctl info`, the output included:  
```
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0302;HW=0903", loaded config BAI00
```  
Now, it only shows:  
```
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0302;HW=0903"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```  
Additionally, when I check available data using `ebusctl find`, everything returns `"no data stored"`:  
```
basv AdaptHeatCurve = no data stored  
basv ContinuosHeating = no data stored  
basv CylinderChargeHyst = no data stored  
...  
basv WaterPressure = no data stored  
scan.08  = Vaillant;BAI00;SW=0302;HW=0903
```  
The boiler is detected, but **the configuration is missing** and no values are returned.  

I have verified that `/etc/ebusd/configs/08.bai.csv` **exists**, but `ebusctl reload` does not fix the issue. The gateway is reachable, and there are no relevant errors in `journalctl -xeu ebusd.service`

### Actual behavior

- eBUSd detects the Vaillant BAI00 boiler but does **not** load the configuration.  
- Running `ebusctl find` returns `"no data stored"` for all parameters.  
- Restarting eBUSd, reloading configs, and verifying connectivity **does not resolve the issue**.  

### Expected behavior

- eBUSd should **load the configuration** for the detected BAI00 device.  
- Running `ebusctl find` should **return actual values** instead of `"no data stored"`.  
- The output of `ebusctl info` should **include**:  
```
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0302;HW=0903", loaded config BAI00

### ebusd version

24.1

### ebusd arguments

EBUSD_OPTS="--configpath=/etc/ebusd/configs --device=ens:192.168.1.111:9999 --scanconfig"

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

x64

### Dockerized

None

### Hardware interface

other

### Related integration

TCP (cmdline client like ebusctl or netcat)

### Logs

### **Logs**  
Output of `ebusctl info`:  
```
version: ebusd 24.1.24.1-33-gd8b2ae1b  
device: 192.168.1.111:9999, TCP, enhanced, firmware 1.1[5210].1[5210]  
signal: acquired  
symbol rate: 24  
max symbol rate: 41  
min arbitration micros: 3  
max arbitration micros: 4  
min symbol latency: 4  
max symbol latency: 7  
scan: finished  
reconnects: 0  
masters: 2  
messages: 34  
conditional: 0  
poll: 0  
update: 0  
address 03: master #11  
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0302;HW=0903"  
address 31: master #8, ebusd  
address 36: slave #8, ebusd  
```  
Output of `ebusctl find`:  
```
basv AdaptHeatCurve = no data stored  
basv ContinuosHeating = no data stored  
basv CylinderChargeHyst = no data stored  
...  
basv WaterPressure = no data stored  
scan.08  = Vaillant;BAI00;SW=0302;HW=0903  
```

wrongisthenewright, 2025-02-25:
Not an expert here, so take my answer with caution.
It may be related to the new kind of online configuration files repo.
Given that you already have your local csv file you may try to force the usage of local conf via dedicated command line option:

`-c, --configpath=PATH`

in your case `--configpath=/etc/ebusd/configs`

Hotelk52339, 2025-02-26:
I tried running `ebusd` with a forced local config path:  

```bash
ebusd --configpath=/etc/ebusd/configs --loglevel=debug
```

The service starts, but the `ebusctl find` command still returns `no data stored`, and `ebusctl scan` either gives no result or returns `ERR: scan already running`.  

Checked logs with:  

```bash
journalctl -u ebusd -n 50 --no-pager
```

I see errors like:
- `Failed to parse PID from file /run/ebusd.pid: Invalid argument`
- `Failed with result 'protocol'`

I also tried running `ebusd` with different parameters, including:  

```bash
ebusd --configpath=/etc/ebusd/configs --loglevel=debug --device=ens:192.168.1.111 --scanconfig
```

But `ebusctl find` still returns `no data stored`.  

The configuration files exist in `/etc/ebusd/configs/`

Any ideas on what could be causing this? Is there anything else I should check?

closed 2025-02-27 by Hotelk52339