john30/ebusd #1501: BUG? ebusctl read -f YieldThisYear not working (15.uih.scv version from next)

gima84, 2025-02-26:
### Description

Hi, thanks a lot for that great piece of software! I am not sure if it is related to ebusd itself, ebusd-configuration or adapter shield v5 (via Wifi). 

After a hardware migration + update of ebusd + new Ebus adapter (eBUS Adapter Shield v5) I have an issue with reading one of the most important (at least for me :-)) field.

The field seems to be available:

`root@ebus-daemon:~/ebus.github.io/next/en/vaillant# ebusctl find |grep YieldThisYear
uih YieldThisYear = no data stored
root@ebus-daemon:~/ebus.github.io/next/en/vaillant# `

When I try to read it, I just got back "ERR: SYN received or ERR: wrong symbol received"

`root@ebus-daemon:~/ebus.github.io/next/en/vaillant# ebusctl read -f YieldThisYear
ERR: SYN received

root@ebus-daemon:~/ebus.github.io/next/en/vaillant# ebusctl read -f YieldThisYear
ERR: wrong symbol received`

My config:

Hardware: Vaillant VWS 83/3

ebusctl I:
`root@ebus-daemon:~/ebus.github.io/next/en/vaillant# ebusctl i
version: ebusd 24.1.24.1
update check: OK, device firmware v5 1[5210] up to date, broadcast.csv: different version available, memory.csv: different version available, vaillant/08.ehp.csv: different version available, vaillant/15.uih.csv: different version available, vaillant/23.ehp.cc.csv: different version available, vaillant/25.ehp.hwc.csv: different version available, vaillant/50.ehp.mc.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/general.csv: different version available, vaillant/scan.csv: different version available
device: ebus-adapter:9999, TCP, enhanced, firmware 1.1[5210].1[5210]
signal: acquired
symbol rate: 41
max symbol rate: 129
min arbitration micros: 0
max arbitration micros: 561
min symbol latency: 0
max symbol latency: 150
scan: finished
reconnects: 0
masters: 6
messages: 634
conditional: 219
poll: 129
update: 66
address 00: master #1
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/08.ehp.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0374;HW=6901", loaded "vaillant/15.uih.csv"
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/25.ehp.hwc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 3f: master #23
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/50.ehp.mc.csv"
address 7f: master #24

`

Any ideas / hints?

### Actual behavior

ebusctl read -f YieldThisYear

ERR: SYN received

or

ERR: wrong symbol received

### Expected behavior

ebusctl read -f YieldThisYear
1234 (kwh)

### ebusd version

24.1

### ebusd arguments

"--scanconfig --device=ens:ebus-adapter:9999 --latency=30 --configpath=/root/ebus.github.io/next/en/ --mqttvar=filter-direction=r|u|^w --loglevel=debug --httpport=8889 --htmlpath=/var/ebusd/html --mqtthost homeassistant --mqttport 1883 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqttjson --mqttuser 'mqtt' --mqttpass 'xxxxx'"

### Operating system

Debian 12 (Bookworm) / Ubuntu 22-23 / Raspberry Pi OS 12 (including lite)

### CPU architecture

arm64

### Dockerized

None

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

Reproducing the issue:
`root@ebus-daemon:~/ebus.github.io/next/en/vaillant# date;ebusctl read -f YieldThisYear;date
Mi 26. Feb 09:33:59 CET 2025
ERR: SYN received

Mi 26. Feb 09:34:01 CET 2025
root@ebus-daemon:~/ebus.github.io/next/en/vaillant# date;ebusctl read -f YieldThisYear;date
Mi 26. Feb 09:34:06 CET 2025
ERR: SYN received

Mi 26. Feb 09:34:07 CET 2025
root@ebus-daemon:~/ebus.github.io/next/en/vaillant# date;ebusctl read -f YieldThisYear;date
Mi 26. Feb 09:34:09 CET 2025
ERR: SYN received

Mi 26. Feb 09:34:10 CET 2025
root@ebus-daemon:~/ebus.github.io/next/en/vaillant# date;ebusctl read -f YieldThisYear;date
Mi 26. Feb 09:34:14 CET 2025
ERR: wrong symbol received

Mi 26. Feb 09:34:15 CET 2025`

[ebusd_log.txt](https://github.com/user-attachments/files/18980294/ebusd_log.txt)

gima84, 2025-02-26:
Today I have switched from Wifi to USB (with RPI 3b+) and it is working fine.

john30, 2025-04-13:
not an ebusd issue then, so closing. feel free to re-open here: https://github.com/john30/ebusd-esp32/issues/

closed 2025-04-13 by john30