john30/ebusd #1506: Ebusd via docker compose local config files not possible

DeltaTango69, 2025-03-01:
### Description

Ich probiere jetzt seit mehreren Tagen lokale csv Dateien zu verwenden. Bekomme es aber leider nicht hin.

Meine docker-compose.yml

```
services:
  ebusd:
    image: john30/ebusd
    container_name: ebusd
    restart: unless-stopped
    network_mode: host

    volumes:
      - ./config:/etc/ebusd
      - ./mqtt/mqtt-hassio.cfg:/ebusd/mqtt-hassio.cfg

    environment:

      EBUSD_DEVICE: "ens:192.168.178.12:9999"
      EBUSD_LATENCY: 10
      EBUSD_CONFIGPATH: "/etc/ebusd"
      # EBUSD_CONFIGPATH: "https://ebus.github.io/next/"
      EBUSD_SCANCONFIG: ""
      EBUSD_POLLINTERVAL: 60

      # MQTT options:

      EBUSD_MQTTHOST: "192.168.178.5"
      EBUSD_MQTTPORT: 1883
      EBUSD_MQTTCLIENTID: "ebusd"
      EBUSD_MQTTUSER: "xxxxx"
      EBUSD_MQTTPASS: "xxxxx"
      EBUSD_MQTTTOPIC: "ebusd"
      EBUSD_MQTTGLOBAL: "ebusd/global/"
      EBUSD_MQTTINT: "/ebusd/mqtt-hassio.cfg"
      EBUSD_MQTTJSON: ""
```

Dateien liegen im docker container an der richtigen Stelle auf die auch die configpath Variable verzweigt:

```
root@Homeserver:/etc/ebusd# ls -lra de/vaillant
total 886
-rwxrwxrwx 1 1000 root   628 Feb 15 23:00 yield8485r.inc
-rwxrwxrwx 1 1000 root   604 Feb 15 23:00 yield8485.inc
-rwxrwxrwx 1 1000 root   604 Feb 15 23:00 yield4445.inc
-rwxrwxrwx 1 1000 root   604 Feb 15 23:00 yield3f40.inc
-rwxrwxrwx 1 1000 root   604 Feb 15 23:00 yield3d43.inc
-rwxrwxrwx 1 1000 root   667 Feb 15 23:00 timertariff.inc
-rwxrwxrwx 1 1000 root   647 Feb 15 23:00 timerhwc.inc
-rwxrwxrwx 1 1000 root   633 Feb 15 23:00 timerhc.inc
-rwxrwxrwx 1 1000 root   677 Feb 15 23:00 timercool.inc
-rwxrwxrwx 1 1000 root   647 Feb 15 23:00 timercc.inc
-rwxrwxrwx 1 1000 root   926 Feb 15 23:00 timer.inc
-rwxrwxrwx 1 1000 root   727 Feb 15 23:00 tempsetpoints.inc
-rwxrwxrwx 1 1000 root   530 Feb 15 23:00 service.inc
-rwxrwxrwx 1 1000 root   792 Feb 15 23:00 scan.csv
-rwxrwxrwx 1 1000 root   238 Feb 15 23:00 roomtempoffset.inc
-rwxrwxrwx 1 1000 root   316 Feb 15 23:00 quick.inc
-rwxrwxrwx 1 1000 root  1435 Feb 15 23:00 mcmode.inc
-rwxrwxrwx 1 1000 root   423 Feb 15 23:00 iotesthp.inc
-rwxrwxrwx 1 1000 root   313 Feb 15 23:00 iotestbsol.inc
-rwxrwxrwx 1 1000 root   348 Feb 15 23:00 iotestbmc.inc
-rwxrwxrwx 1 1000 root   307 Feb 15 23:00 iotestact.inc
-rwxrwxrwx 1 1000 root   407 Feb 15 23:00 iotest630.inc
-rwxrwxrwx 1 1000 root   404 Feb 15 23:00 iotest620.inc
-rwxrwxrwx 1 1000 root   913 Feb 15 23:00 hwcmode.inc
-rwxrwxrwx 1 1000 root  1177 Feb 15 23:00 hcmode.inc
[... 142 lines cut ...]

### Dockerized

latest

### Hardware interface

Adapter Shield v5 via WiFi

### Related integration

MQTT Home Assistant via mqtt-hassio.cfg

### Logs

kein ebusd.log ?

root@Homeserver:/var/log# ls -l
total 44
drwxr-xr-x 1 root root  4096 Nov 12 07:39 apt
-rw-rw---- 1 root utmp     0 Nov 11 00:00 btmp
-rw-r--r-- 1 root root  2507 Nov 12 07:39 dpkg.log
-rw-r--r-- 1 root root  3232 Nov 11 00:00 faillog
-rw-rw-r-- 1 root utmp 29492 Nov 11 00:00 lastlog
-rw-rw-r-- 1 root utmp     0 Nov 11 00:00 wtmp

Koky05, 2025-03-03:
In case you want to use local copy of configuration files you need one level up folder. Because that level contain _templates.csv, broadcast.csv and memory.csv. At least _templates.csv contains definitions of message types.
Starting from here:
[/etc/ebusd](https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x/de)

DeltaTango69, 2025-03-03:
Thank you very much - that's was it. Please add this **IMPORTANT** information in the wiki, so that maybe other users can avoid days of try and error

closed 2025-03-03 by DeltaTango69